MichaIng/DietPi #3656: DietPi-Software | Python3 and Python2 dependants

MichaIng, 2020-07-06:
We need to get rid of installing EOL Python2, so we need to:
1. Clearly separate Python2-Pip and Python3-Pip
2. List Python2 dependants and check whether there are Python3 migrations available, else check or bring this to discussion on software devs.
3. If there is no plan for a Python3 migration, drop support for affected software.

MichaIng, 2020-07-06:
- 🈺 Mopidy: Python3 from Buster on, now as well for ARMv8/arm64. Probably we can arrange some backport for Stretch.
- 🈺 EmonPi: We use an own Python2 fork currently but upstream is already Python3-compatible. Larger project to migrate since I have never uses this piece of software or even the hardware part. Help and testing required to hopefully do DietPi-adjustments within DietPi-Software so that we can pull from the original upstream repo.
- 🈺 MotionEye: Experimental support in progress: https://github.com/ccrisan/motioneye/pull/1805
- 🈴 HTPC-Manager:
  - Original project dead: https://github.com/styxit/HTPC-Manager
  - Fork (that we use) doesn't show much progress lately, Python3 discussion doesn't look promising: https://github.com/Hellowlol/HTPC-Manager/issues/611
If anyone is interested in this piece of software, contribution and fresh wind seems to be desperately required 🚀 💪!
- 🈯 SABnzbd: Python3 release in RC1 stage, expected to be released before DietPi v6.32: https://github.com/sabnzbd/sabnzbd/releases
- 🈴 CouchPotato: Not Python3 compatible, no issue/request so far, even cementation: https://github.com/CouchPotato/CouchPotatoServer/commit/ab25630f8e2cb4f0868ad808297e58142a9e6f5a
Request for Python3 migration desperately required are I cannot see ANY steps in this direction so far 👎: No GitHub issues enabled, forum [down](https://couchpota.to/forum/)
- 🈯 Google AIY
- 🈯 OctoPrint

maartenlangeveld, 2020-09-21:
Samba server (95) installs python2.7

MichaIng, 2020-09-21:
Good find. On Bullseye it pulls Python 3: https://packages.debian.org/bullseye/samba
Until Debian Buster there are many cases where packages pull Python 2 as dependency:
```
2020-09-21 14:50:19 root@VM-Buster:/# apt rdepends python | wc -l
2573
```
This will not change. But at least where we have control, we should try to migrate until Bullseye release, although it currently ships a compatibility package, now called `python2`: https://packages.debian.org/bullseye/python2

MichaIng, 2021-05-18:
Okay, all items where we can currently do something about have been addressed. In case of MotionEye and Firefox Sync Server, we need to wait until upstream migrations have been finished/become stable, but we'll address those in dedicated issues.

CouchPotato will likely (and sadly!) be removed from the list, as it doesn't seem to be developed anymore. Dealt with in dedicated issue as well: #4323

closed 2021-05-18 by MichaIng