MichaIng/DietPi #3658: Update failed: missing directory

JavaDev55, 2020-07-06:
#### Details:
- Date           | Tue Jul  7 07:23:38 AEST 2020
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l) (ID=11)
- Kernel version | Linux DietPi 4.14.66+ #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018 armv7l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | chmod a+x /mnt/dietpi_userdata
- Exit code      | 1
- Software title | DietPi-Patch
#### Steps to reproduce:

1. ... Ran DietPi-Update
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
chmod: cannot access '/mnt/dietpi_userdata': No such file or directory
```

MichaIng, 2020-07-06:
Many thanks for your report.

Did you manually remove or move `dietpi_userdata` to an external drive?
```
ls -Al /mnt/ "$(readlink -f /mnt/dietpi_userdata)"
```

Probably the external drive it was moved to got disconnected. However if you never used it for any actual software or data, you can remove the left symlink. If there is not even a symlink, it is safe anyway to create the directory:
```
[[ -L '/mnt/dietpi_userdata' ]] && rm /mnt/dietpi_userdata
mkdir /mnt/dietpi_userdata
```

JavaDev55, 2020-07-07:
Yes, Michael, it is an HC1 with file system moved to an SSD.

Thx for the quick response 👍

Cheers

Jim

JavaDev55, 2020-07-07:
Just re-read your email ... No I did not move it manually.  I used the
DietPi menu option to move it.

Joulinar, 2020-07-07:
Pls check if the link to your SSD is still active as described by @MichaIng 

`ls -Al /mnt/ "$(readlink -f /mnt/dietpi_userdata)"`

JavaDev55, 2020-07-07:
Yes it is.  Properly plugged in & functional.

MichaIng, 2020-07-07:
Can you please run those commands and paste the output here? Since your initial error message states that either the drive is not properly mounted or the link is wrong:
> chmod: cannot access '/mnt/dietpi_userdata': No such file or directory

The command output allows to derive which of the two cases is true.

JavaDev55, 2020-07-07:
I ran the commands, output below.  Since the root file system has been moved to the SSD with only the kernel on the micro-sd card, nothing would be working if the SSD was in any way disconnected from the SATA connector or not working.  In the output below you can see this mapping.  Thank you for your attention to this issue.

jim@DietPi:~$ ls -Al /mnt/ "$(readlink -f /mnt/dietpi_userdata)"
ls: cannot access '/mnt/dietpi_userdata': No such file or directory
/mnt/:
total 0
jim@DietPi:~$ ls -Al /mnt
total 0
jim@DietPi:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            931M     0  931M   0% /dev
tmpfs           200M   26M  174M  13% /run
**/dev/sda1       229G  1.2G  228G   1% /**
tmpfs           998M     0  998M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           998M     0  998M   0% /sys/fs/cgroup
tmpfs            50M  156K   50M   1% /var/log
tmpfs           997M   32K  997M   1% /tmp
/dev/mmcblk1p1   71M   33M   39M  46% /boot
jim@DietPi:~$ ls /
bin  boot  btmp  dev  DietPi  dpkg.log  etc  home  lib  lost+found  mnt  opt  proc  root  run  sbin  srv  sys  tmp  usr  var  wtmp

jim@DietPi:~$ [[ -L '/mnt/dietpi_userdata' ]] && rm /mnt/dietpi_userdata
jim@DietPi:~$ sudo mkdir /mnt/dietpi_userdata
jim@DietPi:~$ ls -Al /mnt/ "$(readlink -f /mnt/dietpi_userdata)"
/mnt/:
total 4
drwxr-xr-x 2 root root 4096 Jul  8 07:27 dietpi_userdata

/mnt/dietpi_userdata:
total 0

JavaDev55, 2020-07-07:
BTW - Now that the directory exists, I re-ran the update and it ran to completion without error.  Thank You

Joulinar, 2020-07-07:
ah ok so you did not moved the user data directory only. You moved entire RootFS. Than it's fine that there is no link. But strange that you lost the whole user data directory.

MichaIng, 2020-07-07:
Ahh, I think that was exactly fixed with v6.31 😄: https://github.com/MichaIng/DietPi/commit/a9d2ea33f3a546ec263377b769fb4aeaaef8d7f3#diff-42433c15598bdf803f209563b57e8d5b

Before, moving rootfs didn't include any content of /mnt, including empty mount point directories etc.

Joulinar, 2020-07-09:
I would go to close the ticket. Feel free to reopen if needed

closed 2020-07-09 by Joulinar