MichaIng/DietPi #3659: Dietpi-update: Failed - Unable to access any update server.

Smiggel, 2020-07-08:
### Creating a bug report/issue

#### Required Information
[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6
curl: (6) Could not resolve host: raw.githubusercontent.com
[ INFO ] DietPi-Update | No or error response from: https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/server_version-6 ()
[ INFO ] DietPi-Update | Checking mirror: https://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6
curl: (6) Could not resolve host: dietpi.com
[ INFO ] DietPi-Update | No or error response from: https://dietpi.com/downloads/dietpi-update_mirror/master/server_version-6 ()
[FAILED] DietPi-Update | Unable to access any update server. Please check your network connection, then rerun dietpi-update.
If this error persists, please report at: https://github.com/MichaIng/DietPi/issues

#### Additional Information (if applicable)
Running version DietPi v6.30.0

#### Steps to reproduce
1. run dietpi-update

#### Expected behaviour
- update process would have to start?

#### Actual behaviour
- update failes as it can not contact any update server

#### Extra details
- Raspberry Pi is connected by ethernet and accessible. Network is working.

Smiggel, 2020-07-08:
Update is working after reboot of the pi.

closed 2020-07-08 by Smiggel

Joulinar, 2020-07-08:
Hi,

Many thanks for your report. Looks like you have local issues to resolve DNS

```
Could not resolve host: raw.githubusercontent.com
Could not resolve host: dietpi.com
```

Do you use any local DNS server?