MichaIng/DietPi #3660: qBittorrent: port change does not persist after reboot

linxiaopi, 2020-07-08:
### Creating a bug report/issue

#### Required Information
- DietPi version:
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=31
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version:
10.4
buster

- Kernel version:
Linux DietC3Pi0 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux

- SBC model:
RPi 2 Model B (armv7l)

- Power supply used:
5v 2A 

- SDcard used:
SanDisk Ultra 8GB

#### Additional Information (if applicable)
- Software title:
qBittorrent, installed with 'dietpi-software'

#### Steps to reproduce
1. Install qBittorrent from optimized software list
2. Login to qBittorrent Web UI on default port (1340)
3. Options > Web UI > change port to unused port (e.g., 43000) > Save
4. Logout of Web UI
5. Login to qBittorrent Web UI on new port (43000)
6. Logout of Web UI
7. Reboot Raspberry Pi
8. Login to qBittorrent on new port (4300) > FAIL: 'This site can't be reached... refused connection'
9. Login to qBittorrent on default port (1340) > SUCCESS
10. Options > Web UI > port has changed back to 1340

#### Expected behaviour
After changing port of qBittorrent's Web UI and rebooting the Raspberry Pi, should be able to access qBittorrent's Web UI on the new port.

#### Actual behaviour
After reboot, qBittorrent Web UI port changes back to default port (1340).

#### Extra details
This also occurred on the previous version of dietpi (I just update to 6.31.2 and tried to change ports again).

Joulinar, 2020-07-08:
Hi,

many thanks for your report. Changing the settings within the gui will not have any effect because the port is hard coded within the `qbittorrent.service`. You can adjust as follow

1. `nano /etc/systemd/system/qbittorrent.service`
2. change `ExecStart=/usr/bin/qbittorrent-nox --webui-port=43000`
3. save and exit
4. `systemctl daemon-reload`
5. `systemctl restart qbittorrent.service`

now your system should run on `43000`

MichaIng, 2020-07-08:
Probably we should set the port via config file if possible 🤔.

However when changing the service file directly, not that this is not reinstall-persistent. Generally, also when altering services of a dpkg/apt package, it is good to do this via drop-in config overrides, here via `/etc/systemd/system/qbittorrent.service.d/yourOverride.conf`. This is done as well when doing:
`dietpi-services` > `qbittorrent` > `Edit` 🙂

MichaIng, 2020-07-19:
Lol, I accidentally fixed it while dealing with another issue: https://github.com/MichaIng/DietPi/commit/30e6371756ac43f13b334d230325096be8853d77
We already set the port via config file but still did set it via cmd option which overrides the config file. With this change it should work.

linxiaopi, 2020-08-24:
Great! Thanks for the fixing it quickly.
Now, I guess I just need to wait for v6.32 to come out or to do the manual changes.

MichaIng, 2020-08-24:
Or you manually remove the ` --webui-port=1340` from your `/etc/systemd/system/qbittorrent.service` + `systemctl daemon-reload; systemctl restart qbittorrent`.

closed 2020-08-31 by MichaIng