MichaIng/DietPi #3662: Attempting to made rootfs readonly- mount: /: mount point is busy.

kotx, 2020-07-09:
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=31
G_DIETPI_VERSION_RC=2`
- Distro version | `10.4`
- Kernel version | `Linux Monoke 5.4.43-sunxi64 #20.05.2 SMP Tue Jun 2 17:20:17 CEST 2020 aarch64 GNU/Linux`
- SBC model | `NanoPi NEO2 (aarch64)`
- Power supply used | 5V 1A Anker
- SDcard used | PNY microSDHC Elite 16GB

- Software title | DietPi-Drive_Manager
- Was the software title installed freshly or updated/migrated? Yes
- Can this issue be replicated on a fresh installation of DietPi? This is basically a new install, except for a samba server and JellyFin server.

- Bug report ID | `d2ea77de-0c75-4a37-b17a-4ad025004ba7`

#### Steps to reproduce
1. Open DietPi-Drive_Manager
2. Try to set mmcblk0 to readonly

#### Expected behaviour
- Card is set to readonly without issue

#### Actual behaviour
- Error message: [Failed] Could not apply-
- /: mount point is busy.

#### Extra details
- JellyFin server url: https://jellyfin.org (its an Emby clone).

MichaIng, 2020-07-09:
Many thanks for your report. This is what we do, just to check of it has to do with `dietpi-drive_manager` itself:
```
mount -v -o ro,remount /
```
A busy mount point usually means that there is currently a file opened. This can be a database, a log file or something similar.

You could do the following to check for any R/W accessed file on the drive:
```
apt install lsof
lsof | mawk 'NR==1 || $4~/[0-9]+[uw]/ && $9~/^\// && $9!~/^\/(dev|run|tmp|sys|proc)/'
```

Otherwise, a reboot will solve. `dietpi-drive_manager` already altered `/etc/fstab`. To check:
```
grep '[[:blank:]]/[[:blank:]]' /etc/fstab
```
should show the `,ro` mount option.

Note that DietPi scripts do not handle R/O mounted rootfs very gracefully. You might face some errors on boot, but this can be ignored. We'll address this in the future.

Joulinar, 2020-07-09:
It's not 100% the answer you are looking for but maybe something similar. Probably you can use `overlayroot`. Seems to be an interesting alternative to keep your rootFS r/o and simulate a r/w file system for your applications due to overlay. I was playing with it and installation was quite easy. 

https://dietpi.com/phpbb/viewtopic.php?p=24920#p24920

kotx, 2020-07-09:
I installed `iotop` and it seems to be that `ext4lazyinit` almost constantly has a file open in r/w mode.

kotx, 2020-07-09:
I used killall on ext4lazyinit and stopped my samba & jellyfin services and the mount command worked. However, the drive manager complains: "RootFS is currently set to "Read Only (R/O)". DietPi-Drive_Manager requires "Read Write (R/W)" access to function." The drive is currently mounted in R/W mode.

kotx, 2020-07-09:
Changing fstab manually does work. The problem is DietPi-Drive_Manager not working though.

MichaIng, 2020-07-09:
As of the servers/services, it might be possible if you disable file logging for them and for Samba it might be required to configure it being R/O as well (so that clients see it being R/O gracefully). `DietPi-Drive_Manager` reasonably cannot work without rootfs being R/W as /etc/fstab cannot be altered to apply any change. So the schedule would be to first configuring all mounts as required and then remount rootfs R/O, which forces drive manager to exit.

Interesting: https://superuser.com/a/1306332/1183051
Never heard about or saw `ext4lazyinit` before but it seems to be a background service to finish new file system creations in background. So this must definitely not be killed to avoid fs corruption. I am wondering if this was really writing to rootfs or to an external drive that you formatted moreless recently?

Joulinar, 2020-07-09:
`overlayroot` will help to keep your services running even on r/o systems. Just follow the link I posted above.

kotx, 2020-07-09:
> Never heard about or saw `ext4lazyinit` before but it seems to be a background service to finish new file system creations in background. So this must definitely not be killed to avoid fs [...]

I am using an external hard drive from my old DietPi install (SD card became readonly), so I don't know if that has any correlation.

kotx, 2020-07-09:
> `overlayroot` will help to keep your services running even on r/o systems. Just follow the link I posted above. 

I'll check that out, thanks for the advice

kotx, 2020-07-09:
`overlayroot` works great! Thanks for pointing this out to me. I wish the DriveManager could do this automatically when attempting to make rootfs readonly as well, though. Should I open another issue for a feature request?

closed 2020-07-10 by kotx

MichaIng, 2020-07-10:
> Should I open another issue for a feature request?

Would be great. I remembered there was a required about this already which I'll link/merge then: https://dietpi.com/phpbb/viewtopic.php?f=12&t=5978

Joulinar, 2020-07-10:
FR up https://github.com/MichaIng/DietPi/issues/3663