MichaIng/DietPi #3663: Dietpi-Drive_Manager | Allow to mount RootFS r/o using overlay

Joulinar, 2020-07-10:
Enable `dietpi-drive_manager` to be able to mount `RootFS r/o` using overlay. This would allow to protect SD card from write access but keep applications able to use the overlay for temp write activities. Several system/software tasks require write access, otherwise print errors, refuse to start or such. As well DietPi itself does not work silently on `r/o RootFS`.

**Forum**
https://dietpi.com/phpbb/viewtopic.php?p=24920#p24920
https://dietpi.com/phpbb/viewtopic.php?f=12&t=5978

**Github**
https://github.com/MichaIng/DietPi/issues/3662

**Feathub**
https://feathub.com/MichaIng/DietPi/+87

springfielddatarecovery, 2021-07-07:
I will add a $40 bounty to any developer who implements this feature along with a $10 contribution to DietPi. Not much but maybe others will pitch in as well :). If DietPi themselves make the patch then they get the whole $50. The feature has to make it into an official DietPi release to qualify for this bounty, only one bounty will be awarded, if there are multiple bounty claims for some reason I will allocate at my discretion.

ub3dqy, 2026-11-15:
@Joulinar, I can help add a read-only root option with a temporary writable overlay to DietPi-Drive_Manager. Is that the scope you would still want in a PR against dev?

@springfielddatarecovery, is the posted $40 still available, and can you pay in BTC on the Bitcoin network after the feature ships in an official DietPi release?

Dima Mak / ub3dqy

springfielddatarecovery, 2026-11-16:
This bounty is no longer active