MichaIng/DietPi #3667: Haveged not working

chippey5, 2020-07-12:
#### Required Information
- DietPi version | 6.31 rc2 / master
- Distro version | buster
- Kernel version | 5.4.49-rockchip64
- SBC model | ROCKPro64 (aarch64)
- Power supply used | 5V 2A
- SDcard used | eMMC 
- Haveged version | 1.9.1-7

#### Steps to reproduce
1. `/usr/sbin/haveged --Foreground --verbose=1 --write 1024`

#### Expected behaviour

- Haveged should be creating entropy

#### Actual behaviour

```
$ /usr/sbin/haveged --Foreground --verbose=1 --write 1024
Illegal instruction
```

#### Extra details
The same behaviour consists through reboots and reinstallations. Perhaps the program is configured with the wrong parameters for compilation? I've looked around on the [haveged's github page](https://github.com/jirka-h/haveged) and in existing open/closed issues, but I can't make out what could be wrong.

MichaIng, 2020-07-12:
Many thanks for your report.

The pre-installed haveged is installed from official Debian repository, so I doubt that there is such a fundamental bug, however we never know 😉. Please assure:
1. Run it as root. I see `$`, so probably your user is simply not allowed to execute haveged, instead try: `sudo haveged --Foreground --verbose=1 --write 1024`
2. There is a service running by default that should keep the entropy pool at 1M already: `systemctl status haveged`
Probably one cannot run two instances concurrently, at least this would make sense to me.

chippey5, 2020-07-12:
Thanks for your reply.

Correct, I discovered this issue when I checked the systemd services from the beginning and noticed that the issue is not with the service itself, but with haveged. The service returns:

```
$ sudo systemctl status haveged
● haveged.service - Entropy daemon using the HAVEGE algorithm
   Loaded: loaded (/lib/systemd/system/haveged.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Sun 2020-07-12 17:55:48 BST; 2h 6min ago
     Docs: man:haveged(8)
           http://www.issihosts.com/haveged/
  Process: 1471 ExecStart=/usr/sbin/haveged --Foreground --verbose=1 $DAEMON_ARGS (code=killed, signal=ILL)
 Main PID: 1471 (code=killed, signal=ILL)
```

So I checked what the service contained and ran the command it contained `(/usr/sbin/haveged --Foreground --verbose=1 --write 1024)`.

Running the command as root did unfortunately result in the same:

```
# /usr/sbin/haveged --Foreground --verbose=1 --write 1024
Illegal instruction
```

I'm thinking that something needs to be changed in the `./configure` options stated under _The following build options are available to "./configure":_ on the [project's github page](https://github.com/jirka-h/haveged), but I'm not sure what...

MichaIng, 2020-07-12:
As said, this is the official Debian build which has not changed on Buster repo for more than one year: https://metadata.ftp-master.debian.org/changelogs//main/h/haveged/haveged_1.9.1-7_changelog

Please try to reinstall the package:
```
apt install --reinstall libhavege1 haveged
```
Also check for kernel errors, probably a file corruption:
```
dmesg -l emerg,alert,crit,err
```

I'll also see if I can run some test.

chippey5, 2020-07-12:
Well, this is awkward. The one dependency that was needed and that I didn't reinstall before was the culprit. The issue was with `libhavege1`, after reinstalling it, haveged now starts fine:

```
$ sudo systemctl status haveged
● haveged.service - Entropy daemon using the HAVEGE algorithm
   Loaded: loaded (/lib/systemd/system/haveged.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-07-12 20:39:21 BST; 15s ago
     Docs: man:haveged(8)
           http://www.issihosts.com/haveged/
 Main PID: 3122 (haveged)
    Tasks: 1 (limit: 2212)
   Memory: 3.2M
   CGroup: /system.slice/haveged.service
           └─3122 /usr/sbin/haveged --Foreground --verbose=1 -w 1024
```

Many thanks 😃

closed 2020-07-12 by chippey5