MichaIng/DietPi #3668: NFS Server installation on Odroid C2 hangs system

wubwubwubwubwubwubwubwub, 2020-07-12:
### Creating a bug report/issue

#### Required Information
- DietPi version | not sure, cannot boot
- Distro version | not sure, cannot boot
- Kernel version | system can't actually boot due to bug - all references in /boot are for 3.16.85+
- SBC model | Odroid C2
- Power supply used |  Duracell PRO235 2100mA
- SDcard used |  SanDisk Ultra 16 Gb

#### Additional Information (if applicable)
- Software title | nfs server
- Was the software title installed freshly or updated/migrated?  Fresh install
- Can this issue be replicated on a fresh installation of DietPi?  Yes.
  can't boot, so no access to dietpi-bugreport
- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...  On fresh install of a DietPi image, use dietpi-software to install NFS Server
2. ...

#### Expected behaviour
  When the initial configuration and software installation is complete, the system should reboot to a login prompt.
- ...

#### Actual behaviour

- ...On initial reboot, the system hangs with this message:  NFSD: starting 90-second grace period (net ffffffc001d90c80)
Waiting for multiple minutes (by my watch) does not result in any update to the screen.  [90 seconds seems a >very< long time!]

#### Extra details

- ...  I do not think this matters, but during the installation, I do not have any mass storage connected, except the SD card I am booting from.  I also tried to do a basic installation, configuring only my network adapter and locale settings, then after the initial reboot I tried to add NFS Server.  Hung in the same way.

MichaIng, 2020-07-13:
Many thanks for your report. I was just reviewing our install code and we do nothing else than installing the [nfs-kernel-server](https://packages.debian.org/buster/nfs-kernel-server) package and place an export that at least from syntax and functional test is pretty fine.

I found this grace period being something addressing clients that should not try to access too early on a reboot: https://access.redhat.com/solutions/42868
So if I understand correctly, it does not mean that the server/boot waits/hangs for something, but just tells you that clients will have to wait for 90 seconds from that point on before being able to connect. Boot on the other hand should not be interrupted at all.

But obviously the hang still has something to do with the NFS server, the kernel message before is:
```
Installing knfsd
```

I cannot find a hint that the Linux version should be a problem for buster nfs-utils: http://nfs.sourceforge.net/, http://linux-nfs.org/wiki/index.php/Main_Page
Especially because the Debian package is based on pretty outdated source v1.3.4 vs upstream v2.5.1.

I was searching around a bid and was not able to find any hint. For debugging I can only suggest:
1. Reboot a few times without NFS server installed, assuring that there is nothing else that might cause an issue.
2. Install the Stretch package to rule out an issue related to recent changes:
```
cd /tmp
wget http://deb.debian.org/debian/pool/main/n/nfs-utils/nfs-common_1.3.4-2.1_arm64.deb
wget http://deb.debian.org/debian/pool/main/n/nfs-utils/nfs-kernel-server_1.3.4-2.1_arm64.deb
apt install ./nfs-common_1.3.4-2.1_arm64.deb
apt install ./nfs-kernel-server_1.3.4-2.1_arm64.deb
rm nfs-common_1.3.4-2.1_arm64.deb nfs-kernel-server_1.3.4-2.1_arm64.deb
```
- If reboot still works, upgrade to the newer packages to prove issue with those:
```
apt update; apt upgrade
```
- If reboot still works, then its the export itself:
```
echo '/mnt/dietpi_userdata *(rw,async,no_root_squash,fsid=0,crossmnt,no_subtree_check)' > /etc/exports
```
- Boot message verbosity could be increased as well: In `/boot/boot.ini` line `setenv bootargs ...` add `ignore_loglevel` to the end, inside quotes and with space separated, like:
```
setenv bootargs "root=/dev/mmcblk0p2 rootwait ro ${condev} no_console_suspend hdmimode=${m} m_bpp=${m_bpp} vout=${vout} fsck.repair=yes net.ifnames=0 elevator=noop disablehpd=${hpd} ignore_loglevel"
```

wubwubwubwubwubwubwubwub, 2020-07-14:
Since I am curious about how things work, and I had found a workaround by doing a minimal DietPi installation, then using apt install to add nfs-kernel-server, I decided to try and compare the system that didn't boot to the one that did, to see if I could find out what the difference was.
It turns out that the system that could not boot did not have a symlink in /etc/systemd/system/multi-user.target.wants called nfs-kernel-server.service, but the one that booted did have such a link.
I added one that points to /lib/systemd/system/nfs-kernel.server.service.
After adding that symlink, the system booted cleanly.  I believe I have found the source of the problem.

MichaIng, 2020-07-14:
> After adding that symlink, the system booted cleanly. I believe I have found the source of the problem.

Wow, ähm das is extremely strange. All you did is enabling the NFS server service to be started by systemd a bid earlier while otherwise it is started a bid later by DietPi boot scripts. Actually it cannot be the reason as otherwise the boot would hang exactly the same when you disable the NFS server completely (e.g. mask it). Of course the NFS server cannot be required for booting and also the time it is started by systemd is after several DietPi boots scripts ran already, much later then during kernel message phase, that I thought of.

You know what, might it be possible that the late output of the service simply prints over the login prompt messages? So actually the login promt is there but it only not visible? In such case you need to hit return/enter once to get the username input prompt.

Quick test:
```
systemctl disable nfs-kernel-server # This removes the symlink
reboot
```
The re-add the symlink, hence have NFS started a bid earlier:
```
systemctl enable nfs-kernel-server
```

wubwubwubwubwubwubwubwub, 2020-07-14:
This rabbit hole may be deeper than I thought.
I followed your suggestion, and found that when I call systemctl disable nfs-kernel-server
the symlink under /etc/systemd/system/multi-user.target.wants that gets removed is:
nfs-server.
Sitting right next to it is nfs-kernel-server, with my earlier timestamp, and it is ignored by systemctl.
What is also funny is that since I am lazy, when I decided a symlink to nfs-server might help, I made both nfs-server and nfs-kernel-server.  When the system booted successfully, I removed nfs-server and rebooted to see if nfs-kernel-server was the "right" one.  It seems to work just fine.
On my workaround system, I do >not< see nfs-kernel-server symlinks, so I decided that one was part of DietPi's process.

Anyway, removing the /etc/systemd/system/multi-user.target.wants/nfs-kernel-server symlink and rebooting with nfs-server alone is just fine also.
I am not certain about the order of messages during boot for the workaround system, but on this boot, I saw an nfs-server message go by at about 12 seconds.  That is right after eth0 and right before the usb mass storage device messages.  Last boot message is at 15 seconds or so, as usual.

Does any of this make sense to you?

wubwubwubwubwubwubwubwub, 2020-07-14:
OK.  My deepest apologies for wasting any of your time.  You are exactly right - without either symlink, the system boots to what looks like an error, but hitting >enter< a couple of times shows a perfectly functional login prompt, and nfsd is running as it should be.  Oh well.  I have learned something, and I have acquired an even deeper respect for the maintainers of DietPi.
Thank you!

MichaIng, 2020-07-14:
No worries, another argument against starting everything at the end of boot. I often think to revert to usual systemd controlled service start-up at boot.

Btw, nfs-kernel-server.service is a symlink to nfs-server.service, hence enabling either the one or the other has the same effect.

MichaIng, 2020-07-19:
Okay, I mark this as closed. Feel free to reopen if required.

closed 2020-07-19 by MichaIng