MichaIng/DietPi #3669: WireGuard intermittent handshake succes

bbsixzz, 2020-07-12:
Hello,

I have a WireGuard setup that works intermittently over 4G (iPhone SE 2016 latest iOS) following this standard procedure https://www.vikash.nl/setup-wireguard-vpn-server/. 

The server runs on a Pi Zero W using WiFi, the router is a Netgear Orbi (RBR850) and the throughout is very good (both LAN/WAN) without any neighboring networks.

When the connection fails the log shows this:

[code]Handshake did not complete after 5 seconds, retrying (try 2)[/code]

I can see the client connecting to the WireGuard server each time in the router logs, but the handshake fails seemingly at random.

There are periods (say 10 min) when it connects with no issues, then there are other periods where the handshake fails and I give up only to have it work again 30 min later.

I move between 1 to 2 out of 5 4G bars but the connection is stable. 

I bought this Pi Zero as a stopgap solution until https://github.com/MichaIng/DietPi/issues/3285 could be implemented.

Please help if you have any ideas, cheers

Joulinar, 2020-07-12:
Hi,

many thanks for your report. Do you use IPv4 as well as IPv6? Or IPv4 only?

bbsixzz, 2020-07-13:
The router has IPv6 disabled.

This is my wg0.conf

```[Interface]
Address = 10.9.0.1/24
PrivateKey = XXX KXWs=
ListenPort = 51820

PreUp = /boot/dietpi/func/obtain_network_details
PostUp = sysctl net.ipv4.conf.%i.forwarding=1 net.ipv4.conf.$(mawk 'NR==3' /run/dietpi/.network).forwarding=1
PostUp = sysctl net.ipv6.conf.$(mawk 'NR==3' /run/dietpi/.network).accept_ra=2
PostUp = sysctl net.ipv6.conf.%i.forwarding=1 net.ipv6.conf.$(mawk 'NR==3' /run/dietpi/.network).forwarding=1
PostUp = iptables -A FORWARD -i %i -j ACCEPT; iptables -t nat -A POSTROUTING -o $(mawk 'NR==3' /run/dietpi/.network) -j MASQUERADE
PostUp = ip6tables -A FORWARD -i %i -j ACCEPT; ip6tables -t nat -A POSTROUTING -o $(mawk 'NR==3' /run/dietpi/.network) -j MASQUERADE
PostDown = iptables -D FORWARD -i %i -j ACCEPT; iptables -t nat -D POSTROUTING -o $(mawk 'NR==3' /run/dietpi/.network) -j MASQUERADE
PostDown = ip6tables -D FORWARD -i %i -j ACCEPT; ip6tables -t nat -D POSTROUTING -o $(mawk 'NR==3' /run/dietpi/.network) -j MASQUERADE

# Client 1
[Peer]
PublicKey = XXX 3+Xg=
AllowedIPs = 10z.9.0.2/32

# Client 2
[Peer]
PublicKey = XXX Q1QA=
AllowedIPs = 10.9.0.3/32

# Client 3
[Peer]
PublicKey = XXX mNAY=
AllowedIPs = 10.9.0.4/32
```

This is my wg0-client.conf:

```[Interface]
Address = 10.9.0.2/24
PrivateKey = XXX gvFk=

# Comment the following to preserve the clients default DNS server, or force a desired one.
DNS = 192.168.1.1

# Kill switch: Uncomment the following, if the client should stop any network traffic, when disconnected from the VPN server
# NB: This requires "iptables" to be installed, thus will not work on most mobile phones.
#PostUp = iptables -I OUTPUT ! -o %i -m mark ! --mark $(wg show %i fwmark) -m addrtype ! --dst-type LOCAL -j REJECT && ip6tables -I OUTPUT ! -o %i -m mark ! --mark $(wg show %i fwmark) -m addrtype ! --dst-type LOCAL -j REJECT
#PreDown = iptables -D OUTPUT ! -o %i -m mark ! --mark $(wg show %i fwmark) -m addrtype ! --dst-type LOCAL -j REJECT && ip6tables -D OUTPUT ! -o %i -m mark ! --mark $(wg show %i fwmark) -m addrtype ! --dst-type LOCAL -j REJECT

[Peer]
PublicKey = XXX 6hl4=
# Tunnel all network traffic through the VPN:
#       AllowedIPs = 0.0.0.0/0, ::/0
# Tunnel access to server-side local network only:
#       AllowedIPs = 192.168.1.0/24
# Tunnel access to VPN server only:
#       AllowedIPs = 192.168.1.3/32
AllowedIPs = 0.0.0.0/0, ::/0
Endpoint =XXX.mynetgear.com:51820

# Uncomment the following, if you're behind a NAT and want the connection to be kept alive.
#PersistentKeepalive = 25

Joulinar, 2020-07-13:
@bbsixzz 

do you have issues with all clients? Because, only `Client 1` has been connected since the last time Wireguard Server restarted.

The handshake issue, does it happen on client or server side?
Just for testing, can you change on client device the `Endpoint` from `xxx.mynetgear.com` to your external IP address?

bbsixzz, 2020-07-13:
Here's my second client connecting, it's a MacBook using the iPhone as a hotspot:

```
interface: wg0
  public key: XXX 6hl4=
  private key: (hidden)
  listening port: 51820

peer: XXX Q1QA=
  endpoint: XXX 155.39:38709
  allowed ips: 10.9.0.3/32
  latest handshake: 39 seconds ago
  transfer: 211.53 KiB received, 150.43 KiB sent

peer: XXX +Xg=
  endpoint: XXX 155.39:64976
  allowed ips: 10.9.0.2/32
  latest handshake: 1 minute, 43 seconds ago
  transfer: 6.90 MiB received, 30.09 MiB sent

peer: XXX mNAY=
  allowed ips: 10.9.0.4/32

As you can see they are using the same IP, both connect fine; for now.

I've tried changing the `Endpoint` to the external IP address with no success.

The handshake issue shows up in the iPhone WireGuard app log, I have no way of seeing the server side because I'm not on location to connect to it locally. I can see the connection being made in the router log because I can monitor it remotely, so whatever is wrong seems to be happening on the server and/or iOS. It's real tricky to troubleshoot.

I've also rebooted the router and Pi Zero multiple times.

Joulinar, 2020-07-13:
question would be, if your MacBook will have similar issue at same time as your iPhone. Maybe there is a general connection/ internet access issue to your home at this moment? Hard to say. Usually you would need to do package capturing on your router, to see if packages are arriving from external

bbsixzz, 2020-07-13:
Both MacBook and iPhone have the issue simultaneously after some testing.

I activated VPN server on the router which uses OVPN, it always works when WireGuard fails.

This leads me to believe there is either something wrong with WireGuard as a protocol, or DietPi.

Joulinar, 2020-07-14:
hmm I'm using WireGuard without issues and atm you are the only one reporting it. Can you do some package tracing on your router. Would be good to see how network packages will looks like once issue occurs. You should see something like this if all goes fine

[image]

bbsixzz, 2020-07-14:
Actually I have something better, I just realized while connected to the router OVPN that I couldn't access the Pi Zero when WireGuard is failing handshakes. 

When WireGuard works I can also access the machine using the router OVPN.

It seems for some reason the Wi-Fi on the Pi Zero is occasionally failing. 

Any tips on how to check some logs so I can share here next time I have access?

Joulinar, 2020-07-14:
yes indeed a flapping WiFi is a good reason for connection issues.

bbsixzz, 2020-07-14:
Well everything is rock solid now.

Why is auto reconnect disabled by default for the WiFi adapter? This was the culprit, shouldn't it be enabled by default?

On a tangent, are there logs on the Pi to spot the connection dropping? It would be great to have some actual proof.

MichaIng, 2020-07-14:
IMO its good to not have auto-reconnect enabled by default. Its good to know about unstable WiFi in the first place. Auto-reconnect checks connection every 10 seconds, so there are small time frames of failing requests which one can then associate. Also auto-reconnect causes issues when the WiFi is for local network connection only while the default gateway is on a different network adapter. We can reconsider when the planned network setup rework is done, which includes configuring adapters as main (internet) adapters or for other purpose (AP, LAN only, fallback internet adapter, ...) to enable auto-reconnect by default when a WiFi interface is setup as main internet interface.

bbsixzz, 2020-07-14:
Very well!

What about logs I can check for disconnects on the Pi?

MichaIng, 2020-07-14:
```
journalctl -u dietpi-wifi-monitor
```
If you use it, else without running some request log daemon not 100% sure, does wpasupplicant produce a log when the connection is lost? `journalctl -u wpa_supplicant@wlan0`

bbsixzz, 2020-07-14:
```
root@DietPi:~# journalctl -u dietpi-wifi-monitor
-- Logs begin at Sat 2020-07-11 16:15:39 BST, end at Tue 2020-07-14 15:09:03 BST. --
Jul 14 03:25:54 DietPi systemd[1]: Starting DietPi-WiFi-Monitor...
Jul 14 03:25:54 DietPi systemd[1]: Started DietPi-WiFi-Monitor.
Jul 14 03:25:54 DietPi dietpi-wifi-monitor.sh[4626]: Checking connection for: wlan0 via ping to default gateway every 10 seconds
Jul 14 03:26:55 DietPi dietpi-wifi-monitor.sh[4626]: Detected connection loss: wlan0. Reconnecting...
Jul 14 03:26:55 DietPi dietpi-wifi-monitor.sh[4626]: ifdown: waiting for lock on /run/network/ifstate.wlan0
Jul 14 03:27:05 DietPi dhclient[4941]: Killed old client process
Jul 14 03:27:05 DietPi dietpi-wifi-monitor.sh[4626]: Killed old client process
Jul 14 03:27:06 DietPi dhclient[4941]: Internet Systems Consortium DHCP Client 4.4.1
Jul 14 03:27:06 DietPi dietpi-wifi-monitor.sh[4626]: Internet Systems Consortium DHCP Client 4.4.1
Jul 14 03:27:06 DietPi dietpi-wifi-monitor.sh[4626]: Copyright 2004-2018 Internet Systems Consortium.
Jul 14 03:27:06 DietPi dietpi-wifi-monitor.sh[4626]: All rights reserved.
Jul 14 03:27:06 DietPi dietpi-wifi-monitor.sh[4626]: For info, please visit https://www.isc.org/software/dhcp/
Jul 14 03:27:06 DietPi dietpi-wifi-monitor.sh[4626]: Listening on LPF/wlan0/b8:27:eb:f6:99:9e
Jul 14 03:27:06 DietPi dietpi-wifi-monitor.sh[4626]: Sending on   LPF/wlan0/b8:27:eb:f6:99:9e
Jul 14 03:27:06 DietPi dietpi-wifi-monitor.sh[4626]: Sending on   Socket/fallback
Jul 14 03:27:06 DietPi dhclient[4941]: Copyright 2004-2018 Internet Systems Consortium.
Jul 14 03:27:06 DietPi dhclient[4941]: All rights reserved.
Jul 14 03:27:06 DietPi dhclient[4941]: For info, please visit https://www.isc.org/software/dhcp/
Jul 14 03:27:06 DietPi dhclient[4941]: 
Jul 14 03:27:06 DietPi dhclient[4941]: Listening on LPF/wlan0/b8:27:eb:f6:99:9e
Jul 14 03:27:06 DietPi dhclient[4941]: Sending on   LPF/wlan0/b8:27:eb:f6:99:9e
Jul 14 03:27:06 DietPi dhclient[4941]: Sending on   Socket/fallback
Jul 14 03:27:06 DietPi dhclient[4941]: DHCPRELEASE of 192.168.1.3 on wlan0 to 192.168.1.1 port 67
Jul 14 03:27:06 DietPi dietpi-wifi-monitor.sh[4626]: DHCPRELEASE of 192.168.1.3 on wlan0 to 192.168.1.1 port 67
Jul 14 03:27:08 DietPi wpa_supplicant[5003]: Successfully initialized wpa_supplicant
Jul 14 03:27:09 DietPi wpa_supplicant[5028]: wlan0: Trying to associate with SSID 'XXX'
Jul 14 03:27:10 DietPi dhclient[5045]: Internet Systems Consortium DHCP Client 4.4.1
Jul 14 03:27:10 DietPi dhclient[5045]: Copyright 2004-2018 Internet Systems Consortium.
Jul 14 03:27:10 DietPi dietpi-wifi-monitor.sh[4626]: Internet Systems Consortium DHCP Client 4.4.1
Jul 14 03:27:10 DietPi dietpi-wifi-monitor.sh[4626]: Copyright 2004-2018 Internet Systems Consortium.
Jul 14 03:27:10 DietPi dietpi-wifi-monitor.sh[4626]: All rights reserved.
Jul 14 03:27:10 DietPi dietpi-wifi-monitor.sh[4626]: For info, please visit https://www.isc.org/software/dhcp/
Jul 14 03:27:10 DietPi dhclient[5045]: All rights reserved.
Jul 14 03:27:10 DietPi dhclient[5045]: For info, please visit https://www.isc.org/software/dhcp/
Jul 14 03:27:10 DietPi dhclient[5045]: 
Jul 14 03:27:10 DietPi dhclient[5045]: Listening on LPF/wlan0/b8:27:eb:f6:99:9e
Jul 14 03:27:10 DietPi dietpi-wifi-monitor.sh[4626]: Listening on LPF/wlan0/b8:27:eb:f6:99:9e
Jul 14 03:27:10 DietPi dhclient[5045]: Sending on   LPF/wlan0/b8:27:eb:f6:99:9e
Jul 14 03:27:10 DietPi dietpi-wifi-monitor.sh[4626]: Sending on   LPF/wlan0/b8:27:eb:f6:99:9e
Jul 14 03:27:10 DietPi dietpi-wifi-monitor.sh[4626]: Sending on   Socket/fallback
Jul 14 03:27:10 DietPi dietpi-wifi-monitor.sh[4626]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 5
Jul 14 03:27:10 DietPi dhclient[5045]: Sending on   Socket/fallback
Jul 14 03:27:10 DietPi dhclient[5045]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 5
Jul 14 03:27:10 DietPi wpa_supplicant[5028]: wlan0: Associated with 3c:37:86:f4:99:18
Jul 14 03:27:10 DietPi wpa_supplicant[5028]: wlan0: CTRL-EVENT-CONNECTED - Connection to 3c:37:86:f4:99:18 completed [id=0 id_str=]
Jul 14 03:27:10 DietPi wpa_supplicant[5028]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Jul 14 03:27:10 DietPi wpa_supplicant[5028]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=DE
Jul 14 03:27:15 DietPi wpa_supplicant[5028]: wlan0: CTRL-EVENT-DISCONNECTED bssid=3c:37:86:f4:99:18 reason=1 locally_generated=1
Jul 14 03:27:15 DietPi wpa_supplicant[5028]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Jul 14 03:27:15 DietPi wpa_supplicant[5028]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=USER type=COUNTRY alpha2=SE
Jul 14 03:27:15 DietPi wpa_supplicant[5028]: wlan0: Trying to associate with SSID 'XXX'
Jul 14 03:27:15 DietPi dhclient[5045]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 6
Jul 14 03:27:15 DietPi dietpi-wifi-monitor.sh[4626]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 6
Jul 14 03:27:16 DietPi wpa_supplicant[5028]: wlan0: Associated with 3c:37:86:f4:c5:97
Jul 14 03:27:16 DietPi wpa_supplicant[5028]: wlan0: CTRL-EVENT-CONNECTED - Connection to 3c:37:86:f4:c5:97 completed [id=0 id_str=]
Jul 14 03:27:16 DietPi wpa_supplicant[5028]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Jul 14 03:27:16 DietPi wpa_supplicant[5028]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=DE
Jul 14 03:27:21 DietPi dhclient[5045]: No DHCPOFFERS received.
Jul 14 03:27:21 DietPi dietpi-wifi-monitor.sh[4626]: No DHCPOFFERS received.
[... cut ...]
Jul 14 14:29:15 DietPi dhclient[25731]: bound to 192.168.1.3 -- renewal in 42555 seconds.
Jul 14 14:29:15 DietPi dietpi-wifi-monitor.sh[4626]: bound to 192.168.1.3 -- renewal in 42555 seconds.
Jul 14 14:29:15 DietPi dietpi-wifi-monitor.sh[4626]: Completed
```

Is there something wrong with the country code? I tried setting it to SE but DietPi gave me an error.

Tried it again now and it works.

I will monitor this log and see if it disconnects again, at least now with auto reconnect I can reliably get back in.

Joulinar, 2020-07-14:
As you could switch to Static IP instead of DHCP. Maybe it could help to reduce time to start WiFi interface

bbsixzz, 2020-07-16:
Well the Pi still disconnects from WiFi occasionally but always reconnects so WireGuard is always accessible.

That's what matters, I have no intention to troubleshoot further to find out the underlying WiFi issue.

The access point is a Netgear Orbi WiFi 6 router and the Pi is located not far away; there are no obstructions.

Joulinar, 2020-07-16:
Ok can we consider to close the issue?

closed 2020-07-16 by bbsixzz

bbsixzz, 2020-07-19:
I would like to append this finding, I don't know how carefully you read through my log there but the issue is identical.

https://github.com/NixOS/nixpkgs/issues/40089

It seems this error code causes the WiFi to drop:

```
wpa_supplicant[5028]: nl80211: Failed to open /proc/sys/net/ipv4/conf/p2p-dev-wlan0/drop_unicast_in_l2_multicast: No such file or direc
tory
wpa_supplicant[5028]: nl80211: Failed to set IPv4 unicast in multicast filter
wpa_supplicant[5028]: nl80211: Failed to open /proc/sys/net/ipv4/conf/p2p-dev-wlan0/drop_unicast_in_l2_multicast: No such file or direc
tory
wpa_supplicant[5028]: nl80211: Failed to set IPv4 unicast in multicast filter
```

Maybe you want to have a look at that, even though there is a "workaround".

MichaIng, 2020-07-19:
Can you please paste:
```
systemctl status wpa_supplicant
systemctl cat wpa_supplicant
systemctl status wpa_supplicant@wlan0
systemctl cat wpa_supplicant@
```

bbsixzz, 2020-07-19:
```
root@DietPi:~# systemctl status wpa_supplicant
● wpa_supplicant.service - WPA supplicant
   Loaded: loaded (/lib/systemd/system/wpa_supplicant.service; disabled; vendor 
preset: enabled)
   Active: inactive (dead)
```

```
root@DietPi:~# systemctl cat wpa_supplicant
# /lib/systemd/system/wpa_supplicant.service
[Unit]
Description=WPA supplicant
Before=network.target
After=dbus.service
Wants=network.target

[Service]
Type=dbus
BusName=fi.w1.wpa_supplicant1
ExecStart=/sbin/wpa_supplicant -u -s -O /run/wpa_supplicant

[Install]
WantedBy=multi-user.target
Alias=dbus-fi.w1.wpa_supplicant1.service
```

```
root@DietPi:~# systemctl status wpa_supplicant@wlan0
● wpa_supplicant@wlan0.service - WPA supplicant daemon (interface-specific versi
on)
   Loaded: loaded (/lib/systemd/system/wpa_supplicant@.service; disabled; vendor
 preset: enabled)
   Active: inactive (dead)
```

```
root@DietPi:~# systemctl cat wpa_supplicant@
# /lib/systemd/system/wpa_supplicant@.service
[Unit]
Description=WPA supplicant daemon (interface-specific version)
Requires=sys-subsystem-net-devices-%i.device
After=sys-subsystem-net-devices-%i.device
Before=network.target
Wants=network.target

# NetworkManager users will probably want the dbus version instead.

[Service]
Type=simple
ExecStart=/sbin/wpa_supplicant -c/etc/wpa_supplicant/wpa_supplicant-%I.conf -Dnl
80211,wext -i%I

[Install]
Alias=multi-user.target.wants/wpa_supplicant@%i.service
```

MichaIng, 2020-07-19:
Okay no conflicting service active, versions look fine, ifupdown invokes wpa_supplicant differently.

Does this error and unstable WiFi only happen when WireGuard is enabled or as well if not?
And did you try to play with WiFi AP setting, enabling/disabling different modes? `/etc/wpa_supplicant/wpa_supplicant.conf` can be used to toggle WiFi n support, but also check in router/AP.

Best match I found: https://archlinuxarm.org/forum/viewtopic.php?t=12267&p=57231
There is was a firmware issue. You're on latest kernel, right? `apt update && apt full-upgrade`

bbsixzz, 2020-07-20:
DietPi v6.31.2 and I have run apt update && apt full-upgrade.

According to https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md

> Note that with the latest Buster Raspberry Pi OS release, you must ensure that the wpa_supplicant.conf file contains the following information at the top:
> ```
> ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
> update_config=1
> country=<Insert 2 letter ISO 3166-1 country code here>
> ```

Country is missing in DietPi config, I have set `country=SE` now to see if it makes a difference.

I can't test without WG because I only have remote access currently (4G), I think it's unlikely to cause the WiFi issue however.

On a tangent, I checked the country setting in `dietpi-config` and it had reverted from SE to DE.

I set it to SE again only to have it revert to DE.

Maybe it's not important and makes no difference?

[image]

P.S.

How do you mean I can toggle WiFi n support in `wpa_supplicant.conf`, isn't it on by default?

MichaIng, 2020-07-20:
> How do you mean I can toggle WiFi n support in wpa_supplicant.conf, isn't it on by default?

Ah lol sorry I had hostapd config in mind, `wpa_supplicant.conf` doesn't allow that. 
This is what our config head looks like:
```
# Grant all members of group "netdev" permissions to configure WiFi, e.g. via wpa_cli or wpa_gui
ctrl_interface=DIR=/run/wpa_supplicant GROUP=netdev
# Allow wpa_cli/wpa_gui to overwrite this config file
update_config=1
```
- Since `/var/run` is a symlink to `/run`, it's the same, but one should consequently use `/run` nowadays.
- We set the WiFi country code via `/etc/default/crda`, which applies it during early boot. You can check whether it's correct via: `iw reg get`

bbsixzz, 2020-07-20:
```
global
country DE: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5875 @ 80), (N/A, 13), (N/A)
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

MichaIng, 2020-07-21:
Hmm not sure, for debugging:
```
sed -i '/^exec/a\echo "Setting to \$REGDOMAIN" > /root/crda.log' /lib/crda/setregdomain
```
This script should be called on boot by `/lib/udev/rules.d/60-crda.rules`.

`wpasupplicant` can apply it as well, but it should be sufficient to do that once at boot. Interestingly the `country` setting is not even mentioned in the man pages 🤔: https://manpages.debian.org/wpa_supplicant.conf
________
And you can set it manually and check by times, e.g. WiFi reconnect, if it has changed: `iw reg set SE`

But
> Either way, DE or SE shouldn't really matter to keep the connection.

I agree with that.