MichaIng/DietPi #3670: Error while upgrading Diet-pi

illusive-c, 2020-07-13:
Hi, 

I tried to upgrade the dietpi. Looks like it's getting stuck while installing the new PHP version. 

#### Details:
- Date           | Mon Jul 13 12:00:26 +03 2020
- Bug report     | e4260fe0-f634-4dfa-a66b-18185b5be745
- DietPi version | v6.22.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | Linux PiServer 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | apt-get -qq --allow-change-held-packages install php7.2-intl
- Exit code      | 100
- Software title | DietPi-Patch
#### Steps to reproduce:
1. log in as roo
2. dietpi-upgrade
#### Expected behaviour:
Update
#### Actual behaviour:
Getting the error message
│ E: Unable to locate package php7.2-intl                                      │
│ E: Couldn't find any package by glob 'php7.2-intl'                           │
│ E: Couldn't find any package by regex 'php7.2-intl'
#### Extra details:
Installation is not modified. Just the default installation with OwnCloud and lighthttp

#### Additional logs:
```
E: Unable to locate package php7.2-intl
E: Couldn't find any package by glob 'php7.2-intl'
E: Couldn't find any package by regex 'php7.2-intl'
```

illusive-c, 2020-07-13:
Looks like old PHP installation was corrupted and was preventing the new installation somehow... I uninstalled the PHP from the dietpi-software tool and re-installed and getting PHP 7.2.32-1+0~20200710.46+debian9~1.gbp625eb5  from php-v. 

Unfortunately, this also didn't get the owncloud to work. I had to run "apt-get install php7.2-intl" separately and now all is working fine.

MichaIng, 2020-07-13:
Ah yes the PHP upgrade is messy. I'll adjust and simplify it. A recent change in how incremental patches are applied caused this issue where `php7.2-intl` is attempted to be installed before the required repository has been added.

Just assure that dietpi-update finally runs through including all incremental patches that are done after `php7.2-intl` install attempt.

illusive-c, 2020-07-13:
Not sure how the update logic works, so this might be a dumb idea but, maybe all versions of PHP should be deleted before attempting 7.2 deployment? 

After messing around a little with different versions, now all seems to be working fine except when I restart the Pi, it gets stuck in maintenance mode and I have to SSH and "occ maintenance:mode --off" **twice** before it starts working. I'm thinking this might be the result of the "dietpi-update finally" you've mentioned. Should I just run that command in shell?

MichaIng, 2020-07-13:
> Not sure how the update logic works, so this might be a dumb idea but, maybe all versions of PHP should be deleted before attempting 7.2 deployment?

The problem is that now all reinstalls are done bundled at the end of all incremental patches to reduce overhead (multiple dedicated dietpi-software calls) and assure that dietpi-software (which is called with newest code) finds the system with all possibly required incremental patches instead of half done half not. However of course this means that all patches must not rely on any reinstall, and in case of PHP7.0 upgrade to PHP7.2/7.3 when ownCloud and/or Nextcloud and/or Pydio are installed, they do require the PHP reinstall first, which places the repository form where php7.2/7.3-intl can be installed.

I'll now add required module installs and webserver config adjustments right into PHP (re)install code in dietpi-software, which also enables the possibility to uninstall and reinstall PHP, e.g. do have it upgraded to newest version, without breaking dependants like webservers or other PHP web applications. The incremental patches will then only mark the old PHP packages for autoremoval which is done after the reinstalls.

> except when I restart the Pi, it gets stuck in maintenance mode and I have to SSH and "occ maintenance:mode --off" twice before it starts working.

One way to verify, could you paste:
```
grep -E 'CONFIG_AUTO_DIETPI_UPDATES=|CONFIG_WIFI_COUNTRY_CODE=' /boot/dietpi.txt
```

Please also verify that PHP7.2 is the only installed PHP version:
```
dpkg --get-selections '*php*'
```

illusive-c, 2020-07-13:
Man as powerful as Linux package management is, it's hella confusing too...

I've run the commands you've sent. The first one doesn't do anything as far as i can tell. Just returns back.

The output for the second one is this:
`php-apcu                                        install
php-common                                      install
php-igbinary                                    install
php-redis                                       install
php7.2-cli                                      install
php7.2-common                                   install
php7.2-curl                                     install
php7.2-fpm                                      install
php7.2-gd                                       install
php7.2-intl                                     install
php7.2-json                                     install
php7.2-mbstring                                 install
php7.2-mysql                                    install
php7.2-opcache                                  install
php7.2-readline                                 install
php7.2-sqlite3                                  install
php7.2-xml                                      install
php7.2-zip                                      install
`

I think only 7.2 is present now.

MichaIng, 2020-07-13:
Okay jep looks good then. Not sure why `occ maintenance:mode --off` fails then. Does it throw an error message? Since the first attempt is printed to local console only, if you have no monitor to watch:
```
systemctl disable dietpi-postboot
reboot
# login via SSH
systemctl enable --now dietpi-postboot
```
This way you see the very first attempt of starting up PHP and disabling maintenance mode... Now that I'm thinking about it: Actually maintenance mode is not enabled on shutdown/reboot, hence disabling it should be skipped on boot 🤔.

illusive-c, 2020-07-13:
Just did another reboot and indeed it's not stuck in the maintenance mode now. I think all the problems I got with the upgrade is sorted now. Thank you for the fast response! Appreciate the work you guys do!

MichaIng, 2020-07-13:
Fixed: https://github.com/MichaIng/DietPi/commit/3b90fabc5b286f978a5cee52925c14e906f7598e
Changelog: https://github.com/MichaIng/DietPi/commit/e0ce994cc81d93d1ac23a0afeee04d45c72a2ae2

Great to hear that the maintenance mode issue as gone as well 😃.

Phil1988, 2020-07-24:
> Okay jep looks good then. Not sure why `occ maintenance:mode --off` fails then. 

Should be ```ncc maintenance:mode --off``` right ;)

MichaIng, 2020-07-24:
`occ maintenance:mode --off` for ownCloud
`ncc maintenance:mode --off` for Nextcloud

Phil1988, 2020-07-24:
Yeah sorry... I did overread the "owncloud" and actually didnt know that OwnCloud is supported as well...
But I should have known. DietPi just combines EVERYTHING :D

MichaIng, 2020-07-24:
Especially since ownCloud and Nextcloud are still similar enough to share 99% of install and config code (and we do MUCH on those), it would be a waste of quality code to use it only for one option 😄. I remember sharing the same steps for both was one of my earliest contributions to DietPi after improving the Nextcloud install: https://github.com/MichaIng/DietPi/pull/1251

closed 2020-08-31 by MichaIng