MichaIng/DietPi #3671: ERROR: No internet connection while installing node-red

nwfall, 2020-07-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=31
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
buster

- Kernel version | `uname -a`
Linux DietPi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 2 Model B (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
12VAC Transformer -> Rectifier -> 2x4700uF caps -> 5A DC-DC

- SDcard used | (EG: SanDisk ultra)
AData 16GB uSD for boot
SATA HDD 2.5 on X820 storage board for rootfs

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
nodered

- Was the software title installed freshly or updated/migrated?
first time install

- Can this issue be replicated on a fresh installation of DietPi?
I think so. The system is quite fresh.

#### Steps to reproduce
1. sudo dietpi-software
2. install Node-Red from Software Optimised

#### Expected behaviour
- successful installation

#### Actual behaviour
- install script ./node-install.sh interrupted with ERROR No Internet Connection

#### Extra details
I entered subshell and changed all "curl http://..." to "curl https://..." in ./node-install.sh file
After that Installer completed successfully

Joulinar, 2020-07-14:
Hi,

I guess it's already known https://github.com/MichaIng/DietPi/pull/3651

The external install script that is used seems to broken. Therefore we will switch to our own fork

nwfall, 2020-07-14:
Oh, thank you for clarifying this. Shame on me I was unable to find existing issue

closed 2020-07-14 by nwfall