MichaIng/DietPi #3673: update error

gattohub, 2020-07-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
`
- Distro version | `buster
- Kernel version | `Linux DietPi 5.4.49-rockchip64 #20.05.7 SMP PREEMPT Sun Jun 28 18:17:52 UTC 2020 aarch64 GNU/Linux
`
- SBC model | `ROCK Pi 4 (aarch64)
` or (EG: RPi3)
- Power supply used  original rock pi power supply
- SDcard used | (EG: SanDisk ultra) nvme0n1 vendor: HP model: SSD EX900 250GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...I am trying to install MC, with the automatic Dietpi installer, and I get an error in update
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details
#### Details:
- Date           | Tue Jul 14 00:18:42 -03 2020
- DietPi version | v6.31.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | ROCK Pi 4 (aarch64) (ID=72)
- Kernel version | Linux DietPi 5.4.49-rockchip64 #20.05.7 SMP PREEMPT Sun Jun 28 18:17:52 UTC 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian buster InRelease
Get:2 https://deb.debian.org/debian buster-updates InRelease [51.9 kB]
Get:3 https://deb.debian.org/debian-security buster/updates InRelease [65.4 kB]
Get:5 https://deb.debian.org/debian buster-backports InRelease [46.7 kB]
Hit:4 https://apt.armbian.com buster InRelease
Reading package lists...
E: Release file for https://deb.debian.org/debian/dists/buster-updates/InRelease is not valid yet (invalid for another 17h 6min 37s). Updates for this repository will not be applied.
E: Release file for https://deb.debian.org/debian-security/dists/buster/updates/InRelease is not valid yet (invalid for another 43min 27s). Updates for this repository will not be applied.
E: Release file for https://deb.debian.org/debian/dists/buster-backports/InRelease is not valid yet (invalid for another 17h 6min 36s). Updates for this repository will not be applied.

- ...

Joulinar, 2020-07-15:
Many thanks for your report.

It looks your system clock is wrong? Could you try to run: `/boot/dietpi/func/run_ntpd 1`
Aside of that please simply retry `apt update` in a few hours, it might be simply a temporary server issue.

MichaIng, 2020-08-29:
Marking as closed due to missing reply. Feel free to reopen if required.

closed 2020-08-29 by MichaIng