MichaIng/DietPi #3675: Pihole WebInterface Error Failed CORS: null

Phil1988, 2020-07-15:
#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=31
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
9.12

- Kernel version | `uname -a`
Linux DietPi 4.9.0-8-amd64 #1 SMP Debian 4.9.144-3.1 (2019-02-19) x86_64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Virtual Machine (x86_64)

#### Additional Information (if applicable)
I have several programs installed.
Might be an issue between pihole and nextcloud (based on apache)

#### Steps to reproduce
I just updated from 
DiePi v6.28.0 -> v6.31.2
and
Pi-hole Version v4.3.2 -> v5.0

I think the update of Pi-hole caused this issue.

When I do anything on the WebInterface and save, I get this error:
Failed CORS: null vs 192.168.1.77, 2a03:c22:c015:9d00:20c:29ff:fe52:c3ba, 192.168.1.77, pi.hole, localhost

A quick search looks like apache is not giving the right $server_origin back. 
But I am not a programmer :)

Joulinar, 2020-07-15:
Hi,

many thanks for your message. Hmm looks like PiHole did not support Appache2, at least according the statement of Dan. Maybe @MichaIng could have a look, if we could mitigate it from DietPi side 🤔

MichaIng, 2020-07-15:
I tested Apache2 after Pi-hole v5 release, to generally it works fine. Yes Pi-hole officially only supports their complete setup including Lighttpd install + own config + Pi-hole in webroot etc, so keeping an eye on other webservers compatibility is a task for us, respectively the part of the community which requires a different webserver setup.

I'm currently retesting.

Phil1988, 2020-07-15:
Let me know if I can do anything for you to help

Phil1988, 2020-07-15:
I am doing so with latest edge and chrome and will report back

MichaIng, 2020-07-15:
Did you install via dietpi-software? As there is is skipped. Otherwise you would have two conflicting webservers installed.

MichaIng, 2020-07-15:
Can you instead check this:
```
echo "<?php echo '<pre>'.print_r(\$_SERVER, true).'</pre>'; ?>" > /var/www/test.php
```
Then open `your.domain.org/test.php` in browser and see if HTTP_ORIGIN is set.

MichaIng, 2020-07-15:
Jep that is expected. As said, two webservers cannot run concurrently on the same port. Hmm I guess now you really have to uninstall and reinstall as the Pi-hole installed, with webserver selected will alter webroot content etc.

Phil1988, 2020-07-15:
No problem.
I will uninstall and reinstall via dietpi-software and feedback :)

MichaIng, 2020-07-15:
> not sure what to pick, but I will jsut uninstall all depencies for now:

NOPE not do that!

Phil1988, 2020-07-15:
yeah... as soon as it asked me to uninstall cron I realizied its not "that good" :D 

THe good point is... I have a snapshot and will start again from there :)

MichaIng, 2020-07-15:
> THe good point is... I have a snapshot and will start again from there :)

Jep, keep and update that for any such things 👍.
_________
But a good reminder that this uninstall point still needs action. At least instead of purging those packages, they could instead only being marked as "auto" installed, which allows `apt-get autoremove` to remove only those which are not required for anything else. On the other hand "sudo", "wget", "unzip", "whiptail", "ping", you want to keep that in any case... Probably this step should be completely skipped. Someone who knows what all this is will be able to uninstall manually non-required leftovers, someone who does not know, should not remove then anyway. Good about DietPi is that e.g. webserver, PHP, SQLite are separate software IDs which can be uninstalled separately as well.

Phil1988, 2020-07-15:
uninstall worked flawless (i just hit "no" to the depencies).
Installation gets to this:
[image]

MichaIng, 2020-07-15:
Jep this is because "ping" was uninstalled. Please try the following:
```
apt install iputils-ping iproute2 curl cron sudo wget
```

Phil1988, 2020-07-15:
well... 

I hit exit and got this:
[image]

installing ping etc. gave me this 
[image]

Trying to start dietpi-software again gives me the error with the ping.... one.one.one again

MichaIng, 2020-07-15:
Ah I guess because of Pi-hole uninstall the DNS nameserver entry was left invalid. Please try:
```
echo 'nameserver 1.1.1.1' > /etc/resolv.conf
```
__________
Next test:
```
cat << _EOF_ > /var/www/post.php
<?php
\$url = 'http://localhost/test.php';
\$data = array('some' => 'content');

\$options = array(
    'http' => array(
        'header'  => "Content-type: application/x-www-form-urlencoded\r\n",
        'method'  => 'POST',
        'content' => http_build_query(\$data)));

\$response = file_get_contents(\$url, false, stream_context_create(\$options));

echo \$response;
?>
_EOF_
```
Then open `your.domain.org/post.php` in browser. This does a POST request, hence the origin is the webserver itself.

Phil1988, 2020-07-15:
I will do this soon... I just rebooted the system and was able to start the installation of pihole

MichaIng, 2020-07-15:
Probably I should do those tests on Stretch to assure exact same environment 🤔.

Phil1988, 2020-07-15:
going to have dinner now and will report back when finished (eating ...) 

I just got this error/message while installing:
[image]

it said installation is completed anyway and rebooted

MichaIng, 2020-07-15:
If you selected "Ignore" then the install is left unfinished. If you can see the last output of the installer, that would be interesting. Else simply retry as dietpi-software has it not marked as installed. Will be quicker since all dependencies are installed already.
_________
Further research about that CORS thing: https://subscription.packtpub.com/book/web_development/9781784393779/1/ch01lvl1sec09/how-cors-works-the-header-and-the-request
> Techniques have been proposed to first match $_SERVER['HTTP_ORIGIN'] to an allowed list, then write the header that allows the matched origin. Since $_SERVER['HTTP_ORIGIN'] is not reliable, or the requesting domain may be served via a CDN that does not match the expected domain, this technique may not work.

Pi-hole exactly follows this approach, checking this variable first, then setting the header accordingly. However the article underlines that HTTP_ORIGIN is not reliable.

Phil1988, 2020-07-15:
I did 
`echo 'nameserver 1.1.1.1' > /etc/resolv.conf`

and then
`cat << _EOF_ > /var/www/post.php
<?php
\$url = 'http://localhost/test.php';
\$data = array('some' => 'content');

\$options = array(
    'http' => array(
        'header'  => "Content-type: application/x-www-form-urlencoded\r\n",
        'method'  => 'POST',
        'content' => http_build_query(\$data)));

\$response = file_get_contents(\$url, false, stream_context_create(\$options));

echo \$response;
?>
_EOF_`

Going to URL/post.php shows an empty page

MichaIng, 2020-07-15:
`echo "<?php echo '<pre>'.print_r(\$_SERVER, true).'</pre>'; ?>" > /var/www/test.php` is still required, I guess that is missing due to recovered snapshot.

Phil1988, 2020-07-15:
you are right :) 

the test.php was missing.

Now the post.php gives this:

> Array
(
    [SCRIPT_URL] => /test.php
    [SCRIPT_URI] => http://localhost/test.php
    [HTTP_HOST] => localhost
    [HTTP_CONNECTION] => close
    [CONTENT_LENGTH] => 12
    [CONTENT_TYPE] => application/x-www-form-urlencoded
    [PATH] => /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
    [SERVER_SIGNATURE] => 
Apache/2.4.25 (Debian) Server at localhost Port 80

    [SERVER_SOFTWARE] => Apache/2.4.25 (Debian)
    [SERVER_NAME] => localhost
    [SERVER_ADDR] => ::1
    [SERVER_PORT] => 80
    [REMOTE_ADDR] => ::1
    [DOCUMENT_ROOT] => /var/www
    [REQUEST_SCHEME] => http
    [CONTEXT_PREFIX] => 
    [CONTEXT_DOCUMENT_ROOT] => /var/www
    [SERVER_ADMIN] => webmaster@localhost
    [SCRIPT_FILENAME] => /var/www/test.php
    [REMOTE_PORT] => 53496
    [GATEWAY_INTERFACE] => CGI/1.1
    [SERVER_PROTOCOL] => HTTP/1.0
    [REQUEST_METHOD] => POST
    [QUERY_STRING] => 
    [REQUEST_URI] => /test.php
    [SCRIPT_NAME] => /test.php
    [PHP_SELF] => /test.php
    [REQUEST_TIME_FLOAT] => 1594834475.345
    [REQUEST_TIME] => 1594834475
)

Phil1988, 2020-07-15:
Im sure that a fresh install would solve all problems, but setting up and configurre all the programs/servers I am using is a bit of work...

so for the current state it looks that the installation of pihole is not possible because /etc/pihole/setupVars.conf is missing - which might have something to do with dnsmasq  (what I often found but cant understand )

MichaIng, 2020-07-15:
> Looks like MichaIng is happy shopping or enjoying his well deserved rest.

Happy walking through the forest after rain stopped, slippy mud down the hill and loaded with food for the next days up the hill, so not the classic shopping joy, but good to get the required sport when sitting all the day 😄.

I also tested successfully on Stretch and as well the two test scripts show no HTTP_ORIGIN variable. Tomorrow I'll add some debug code to the Pi-hole script to check it there explicitly, as I'm still not sure if this variable can somehow be requested or is created/added only in certain circumstances.

Phil1988, 2020-07-16:
Haha! sounds like a healthy challenge ^^

If I can do any testings or verifying, just let me know.

I did a snapshot of the "after uninstall pihole and unable to reinstall because of missing /etc/pihole/setupVars.conf"-state :D 

Now I am back to the state before I updated.
I am reallly curious what causes this issue and broke the WebInterface saving ability.

Sadly I had a full disk on my VM because nextcloud did not properly deleted the garbage collected.
Thats where I had to delete all the snapshots made to beeing able to backup and mirror the whole disk.
So I can not track it as I would normally.

I will randomly check here today whenever I have time.

Phil1988, 2020-07-16:
I am home now, so if I need to test something for you, let me know.
I am willing to follow your commands sir :)

MichaIng, 2020-07-16:
Ah sorry, was consumed by some website/meta tasks today. I'll do the additional testing tomorrow to see which results/commands are interesting/relevant.

Phil1988, 2020-07-17:
No problem. Just wanted to inform you, that testing would have been possible on my side.
Ill be gone over the weekend. 
So maybe on sunday or monday :)

MichaIng, 2020-07-18:
I added code to the Pi-hole script and whatever I do, `$_SERVER['HTTP_ORIGIN']` is not set in my case, neither with Opera desktop, nor with Edge (non-Chromium) nor with Opera mobile.

Found further source that the existence cannot be trusted: https://stackoverflow.com/questions/41231116/serverhttp-origin-doesnt-work
However why the hack it IS set in your case to `null` 🤔.

MichaIng, 2020-07-18:
@Phil1988 
Please check your Apache2 config for `Header set Referrer-Policy "no-referrer"` oder `Header always set Referrer-Policy "no-referrer"`, which makes Apache2 setting HTTP_ORIGIN to "null" for privacy reasons. But it breaks Pi-hole.

Currently verifying with a fresh Pi-hole install, only with their install (+Lighttpd) setting the same policy there.

MichaIng, 2020-07-18:
Found an elegant solution: https://github.com/MichaIng/DietPi/commit/f2b8d2a47eaa41078df22026bda46d7c6509f058

Phil1988, 2020-07-20:
Hey MichaIng! 
I will check the `Header set Referrer-Policy "no-referrer` thing, when I am back home--- like in 2-3 hours :)

MichaIng, 2020-07-20:
Yes edit `/etc/apache2/sites-available/dietpi-pihole.conf` and add the two lines just as in this commit: https://github.com/MichaIng/DietPi/commit/f2b8d2a47eaa41078df22026bda46d7c6509f058
Then reload Apache: `systemctl reload apache2`
That should be it.

Phil1988, 2020-07-20:
I went back to the according snapshot (DietPi v6.31.2, Pihole v5) and can confirm that the commit #f2b8d2a is working flawless!

Good job again MichaIng!!

closed 2020-07-20 by Phil1988