MichaIng/DietPi #3676: Orange Pi One Plus

mrbluecoat, 2020-07-15:
### Creating an image request

#### Formal device information
- Device name | Orange Pi One Plus
- Official product URL | http://www.orangepi.org/OrangePiOneplus/
- Image download URL | https://www.armbian.com/orange-pi-one-plus/

#### Is the SBC officially supported by the Debian installer?

No

#### If not, is a reliable 3rd party Debian image available for this SBC?

Yes (see link above)

#### Vote for this image on FeatHub: https://feathub.com/MichaIng/DietPi/

mrbluecoat, 2020-07-15:
Here's a quick proof of concept (no `DietPi-Imager`): https://drive.google.com/file/d/1jZYNSMlTOvVXy_tSY0IYgfrwydPjma0f/view?usp=sharing

Only issues I encountered:
```
rmdir: failed to remove '/var/run/dbus': Directory not empty
...
mount -o remount /tmp
```

The second command I had to replace with a benign command (`ls /tmp`)

```
CPU Performance : Duration = 10.48 seconds (lower is faster)
CPU Temp        : Idle = 46'c | Full load = 73'c
RootFS          : Write = 20 MiB/s | Read = 21 MiB/s
RAM             : Write = 511 MiB/s | Read = 901 MiB/s

# sdcard
Test size = 100 MiB
WRITE     = 20 MiB/s
READ      = 21 MiB/s
```
Initial impressions: decent performance but runs hot in the clear acrylic case its sold with.

ricardoandren, 2020-07-15:
Yeah, I think the orange pi community (armbian) has done a great job I think dietpi could support all boards now, their latest images are pretty good (h3 and h5)

MichaIng, 2020-07-15:
Many thanks for your request.

There is indeed no problem with OrangePi, same as Banana Pi boards, the generic Armbian images work on them the same as on many other SBCs we support. Its more that opening those box(es) means a potential massive number of additional SBCs and I'm currently from workload on not able to create on maintain those. For this reason currently from my side I'd not work on them, however the preparation script should work fine: https://github.com/MichaIng/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh

> rmdir: failed to remove '/var/run/dbus': Directory not empty
> ...
> mount -o remount /tmp

Where did you face them (as those commands are not part of DietPi-PREP script)?

When using DietPi-Imager, use the one from dev branch: https://github.com/MichaIng/DietPi/blob/dev/.meta/dietpi-imager
I broke it shortly before v6.31 release 😅.

I plan do add "Generic RK3399", "Generic H3", "Generic H5" and such as hardware IDs, selectable via DietPi-PREP. This will allow to create images as well for new/unsupported devices, if the SoC is not a very rare/special one. Since we also check the SoC do derive e.g. required GPU driver and enable/disable install and config options, this will allow better support compared to choosing "Generic device".

And of course, if someone wants to contribute by creating/testing/updating an image for a certain SBC, this is highly welcome.

mrbluecoat, 2020-07-15:
> Where did you face them (as those commands are not part of DietPi-PREP script)?

I literally imaged the vanilla Armbian Buster using Etcher, plugged it into the Orange Pi One Plus and ran the commands on https://github.com/MichaIng/DietPi/issues/1285#issue-280771944   I don't think it's an issue, just mentioned it in for completeness.

> I broke it shortly before v6.31 release

Ah, that's why it wasn't working  :smile:  I thought maybe my tech skills were getting rusty...

The process was very easy so I imagine anyone can do what I did in a few minutes to get DietPi working on the Orange Pi One Plus so feel free to close this ticket.

mrbluecoat, 2020-07-25:
Here's a quick proof of concept using `DietPi-Imager` and nightly Armbian image:
https://drive.google.com/file/d/1iBWKeDXRvIbCYtS20aYbH-nN8Z9E-ejl/view?usp=sharing

MichaIng, 2021-03-12:
A generic Allwinner H6 model is added via: #4193
While it's no full Orange Pi One Plus support, at least GPU acceleration and other SoC-specific tweaks can be applied on a generic way.

closed 2021-03-12 by mrbluecoat