MichaIng/DietPi #3677: Error deitpi-software last update

gael1980, 2020-07-16:
### Creating a bug report/issue

#### Required Information
- DietPi version : 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=31
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version : Buster
- Kernel version : Linux DietPi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux
- SBC model : RPi 3 Model B (armv7l)
- SDcard used : SanDisk 16 Go

#### Additional Information (if applicable)
- Software title : Dietpi-software
- Was the software title updated to last version of dietpi

- Bug report ID : 8f31027a-ac32-4067-ba7e-45fc26a5a55a

#### Expected behaviour
since the last update, I can't launch dietpi anymore.
root@DietPi:~# dietpi-software
/boot/dietpi/dietpi-software: ligne 6594: erreur de syntaxe : fin de fichier prématurée

Joulinar, 2020-07-16:
Hi,

Many thanks for your report. Should be similar to this. #3652 

A fix should be provided by MOTD.

gael1980, 2020-07-16:
Yeah, it's the same problem.
the solution works perfectly
Thank you for your help

closed 2020-07-16 by gael1980