MichaIng/DietPi #3679: Still can't up date  to new version of Diet pi.... info below was never resolved from the last time, just lived with it...

blindtrust, 2020-07-16:
#### Details:
- Date           | Thu Jul 16 16:00:21 BST 2020
- Bug report     | e228a8e5-9f96-49b6-a40f-e420fca823ab
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -qq upgrade
- Exit code      | 139
- Software title | DietPi-Update
#### Steps to reproduce:

1. just run dietpi-update
2. ...
#### Expected behaviour:

- ... it should update.

#### Actual behaviour:

- ...errors out with t is command.. 

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait... 
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait... 
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -qq upgrade
[ INFO ] DietPi-Update | APT upgrade, please wait... 
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -qq upgrade
[ INFO ] DietPi-Bugreport | Packing upload archive, please wait...
[ INFO ] DietPi-Bugreport | Checking URL: ssh.dietpi.com, please wait...  (1/3)
URL transformed to HTTPS due to an HSTS policy
Spider mode enabled. Check if remote file exists.
--2020-07-16 16:00:38--  https://ssh.dietpi.com/
Resolving ssh.dietpi.com (ssh.dietpi.com)... 185.101.93.93
Connecting to ssh.dietpi.com (ssh.dietpi.com)|185.101.93.93|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 125004 (122K) [text/html]
Remote file exists and could contain further links,
but recursion is disabled -- not retrieving.

[  OK  ] DietPi-Bugreport | Checking URL: ssh.dietpi.com
[  OK  ] DietPi-Bugreport | Bug report sent, reference code: e228a8e5-9f96-49b6-a40f-e420fca823ab

#### Extra details:
two screen shots attaced show what the GUI shows.
- ...
#### Additional logs:
```
<img width="1197" alt="Screen Shot 2020-07-16 at 11 00 28 AM" src="https://user-images.githubusercontent.com/14252048/87687990-6726d580-c754-11ea-88d4-608d4fb073a2.png">

```

Joulinar, 2020-07-16:
Hi,

Many thanks for your report. Pls can you run following on command line. 

```
apt update 
apt upgrade
```

blindtrust, 2020-07-16:
the dreaded Segmentation fault after running the two commands... it seems to almost work...

root@DietPi:~# apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
67 packages can be upgraded. Run 'apt list --upgradable' to see them.
root@DietPi:~# apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  base-files bind9-host ca-certificates curl dnsutils e2fsprogs firmware-atheros firmware-brcm80211 firmware-iwlwifi firmware-misc-nonfree firmware-realtek
  git git-man idn2 iputils-ping libapt-inst2.0 libapt-pkg5.0 libbind9-161 libcom-err2 libcurl3-gnutls libcurl4 libdns-export1104 libdns1104 libext2fs2
  libgnutls30 libicu63 libidn2-0 libirs161 libisc-export1100 libisc1100 libisccc161 libisccfg163 libldap-2.4-2 libldap-common liblwres161 libpam-modules
  libpam-modules-bin libpam-runtime libpam0g libraspberrypi-bin libraspberrypi0 libss2 libssl1.1 libsystemd0 libtiff5 libudev1 openssl php7.3-cli
  php7.3-common php7.3-curl php7.3-fpm php7.3-gd php7.3-json php7.3-mbstring php7.3-opcache php7.3-readline php7.3-sqlite3 php7.3-xml php7.3-zip
  raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods sudo systemd systemd-sysv tzdata udev
67 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/131 MB of archives.
After this operation, 2,100 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
**Segmentation fault**
root@DietPi:~#

Joulinar, 2020-07-16:
Do you have sufficient space left on your device?

blindtrust, 2020-07-16:
should have plenty...

Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       29987824 2945920  26105068  11% /
devtmpfs          494908       0    494908   0% /dev
tmpfs             499516    2704    496812   1% /dev/shm
tmpfs             499516   56460    443056  12% /run
tmpfs               5120       0      5120   0% /run/lock
tmpfs             499516       0    499516   0% /sys/fs/cgroup
tmpfs            1047552    7028   1040524   1% /tmp
tmpfs              51200    1488     49712   3% /var/log
/dev/mmcblk0p1    258096   55337    202759  22% /boot
root@DietPi:~#

blindtrust, 2020-07-16:
root@DietPi:~# df -BM
Filesystem     1M-blocks  Used Available Use% Mounted on
/dev/root         29285M 2877M    25494M  11% /
devtmpfs            484M    0M      484M   0% /dev
tmpfs               488M    3M      486M   1% /dev/shm
tmpfs               488M   56M      433M  12% /run
tmpfs                 5M    0M        5M   0% /run/lock
tmpfs               488M    0M      488M   0% /sys/fs/cgroup
tmpfs              1023M    7M     1017M   1% /tmp
tmpfs                50M    2M       49M   3% /var/log
/dev/mmcblk0p1      253M   55M      199M  22% /boot

MichaIng, 2020-07-16:
Please check for file system errors:
```
dmesg -l emerg,alert,crit,err
```
If none, try to reinstall only APT first:
```
apt-get -y install --reinstall apt
apt-get -y upgrade
```

blindtrust, 2020-07-16:
root@DietPi:~# dmesg -l emerg,alert,crit,err
[    0.890128] vc_vchi_sm_init: failed to open VCHI service (-1)
[    0.890136] [vc_sm_connected_init]: failed to initialize shared memory service
[   31.671552] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[   31.671574] [vc_sm_connected_init]: failed to initialize shared memory service

did not run the other commands yet want to see what you say! thank you .

MichaIng, 2020-07-16:
These are expected, all fine. Please try the other two.

blindtrust, 2020-07-16:
decidedroot@DietPi:~# apt-get -y upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  base-files bind9-host ca-certificates curl dnsutils e2fsprogs firmware-atheros firmware-brcm80211 firmware-iwlwifi firmware-misc-nonfree firmware-realtek
  git git-man idn2 iputils-ping libapt-inst2.0 libapt-pkg5.0 libbind9-161 libcom-err2 libcurl3-gnutls libcurl4 libdns-export1104 libdns1104 libext2fs2
  libgnutls30 libicu63 libidn2-0 libirs161 libisc-export1100 libisc1100 libisccc161 libisccfg163 libldap-2.4-2 libldap-common liblwres161 libpam-modules
  libpam-modules-bin libpam-runtime libpam0g libraspberrypi-bin libraspberrypi0 libss2 libssl1.1 libsystemd0 libtiff5 libudev1 openssl php7.3-cli
  php7.3-common php7.3-curl php7.3-fpm php7.3-gd php7.3-json php7.3-mbstring php7.3-opcache php7.3-readline php7.3-sqlite3 php7.3-xml php7.3-zip
  raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods sudo systemd systemd-sysv tzdata udev
67 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/131 MB of archives.
After this operation, 2,100 kB of additional disk space will be used.
Segmentation fault
root@DietPi:~#  to run the commands.. here are the results...

**root@DietPi:~# apt-get -y install --reinstall apt**
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 67 not upgraded.
Need to get 1,349 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf apt armhf 1.8.2.1 [1,349 kB]
Fetched 1,349 kB in 3s (457 kB/s)
Segmentation fault
root@DietPi:~#

MichaIng, 2020-07-17:
Hmm, can you try to reinstall with case dpkg:
```
cd /tmp
wget http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/apt_1.8.2.1_armhf.deb
dpkg -i apt_1.8.2.1_armhf.deb
```
If this fails, the not apt but dpkg is the issue. Then we are still able to replace the binaries from the package manually.

blindtrust, 2020-07-17:
**root@DietPi:/tmp# wget http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/apt_1.8.2.1_armhf.deb**
--2020-07-17 12:58:03--  http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/apt_1.8.2.1_armhf.deb
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... 93.93.128.193, 2a00:1098:0:80:1000:75:0:3
Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: http://mirror.sjc02.svwh.net/raspbian/raspbian/pool/main/a/apt/apt_1.8.2.1_armhf.deb [following]
--2020-07-17 12:58:04--  http://mirror.sjc02.svwh.net/raspbian/raspbian/pool/main/a/apt/apt_1.8.2.1_armhf.deb
Resolving mirror.sjc02.svwh.net (mirror.sjc02.svwh.net)... 72.5.72.15, 2606:c680:0:b:3830:34ff:fe66:6663
Connecting to mirror.sjc02.svwh.net (mirror.sjc02.svwh.net)|72.5.72.15|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1349032 (1.3M) [application/octet-stream]
Saving to: ‘apt_1.8.2.1_armhf.deb’

apt_1.8.2.1_armhf.deb                   100%[===============================================================================>]   1.29M  78.4KB/s    in 17s     

2020-07-17 12:59:06 (76.8 KB/s) - ‘apt_1.8.2.1_armhf.deb’ saved [1349032/1349032]

root@DietPi:/tmp# dpkg -i apt_1.8.2.1_armhf.deb
(Reading database ... 25095 files and directories currently installed.)
Preparing to unpack apt_1.8.2.1_armhf.deb ...
Unpacking apt (1.8.2.1) over (1.8.2.1) ...
Setting up apt (1.8.2.1) ...
apt-daily-upgrade.timer is a disabled or a static unit not running, not starting it.
apt-daily.timer is a disabled or a static unit not running, not starting it.
Processing triggers for libc-bin (2.28-10+rpi1) ...

root@DietPi:/tmp# apt update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 https://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 https://archive.raspberrypi.org/debian buster/main armhf Packages [330 kB]
Fetched 13.4 MB in 12s (1,119 kB/s)                                                                                                                            
Reading package lists... Done
Building dependency tree       
Reading state information... Done
67 packages can be upgraded. Run 'apt list --upgradable' to see them.

root@DietPi:/tmp# apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  base-files bind9-host ca-certificates curl dnsutils e2fsprogs firmware-atheros firmware-brcm80211 firmware-iwlwifi firmware-misc-nonfree firmware-realtek
  git git-man idn2 iputils-ping libapt-inst2.0 libapt-pkg5.0 libbind9-161 libcom-err2 libcurl3-gnutls libcurl4 libdns-export1104 libdns1104 libext2fs2
  libgnutls30 libicu63 libidn2-0 libirs161 libisc-export1100 libisc1100 libisccc161 libisccfg163 libldap-2.4-2 libldap-common liblwres161 libpam-modules
  libpam-modules-bin libpam-runtime libpam0g libraspberrypi-bin libraspberrypi0 libss2 libssl1.1 libsystemd0 libtiff5 libudev1 openssl php7.3-cli
  php7.3-common php7.3-curl php7.3-fpm php7.3-gd php7.3-json php7.3-mbstring php7.3-opcache php7.3-readline php7.3-sqlite3 php7.3-xml php7.3-zip
  raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods sudo systemd systemd-sysv tzdata udev
67 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/131 MB of archives.
After this operation, 2,100 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
**Segmentation fault**
root@DietPi:/tmp#

re-ran dietpi-update for fun here is how it errored out.

                    ┌──────────────────────────────────────────────────┤ DietPi-Update ├───────────────────────────────────────────────────┐
                    │ APT upgrade                                                                                                          │ 
                    │  - Command: apt-get -qq upgrade                                                                                      │ 
                    │  - Exit code: 139                                                                                                    │ 
                    │  - DietPi version: v6.28.0 (MichaIng/master) | HW_MODEL: 3 | HW_ARCH: 2 | DISTRO: 5                                  │ 
                    │  - Image creator: DietPi Core Team                                                                                   │ 
                    │  - Pre-image: Raspbian Lite                                                                                          │ 
                    │  - Error log:                                                                                                        │ 
                    │

MichaIng, 2020-07-17:
Okay, `apt` is somehow broken, although probably not apt itself but something it invokes. Strange that `apt-get install apt` has the segfault after the package has been downloaded while `apt upgrade` has it before anything is downloaded 🤔. Can you try the other apt commands:
```
apt-cache show apt
apt-mark auto apt
```
Probably debconf pre-configuration? Test:
```
mv /etc/apt/apt.conf.d/70debconf{,_bak}
apt upgrade
```

blindtrust, 2020-07-17:
I am happy to try, here are the results.....

root@DietPi:/tmp# apt-cache show apt
Package: apt
Version: 1.8.2.1
Architecture: armhf
Maintainer: APT Development Team <deity@lists.debian.org>
Installed-Size: 3717
Depends: adduser, gpgv | gpgv2 | gpgv1, raspbian-archive-keyring, libapt-pkg5.0 (>= 1.7.0~alpha3~), libc6 (>= 2.15), libgcc1 (>= 1:3.5), libgnutls30 (>= 3.6.6), libseccomp2 (>= 1.0.1), libstdc++6 (>= 5.2)
Recommends: ca-certificates
Suggests: apt-doc, aptitude | synaptic | wajig, dpkg-dev (>= 1.17.2), gnupg | gnupg2 | gnupg1, powermgmt-base
Breaks: apt-transport-https (<< 1.5~alpha4~), apt-utils (<< 1.3~exp2~), aptitude (<< 0.8.10)
Replaces: apt-transport-https (<< 1.5~alpha4~), apt-utils (<< 1.3~exp2~)
Provides: apt-transport-https (= 1.8.2.1)
Priority: important
Section: admin
Filename: pool/main/a/apt/apt_1.8.2.1_armhf.deb
Size: 1349032
SHA256: 5070a471708b6d172d32d273e436c7bd8048bec9e37a2ad58f2ae94b39f14b46
SHA1: a8564c2c0dc1f0bf2ee433c0a7424b3918dbf745
MD5sum: 5673555a167ffcd17bc3d409ae1278fc
Description: commandline package manager
 This package provides commandline tools for searching and
 managing as well as querying information about packages
 as a low-level access to all features of the libapt-pkg library.
 .
 These include:
  * apt-get for retrieval of packages and information about them
    from authenticated sources and for installation, upgrade and
    removal of packages together with their dependencies
  * apt-cache for querying available information about installed
    as well as installable packages
  * apt-cdrom to use removable media as a source for packages
  * apt-config as an interface to the configuration settings
  * apt-key as an interface to manage authentication keys
Description-md5: 9fb97a88cb7383934ef963352b53b4a7

root@DietPi:/tmp# apt-mark auto apt
apt was already set to automatically installed.
root@DietPi:/tmp# 

root@DietPi:/tmp# mv /etc/apt/apt.conf.d/70debconf{,_bak}
root@DietPi:/tmp# apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  base-files bind9-host ca-certificates curl dnsutils e2fsprogs firmware-atheros firmware-brcm80211 firmware-iwlwifi firmware-misc-nonfree firmware-realtek
  git git-man idn2 iputils-ping libapt-inst2.0 libapt-pkg5.0 libbind9-161 libcom-err2 libcurl3-gnutls libcurl4 libdns-export1104 libdns1104 libext2fs2
  libgnutls30 libicu63 libidn2-0 libirs161 libisc-export1100 libisc1100 libisccc161 libisccfg163 libldap-2.4-2 libldap-common liblwres161 libpam-modules
  libpam-modules-bin libpam-runtime libpam0g libraspberrypi-bin libraspberrypi0 libss2 libssl1.1 libsystemd0 libtiff5 libudev1 openssl php7.3-cli
  php7.3-common php7.3-curl php7.3-fpm php7.3-gd php7.3-json php7.3-mbstring php7.3-opcache php7.3-readline php7.3-sqlite3 php7.3-xml php7.3-zip
  raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods sudo systemd systemd-sysv tzdata udev
67 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/131 MB of archives.
After this operation, 2,100 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
**Segmentation fault**
root@DietPi:/tmp#

MichaIng, 2020-07-17:
Okay lets upgrade some packages manually and see if that helps (especially the libapt* ones):
```
cd /tmp
wget http://raspbian.raspberrypi.org/raspbian/pool/main/b/base-files/base-files_10.3+rpi1+deb10u4_armhf.deb
dpkg -i base-files_10.3+rpi1+deb10u4_armhf.deb
rm base-files_10.3+rpi1+deb10u4_armhf.deb
wget http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/libapt-inst2.0_1.8.2.1_armhf.deb
dpkg -i libapt-inst2.0_1.8.2.1_armhf.deb
rm libapt-inst2.0_1.8.2.1_armhf.deb
wget http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/libapt-pkg5.0_1.8.2.1_armhf.deb
dpkg -i libapt-pkg5.0_1.8.2.1_armhf.deb
rm libapt-pkg5.0_1.8.2.1_armhf.deb
wget http://raspbian.raspberrypi.org/raspbian/pool/main/o/openssl/libssl1.1-udeb_1.1.1d-0+deb10u3_armhf.udeb
dpkg -i libssl1.1-udeb_1.1.1d-0+deb10u3_armhf.udeb
rm libssl1.1-udeb_1.1.1d-0+deb10u3_armhf.udeb
wget http://raspbian.raspberrypi.org/raspbian/pool/main/o/openssl/openssl_1.1.1d-0+deb10u3_armhf.deb
dpkg -i openssl_1.1.1d-0+deb10u3_armhf.deb
rm openssl_1.1.1d-0+deb10u3_armhf.deb
apt-get -y upgrade
```

blindtrust, 2020-07-17:
seemed to have issues with....
root@DietPi:/tmp# dpkg -i libssl1.1-udeb_1.1.1d-0+deb10u3_armhf.udeb then others after that... 

root@DietPi:/tmp# wget http://raspbian.raspberrypi.org/raspbian/pool/main/b/base-files/base-files_10.3+rpi1+deb10u4_armhf.deb
--2020-07-17 14:26:27--  http://raspbian.raspberrypi.org/raspbian/pool/main/b/base-files/base-files_10.3+rpi1+deb10u4_armhf.deb
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... 93.93.128.193, 2a00:1098:0:80:1000:75:0:3
Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: http://mirror.sjc02.svwh.net/raspbian/raspbian/pool/main/b/base-files/base-files_10.3+rpi1+deb10u4_armhf.deb [following]
--2020-07-17 14:26:28--  http://mirror.sjc02.svwh.net/raspbian/raspbian/pool/main/b/base-files/base-files_10.3+rpi1+deb10u4_armhf.deb
Resolving mirror.sjc02.svwh.net (mirror.sjc02.svwh.net)... 72.5.72.15, 2606:c680:0:b:3830:34ff:fe66:6663
Connecting to mirror.sjc02.svwh.net (mirror.sjc02.svwh.net)|72.5.72.15|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 70016 (68K) [application/octet-stream]
Saving to: ‘base-files_10.3+rpi1+deb10u4_armhf.deb’

base-files_10.3+rpi1+deb10u4_armhf.deb  100%[===============================================================================>]  68.38K   188KB/s    in 0.4s    

2020-07-17 14:27:10 (188 KB/s) - ‘base-files_10.3+rpi1+deb10u4_armhf.deb’ saved [70016/70016]

root@DietPi:/tmp# dpkg -i base-files_10.3+rpi1+deb10u4_armhf.deb
(Reading database ... 25095 files and directories currently installed.)
Preparing to unpack base-files_10.3+rpi1+deb10u4_armhf.deb ...
Unpacking base-files (10.3+rpi1+deb10u4) over (10.3+rpi1+deb10u2) ...
Setting up base-files (10.3+rpi1+deb10u4) ...
Installing new version of config file /etc/debian_version ...
Processing triggers for install-info (6.5.0.dfsg.1-4+b1) ...
root@DietPi:/tmp# rm base-files_10.3+rpi1+deb10u4_armhf.deb
root@DietPi:/tmp# wget http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/libapt-inst2.0_1.8.2.1_armhf.deb
--2020-07-17 14:27:35--  http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/libapt-inst2.0_1.8.2.1_armhf.deb
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... 93.93.128.193, 2a00:1098:0:80:1000:75:0:3
Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: http://mirror.sjc02.svwh.net/raspbian/raspbian/pool/main/a/apt/libapt-inst2.0_1.8.2.1_armhf.deb [following]
--2020-07-17 14:27:36--  http://mirror.sjc02.svwh.net/raspbian/raspbian/pool/main/a/apt/libapt-inst2.0_1.8.2.1_armhf.deb
Resolving mirror.sjc02.svwh.net (mirror.sjc02.svwh.net)... 72.5.72.15, 2606:c680:0:b:3830:34ff:fe66:6663
Connecting to mirror.sjc02.svwh.net (mirror.sjc02.svwh.net)|72.5.72.15|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 199356 (195K) [application/octet-stream]
Saving to: ‘libapt-inst2.0_1.8.2.1_armhf.deb’

libapt-inst2.0_1.8.2.1_armhf.deb        100%[===============================================================================>] 194.68K  76.3KB/s    in 2.6s    

2020-07-17 14:27:53 (76.3 KB/s) - ‘libapt-inst2.0_1.8.2.1_armhf.deb’ saved [199356/199356]

root@DietPi:/tmp# dpkg -i libapt-inst2.0_1.8.2.1_armhf.deb
(Reading database ... 25095 files and directories currently installed.)
Preparing to unpack libapt-inst2.0_1.8.2.1_armhf.deb ...
Unpacking libapt-inst2.0:armhf (1.8.2.1) over (1.8.2) ...
Setting up libapt-inst2.0:armhf (1.8.2.1) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
root@DietPi:/tmp# rm libapt-inst2.0_1.8.2.1_armhf.deb
root@DietPi:/tmp# wget http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/libapt-pkg5.0_1.8.2.1_armhf.deb
--2020-07-17 14:28:14--  http://raspbian.raspberrypi.org/raspbian/pool/main/a/apt/libapt-pkg5.0_1.8.2.1_armhf.deb
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... 93.93.128.193, 2a00:1098:0:80:1000:75:0:3
Connecting to raspbian.raspberrypi.org (raspbian.raspberrypi.org)|93.93.128.193|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: http://mirror.sjc02.svwh.net/raspbian/raspbian/pool/main/a/apt/libapt-pkg5.0_1.8.2.1_armhf.deb [following]
--2020-07-17 14:28:14--  http://mirror.sjc02.svwh.net/raspbian/raspbian/pool/main/a/apt/libapt-pkg5.0_1.8.2.1_armhf.deb
Resolving mirror.sjc02.svwh.net (mirror.sjc02.svwh.net)... 72.5.72.15, 2606:c680:0:b:3830:34ff:fe66:6663
Connecting to mirror.sjc02.svwh.net (mirror.sjc02.svwh.net)|72.5.72.15|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 854848 (835K) [application/octet-stream]
Saving to: ‘libapt-pkg5.0_1.8.2.1_armhf.deb’

libapt-pkg5.0_1.8.2.1_armhf.deb         100%[===============================================================================>] 834.81K   345KB/s    in 2.4s    

2020-07-17 14:28:17 (345 KB/s) - ‘libapt-pkg5.0_1.8.2.1_armhf.deb’ saved [854848/854848]

root@DietPi:/tmp# dpkg -i libapt-pkg5.0_1.8.2.1_armhf.deb
[... 98 lines cut ...]
HTTP request sent, awaiting response... 200 OK
Length: 73968 (72K) [application/octet-stream]
Saving to: ‘libidn2-0_2.0.5-1+deb10u1_armhf.deb’

libidn2-0_2.0.5-1+deb10u1_armhf.deb     100%[===============================================================================>]  72.23K   181KB/s    in 0.4s    

2020-07-17 14:34:34 (181 KB/s) - ‘libidn2-0_2.0.5-1+deb10u1_armhf.deb’ saved [73968/73968]
[... cut ...]
                  Depends: libcrypto1.1-udeb (>= 1.1.1d) but it is not installable
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
(Reading database ... 25096 files and directories currently installed.)
Preparing to unpack libidn2-0_2.0.5-1+deb10u1_armhf.deb ...
Unpacking libidn2-0:armhf (2.0.5-1+deb10u1) over (2.0.5-1) ...
Setting up libidn2-0:armhf (2.0.5-1+deb10u1) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
root@DietPi:/tmp# rm libidn2-0_2.0.5-1+deb10u1_armhf.deb
root@DietPi:/tmp#

blindtrust, 2020-07-17:
root@DietPi:/tmp# apt-get -y upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 libssl1.1-udeb : Depends: libc6-udeb (>= 2.28) but it is not installable
                  Depends: libcrypto1.1-udeb (>= 1.1.1d) but it is not installable
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
root@DietPi:/tmp# apt --fix-browken install
E: Command line option --fix-browken is not understood in combination with the other options
root@DietPi:/tmp# apt --fix-broken install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following packages will be REMOVED:
  libssl1.1-udeb
0 upgraded, 0 newly installed, 1 to remove and 63 not upgraded.
1 not fully installed or removed.
After this operation, 460 kB disk space will be freed.
Do you want to continue? [Y/n] Y
Segmentation fault
root@DietPi:/tmp#

MichaIng, 2020-07-17:
Ah sorry it should have been this one:
```
wget http://raspbian.raspberrypi.org/raspbian/pool/main/o/openssl/libssl1.1_1.1.1d-0+deb10u3_armhf.deb
dpkg -i libssl1.1_1.1.1d-0+deb10u3_armhf.deb
rm libssl1.1_1.1.1d-0+deb10u3_armhf.deb
```

blindtrust, 2020-07-17:
etc/apt/apt.conf.d/01autoremove-kernels
// DO NOT EDIT! File autogenerated by /etc/kernel/postinst.d/apt-auto-removal
APT::NeverAutoRemove
{
};
/* Debug information:
# dpkg list:
# list of installed kernel packages:

# list of different kernel versions:

# Installing kernel:  (4.19.75-v8+)
# Running kernel: ignored (4.19.57-v7+)
# Last kernel: 
# Previous kernel: 
# Kernel versions list to keep:

# Kernel packages (version part) to protect:
4\.19\.57-v7\+
4\.19\.75-v8\+
*/
/etc/apt/apt.conf.d/70debconf_bak
// Pre-configure all packages with debconf before they are installed.
// If you don't like it, comment it out.
DPkg::Pre-Install-Pkgs {"/usr/sbin/dpkg-preconfigure --apt || true";};
/etc/apt/apt.conf.d/98-dietpi-no_translations
Acquire::Languages "none";
/etc/apt/apt.conf.d/99-dietpi-forceconf
Dpkg::options {
   "--force-confdef";
   "--force-confold";
}
/etc/apt/apt.conf.d/99-dietpi-force-ipv4
Acquire::ForceIPv4 "true";
/etc/apt/apt.conf.d/99-dietpi-norecommends
APT::Install-Recommends "false";
APT::Install-Suggests "false";
APT::AutoRemove::RecommendsImportant "false";
APT::AutoRemove::SuggestsImportant "false";
root@DietPi:~#

MichaIng, 2020-07-17:
Ooops, I'm dumb, 70debconf_bak is still parsed with this suffix, we need to move it out of the directory:
```
mv /etc/apt/apt.conf.d/70debconf_bak /root/70debconf_bak
apt upgrade
```

blindtrust, 2020-07-17:
root@DietPi:~# mv /etc/apt/apt.conf.d/70debconf_bak /root/70debconf_bak
root@DietPi:~# apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  bind9-host ca-certificates curl dnsutils e2fsprogs firmware-atheros firmware-brcm80211 firmware-iwlwifi firmware-misc-nonfree firmware-realtek git git-man
  idn2 iputils-ping libbind9-161 libcom-err2 libcurl3-gnutls libcurl4 libdns-export1104 libdns1104 libext2fs2 libgnutls30 libicu63 libirs161 libisc-export1100
  libisc1100 libisccc161 libisccfg163 libldap-2.4-2 libldap-common liblwres161 libpam-modules libpam-modules-bin libpam-runtime libpam0g libraspberrypi-bin
  libraspberrypi0 libss2 libssl1.1 libsystemd0 libtiff5 libudev1 openssl php7.3-cli php7.3-common php7.3-curl php7.3-fpm php7.3-gd php7.3-json php7.3-mbstring
  php7.3-opcache php7.3-readline php7.3-sqlite3 php7.3-xml php7.3-zip raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods sudo systemd systemd-sysv
  tzdata udev
63 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/130 MB of archives.
After this operation, 2,106 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Segmentation fault
root@DietPi:~#

MichaIng, 2020-07-17:
Okay, it's not debconf. However please apply the two upgrades above:
```
dpkg -P libssl1.1-udeb
wget http://raspbian.raspberrypi.org/raspbian/pool/main/o/openssl/libssl1.1_1.1.1d-0+deb10u3_armhf.deb
dpkg -i libssl1.1_1.1.1d-0+deb10u3_armhf.deb
rm libssl1.1_1.1.1d-0+deb10u3_armhf.deb
wget http://raspbian.raspberrypi.org/raspbian/pool/main/g/gnutls28/libgnutls30_3.6.7-4+deb10u4_armhf.deb
dpkg -i libgnutls30_3.6.7-4+deb10u4_armhf.deb
rm libgnutls30_3.6.7-4+deb10u4_armhf.deb
```

blindtrust, 2020-07-17:
root@DietPi:/tmp# tail -20 log.strace-aptget
write(1, "  libpam0g libraspberrypi-bin li"..., 75  libpam0g libraspberrypi-bin libraspberrypi0 libss2 libssl1.1 libsystemd0
) = 75
write(1, "  libtiff5 libudev1 openssl php7"..., 76  libtiff5 libudev1 openssl php7.3-cli php7.3-common php7.3-curl php7.3-fpm
) = 76
write(1, "  php7.3-gd php7.3-json php7.3-m"..., 71  php7.3-gd php7.3-json php7.3-mbstring php7.3-opcache php7.3-readline
) = 71
write(1, "  php7.3-sqlite3 php7.3-xml php7"..., 62  php7.3-sqlite3 php7.3-xml php7.3-zip raspberrypi-bootloader
) = 62
write(1, "  raspberrypi-kernel raspberrypi"..., 75  raspberrypi-kernel raspberrypi-sys-mods sudo systemd systemd-sysv tzdata
) = 75
write(1, "  udev\n", 7  udev
)                 = 7
statfs64("/var/cache/apt/archives/", 88, 0x7efe15d0) = 0
write(1, "62 upgraded, 0 newly installed, "..., 19462 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/129 MB of archives.
After this operation, 2,105 kB of additional disk space will be used.
Do you want to continue?) = 194
write(1, " [Y/n] ", 7 [Y/n] )                  = 7
fstat64(0, {st_mode=S_IFCHR|0600, st_rdev=makedev(0x88, 0), ...}) = 0
read(0, root@DietPi:/tmp#

MichaIng, 2020-07-17:
Hmm could you do the strace again with non-interactive call:
```
cd /tmp
strace -f apt-get -qq upgrade >log.strace-aptget 2>&1
cat log.strace-aptget # If this is too much, better the below:
tail -20 log.strace-aptget
```

blindtrust, 2020-07-17:
root@DietPi:~# cd /tmp
root@DietPi:/tmp# strace -f apt-get -qq upgrade >log.strace-aptget 2>&1
Segmentation fault
root@DietPi:/tmp#

blindtrust, 2020-07-17:
maybe just rebuild my dietpi install? then reinstall pihole???

Joulinar, 2020-07-17:
if that is possible and did not cause to much trouble, it might be fastes way to get back to normal state

MichaIng, 2020-07-18:
And what is now the content of the created log?
```
tail -20 log.strace-aptget
```
Lets give it a last try (also since I am interested on what actually broke) and if this log does not contain something helpful, yes, I also suggest a fresh install. Copy the current SD card content somewhere else so you can recover any custom configs, in case made.

blindtrust, 2020-07-18:
root@DietPi:/tmp# cd /tmp
root@DietPi:/tmp# strace -f apt-get -qq upgrade >log.strace-aptget 2>&1
Segmentation fault
root@DietPi:/tmp# tail -20 log.strace-aptget
brk(0x1e04000)                          = 0x1e04000
getgroups32(65536, [0])                 = 1
setgroups32(1, [65534])                 = 0
setresgid32(-1, 65534, -1)              = 0
setresuid32(-1, 103, -1)                = 0
setresuid32(-1, 0, -1)                  = 0
setresgid32(-1, 0, -1)                  = 0
setgroups32(1, [0])                     = 0
close(5)                                = 0
stat64("/var/log/apt/", {st_mode=S_IFDIR|0755, st_size=100, ...}) = 0
unlink("/var/log/apt/eipp.log.xz")      = 0
openat(AT_FDCWD, "/var/log/apt/eipp.log.xz", O_WRONLY|O_CREAT|O_EXCL|O_LARGEFILE, 0644) = 5
fcntl64(5, F_GETFL)                     = 0x20001 (flags O_WRONLY|O_LARGEFILE)
mmap2(NULL, 13123584, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x72ee6000
mmap2(NULL, 17047552, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x71ea4000
mmap2(NULL, 67112960, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x6dea3000
fcntl64(5, F_SETFD, FD_CLOEXEC)         = 0
fstat64(5, {st_mode=S_IFREG|0644, st_size=0, ...}) = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x94d5384} ---
+++ killed by SIGSEGV +++

MichaIng, 2020-07-18:
Hmm its crashing shortly after accessing `/var/log/apt/eipp.log.xz`, although not sure if it is still related.
In my case (x86_64 VM) it looks like this:
```
stat("/var/log/apt/", {st_mode=S_IFDIR|0755, st_size=100, ...}) = 0
unlink("/var/log/apt/eipp.log.xz")      = 0
openat(AT_FDCWD, "/var/log/apt/eipp.log.xz", O_WRONLY|O_CREAT|O_EXCL, 0644) = 5
fcntl(5, F_GETFL)                       = 0x8001 (flags O_WRONLY|O_LARGEFILE)
mmap(NULL, 13123584, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f9bb0955000
mmap(NULL, 17047552, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f9baf913000
mmap(NULL, 67112960, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f9bab912000
fcntl(5, F_SETFD, FD_CLOEXEC)           = 0
fstat(5, {st_mode=S_IFREG|0644, st_size=0, ...}) = 0
write(5, "\3757zXZ\0\0\4\346\326\264F\2\0!\1\26\0\0\0t/\345\243\341\10>-\6]\0)"..., 4096) = 4096
write(5, "\245\5\374y,\253-\21\4M\246.4(L\210\300\330w\275\1\vU\303\376\3038-^d\355\341"..., 4096) = 4096
munmap(0x7f9bab912000, 67112960)        = 0
munmap(0x7f9baf913000, 17047552)        = 0
munmap(0x7f9bb0955000, 13123584)        = 0
...
```
What I am wondering now is why in your case `stat64` and `fcntl64` is called even that it is a 32-bit OS while in my case there is no `64` suffix but a 64-bit OS 🤔.

blindtrust, 2020-07-19:
root@DietPi:~# free -m

              total        used        free      shared  buff/cache   available
Mem:            975          48         637          17         289         858
Swap:          1072           0        1072

root@DietPi:~#

MichaIng, 2020-07-19:
Last attempt
```
/boot/dietpi/func/dietpi-set_software apt-cache reset
apt update
apt upgrade
```
Else last idea I have is to manually update further packages, especially the kernel, as we did for a few above. Probably there is indeed something broken with kernel/firmware which might be fixed with the kernel update.

blindtrust, 2020-07-19:
root@192.168.1.4's password: 
 ─────────────────────────────────────────────────────
 DietPi v6.28.0 : Update available
 ─────────────────────────────────────────────────────
 - Device model : RPi 3 Model B (armv7l)
 - CPU temp : 52'C : 125'F (Running warm, but safe)
 - LAN IP : 192.168.1.4 (eth0)
 - MOTD : Help testing the new RPi3/RPi4 64-bit image:
          https://github.com/MichaIng/DietPi/issues/3570
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image           : DietPi Core Team (pre-image: Raspbian Lite)
 Web             : https://dietpi.com | https://twitter.com/dietpi_
 Patreon Legends : Bryce
 Donate          : https://dietpi.com/#donate
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-update   : Run now to update DietPi from v6.28.0 to v6.31.2.

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi:~# /boot/dietpi/func/dietpi-set_software apt-cache reset

 DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: apt-cache (reset)

[  OK  ] apt-cache reset | Completed
root@DietPi:~# apt update\
> exit
E: Invalid operation updateexit
root@DietPi:~# 
root@DietPi:~# 
root@DietPi:~# 
root@DietPi:~# apt update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 https://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 https://archive.raspberrypi.org/debian buster/main armhf Packages [330 kB]
Fetched 13.4 MB in 11s (1,224 kB/s)                                                                                                                            
Reading package lists... Done
Building dependency tree       
Reading state information... Done
62 packages can be upgraded. Run 'apt list --upgradable' to see them.
root@DietPi:~# apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  bind9-host ca-certificates curl dnsutils e2fsprogs firmware-atheros firmware-brcm80211 firmware-iwlwifi firmware-misc-nonfree firmware-realtek git
  git-man idn2 iputils-ping libbind9-161 libcom-err2 libcurl3-gnutls libcurl4 libdns-export1104 libdns1104 libext2fs2 libicu63 libirs161 libisc-export1100
  libisc1100 libisccc161 libisccfg163 libldap-2.4-2 libldap-common liblwres161 libpam-modules libpam-modules-bin libpam-runtime libpam0g libraspberrypi-bin
  libraspberrypi0 libss2 libssl1.1 libsystemd0 libtiff5 libudev1 openssl php7.3-cli php7.3-common php7.3-curl php7.3-fpm php7.3-gd php7.3-json
  php7.3-mbstring php7.3-opcache php7.3-readline php7.3-sqlite3 php7.3-xml php7.3-zip raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods sudo
  systemd systemd-sysv tzdata udev
62 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 129 MB of archives.
After this operation, 2,105 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:2 https://archive.raspberrypi.org/debian buster/main armhf libpam0g armhf 1.3.1-5+rpt2 [121 kB]
Get:1 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf libsystemd0 armhf 241-7~deb10u4+rpi1 [305 kB]
Get:4 https://archive.raspberrypi.org/debian buster/main armhf libpam-modules-bin armhf 1.3.1-5+rpt2 [103 kB]
Get:5 https://archive.raspberrypi.org/debian buster/main armhf libpam-modules armhf 1.3.1-5+rpt2 [290 kB]
Get:3 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf systemd armhf 241-7~deb10u4+rpi1 [3,255 kB]
[... 35 lines cut ...]
Get:41 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf libisc-export1100 armhf 1:9.11.5.P4+dfsg-5.1+deb10u1 [360 kB]                       
Get:42 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf libdns-export1104 armhf 1:9.11.5.P4+dfsg-5.1+deb10u1 [871 kB]                       
Get:43 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf libicu63 armhf 63.1-6+deb10u1 [7,973 kB]                                            
Get:59 https://archive.raspberrypi.org/debian buster/main armhf libraspberrypi-bin armhf 1.20200601-1 [339 kB]                                              
Get:60 https://archive.raspberrypi.org/debian buster/main armhf libraspberrypi0 armhf 1.20200601-1 [850 kB]                                                 
Get:61 https://archive.raspberrypi.org/debian buster/main armhf raspberrypi-bootloader armhf 1.20200601-1 [4,870 kB]                                        
Get:62 https://archive.raspberrypi.org/debian buster/main armhf raspberrypi-sys-mods armhf 20200514 [10.6 kB]                                               
Get:44 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf libss2 armhf 1.44.5-1+deb10u3 [71.0 kB]                                             
Get:45 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf libtiff5 armhf 4.1.0+git191117-2~deb10u1 [249 kB]                                   
Get:46 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf php7.3-zip armhf 7.3.19-1~deb10u1 [17.7 kB]                                         
[... cut ...]
Get:57 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf php7.3-common armhf 7.3.19-1~deb10u1 [522 kB]                                       
Get:58 http://mirror.sjc02.svwh.net/raspbian/raspbian buster/main armhf sudo armhf 1.8.27-1+deb10u2 [1,201 kB]                                              
Fetched 129 MB in 23s (5,558 kB/s)                                                                                                                          
**Segmentation fault**
root@DietPi:~#

blindtrust, 2020-07-19:
Thank you so much for you time trying to fix this... Just wanted to say that!

MichaIng, 2020-07-19:
Last attempt is upgrading the kernel:
```
cd /tmp
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20200601-1_armhf.deb
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20200601-1_armhf.deb
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-bin_1.20200601-1_armhf.deb
wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi0_1.20200601-1_armhf.deb
dpkg -i *raspberrypi*.deb
reboot
```

blindtrust, 2020-07-19:
root@192.168.1.4's password: 
 ─────────────────────────────────────────────────────
 DietPi v6.28.0 : Update available
 ─────────────────────────────────────────────────────
 - Device model : RPi 3 Model B (armv7l)
 - CPU temp : 53'C : 127'F (Running warm, but safe)
 - LAN IP : 192.168.1.4 (eth0)
 - MOTD : Help testing the new RPi3/RPi4 64-bit image:
          https://github.com/MichaIng/DietPi/issues/3570
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image           : DietPi Core Team (pre-image: Raspbian Lite)
 Web             : https://dietpi.com | https://twitter.com/dietpi_
 Patreon Legends : Bryce
 Donate          : https://dietpi.com/#donate
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-update   : Run now to update DietPi from v6.28.0 to v6.31.2.

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi:~# cd /tmp
root@DietPi:/tmp# wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20200601-1_armhf.deb
--2020-07-19 23:30:12--  http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20200601-1_armhf.deb
Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 46.235.230.122, 46.235.231.111, 46.235.231.145, ...
Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|46.235.230.122|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 71676420 (68M) [application/x-debian-package]
Saving to: ‘raspberrypi-kernel_1.20200601-1_armhf.deb’

raspberrypi-kernel_1.20200601-1_armhf.d 100%[===============================================================================>]  68.36M  8.96MB/s    in 8.5s    

2020-07-19 23:30:21 (8.09 MB/s) - ‘raspberrypi-kernel_1.20200601-1_armhf.deb’ saved [71676420/71676420]

root@DietPi:/tmp# wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20200601-1_armhf.deb
--2020-07-19 23:30:42--  http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20200601-1_armhf.deb
Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 46.235.231.145, 46.235.231.151, 93.93.130.212, ...
Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|46.235.231.145|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4870192 (4.6M) [application/x-debian-package]
Saving to: ‘raspberrypi-bootloader_1.20200601-1_armhf.deb’

raspberrypi-bootloader_1.20200601-1_arm 100%[===============================================================================>]   4.64M  4.18MB/s    in 1.1s    

2020-07-19 23:30:43 (4.18 MB/s) - ‘raspberrypi-bootloader_1.20200601-1_armhf.deb’ saved [4870192/4870192]

root@DietPi:/tmp# wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-bin_1.20200601-1_armhf.deb
--2020-07-19 23:30:52--  http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-bin_1.20200601-1_armhf.deb
Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 46.235.230.122, 46.235.231.111, 46.235.231.145, ...
Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|46.235.230.122|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 339300 (331K) [application/x-debian-package]
Saving to: ‘libraspberrypi-bin_1.20200601-1_armhf.deb’

libraspberrypi-bin_1.20200601-1_armhf.d  30%[=======================>                                                        ] 100.17K   381KB/s               
libraspberrypi-bin_1.20200601-1_armhf.d 100%[===============================================================================>] 331.35K   910KB/s    in 0.4s    

2020-07-19 23:30:53 (910 KB/s) - ‘libraspberrypi-bin_1.20200601-1_armhf.deb’ saved [339300/339300]

root@DietPi:/tmp# wget http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi0_1.20200601-1_armhf.deb
--2020-07-19 23:30:53--  http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi0_1.20200601-1_armhf.deb
Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 93.93.135.117, 93.93.135.118, 93.93.135.141, ...
Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|93.93.135.117|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 850168 (830K) [application/x-debian-package]
[... 516 lines cut ...]

rebott seemed to work....

root@192.168.1.4's password: 
 ─────────────────────────────────────────────────────
 DietPi v6.28.0 : Update available
 ─────────────────────────────────────────────────────
 - Device model : RPi 3 Model B (armv7l)
 - CPU temp : 54'C : 129'F (Running warm, but safe)
 - LAN IP : 192.168.1.4 (eth0)
 - MOTD : Help testing the new RPi3/RPi4 64-bit image:
          https://github.com/MichaIng/DietPi/issues/3570
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image           : DietPi Core Team (pre-image: Raspbian Lite)
 Web             : https://dietpi.com | https://twitter.com/dietpi_
 Patreon Legends : Bryce
 Donate          : https://dietpi.com/#donate
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-update   : Run now to update DietPi from v6.28.0 to v6.31.2.

ran dietpi-update just to see.............   
still got error on apt-get -qq upgrade

MichaIng, 2020-07-19:
Okay, sadly I'm out of ideas, besides manually upgrading all the other packages which at least allows the DietPi upgrade for now. No idea what might cause this segfault and at least I can't get useful information from the error strace.

If reflashing DietPi from scratch is not an option for you, you might find help at https://superuser.com/ or one of the other StackExchange sites: https://stackexchange.com/sites

blindtrust, 2020-07-19:
THANKS for all you have done! It works just fine with pihole as of now so I won't change it. I will rebuild it when I have some time in the near future.  I really like all the work you do with dietpi and really enjoy seeing all the changes after updates. Thanks again for the great work and support! Thanks for taking the time!

Joulinar, 2020-07-20:
Ok I guess we will close the issue even if we did not found the cause of the issue correct?

Joulinar, 2020-07-29:
ok I will close this issue, feel free to reopen if needed.

closed 2020-07-29 by Joulinar