MichaIng/DietPi #3682: Allow an option to expand the partition on boot like Raspbian

8bit-coder, 2020-07-18:
Raspbian does a thing on first time boot where it extends the partition of the os(probably a few gigs) to fill the rest of the drive. I'm having an issue where after using balenaetcher on a 512gb hard drive, balenaetcher still only creates a 768 mb large partition, even though i have 500 more gigs in the drive. This causes all sorts of issues in dietpi because it always errors out saying that there's not enough space.

Joulinar, 2020-07-18:
Hi,

Many thanks for you report. Usually the. Same happen on DietPi and the RootsFS should be extended. But it should be possible as well using Drive Manager to do the same.

8bit-coder, 2020-07-18:
How would I use the driver manager to do so?

MichaIng, 2020-07-18:
First of all please check the log why the automated expansion did not work:
```
cat /var/tmp/dietpi/logs/fs_partition_resize.log
```

`dietpi-drive_manager` will just re-enable the same service, do it might fail the same way and depending on above error message we might need to do the expansion manually or fix some other issue first, lets see.

8bit-coder, 2020-07-18:
Hmmm, I decided to just try reformatting the drive with dietpi and suddenly the expansion worked. I guess something went wrong the first time but I don't have the log files to do anything about it. Thanks for the help anyways.

closed 2020-07-18 by 8bit-coder