MichaIng/DietPi #3683: Add Software install dialog when install Desktop

TamaTamaGoGo, 2020-07-18:
### Add Software install dialog when install Desktop

#### Describe of request:

Add Software install dialog when install Desktop.
Like for example
 - [FireFox-esr] firefox-esr
 - [Chromium] ?
 - [Midori] midori
 - [epiphany] epiphany-browser

**Another details will add here when some idea appear or good idea at discuss**

MichaIng, 2020-07-18:
#### Background:
- Currently we pre-install Firefox with all desktops.
- It is not visible on all desktops, i.e. the default browser icon is there and opens Firefox, but no Firefox icon is shown, but hidden in menu instead.
- As we have Chromium as separate install option, when selected, users have two browsers installed.

#### Implementation ideas:
- Make Firefox a dedicated software title with own ID, like Chromium is. This also enables us to implement some enhancements, e.g. having browser cache in tmpfs or on external drive to enhance performance and reduce SD card writes.
- Browser preference system like we have for webservers, including the option "none".
- Add interactive preference selections for both preference systems when a desktop respectively a webserver dependant is installed. I think we can even derive if a preference has been actively chosen before to then skip the obsolete dialog.
- Add further browsers by times.

TamaTamaGoGo, 2020-07-19:
[NOTE]
Share with Owner (@MichaIng ) and ( @TamaTamaGoGo )

**Now empty**

TamaTamaGoGo, 2020-07-19:
I know how to use tmpfs for Chromium

    sudo nano /etc/fstab

And add

    tmpfs  /home/pi/.cache/chromium/Default/Cache  tmpfs  defaults,size=768m,noatime,mode=1777  0  0

MichaIng, 2020-07-19:
tmpfs cache for Chromium (browsers in general) or on low RAM devices cache on external drive (i.e. dietpi_userdata) is a great idea, however its again a separate topic, lets concentrate on a browser selection for now 😉. Also additional browsers are a separate topic, especially when we want to implement nicely (e.g. according to cache, probably GPU acceleration).

> Description: The old version of Chrome

Chromium is the underlying open source project, while "Chrome" is just Googles Chromium-based browser, like Opera, Brave, Microsofts Chromium Edge, Vivaldi etc, all are Chromium browsers. On Debian/Raspbian you get a very basic Chromium version without any branding or additions, indeed being visually close to "Chrome" but e.g. without their specific telemetry additions.

TamaTamaGoGo, 2020-07-19:
Hm...?
What should we describe shortly...

TamaTamaGoGo, 2020-07-28:
With some researching I found these things

**Midori**
YouTube said YouTube couldn't find corresponding video type.

**epiphany**
YouTube said YouTube couldn't find corresponding video type.

When I installed epiphany browser with following this command
`sudo apt-get install epiphany-browser`

epiphany doesn't appeared to Application Menu.

MichaIng, 2021-05-18:
Done: #4314
For now Firefox has become an independent software option for this but otherwise no new browser has been implemented yet. `epiphany-browser` was an idea, Brave and Opera for x86_64. But I'll now concentrate on the other planned tasks for v7.2.

closed 2021-05-18 by MichaIng