MichaIng/DietPi #3684: Reducing or disabling SWAP-file on RPiCM

danergo, 2020-07-20:
Hi, this question might be a duplicate, in this case sorry, I could not find a true answer.

On RPiCM there is an onboard 4GB (3.6GB actually) storage which is not too much, and SWAP file eats 1.5GB of it.
DietPi itself is small (that's why I like it), but I'm running out of space, and SWAP file is almost 50% of the whole storage.

Is it safe to shrink or disable the SWAP file? (Yes, I know it depends on usage. For my own purposes I personally don't need the SWAP) But for DietPi to run smooth is it needed, or not mandatory?

Thanks a lot!

MichaIng, 2020-07-20:
The default swap file size is only to cover RAM usage of most installs and software usage we offer. Of course you can reduce or completely remove it if your maximum RAM usage allows. Check `free -m` (available memory) or `htop` to check current usage and then I'd assure that total memory (including swap) is at least double of size of what is currently used, to cover peaks without OOM killer intervention.

Swap size can be adjusted via `dietpi-drive_manager` > `/`, or quickly:
```
/boot/dietpi/func/dietpi-set_swapfile 512 # Sets swap size to 512 MiB
/boot/dietpi/func/dietpi-set_swapfile 0 # Removes the swap file completely
```

And as a general recommendation: If your usage rises, best practice is to add an external drive and move dietpi_userdata + the swap file there, so minimal writes are done to the internal SD and its space is fully preserved for system/OS files.

danergo, 2020-07-20:
htop shows: 132/480M for MEM, and 150M/1.53G for SWP.

Does it mean 132 is used from 480 of physical memory and 150 from SWP, altogether 282MB?

MichaIng, 2020-07-20:
Yes that's what it means. However, as far as I can say, the 150M in swap are probably not even used, AFAIK its data is not cleared immediately when becoming obsolete but only when its touched the next time, or such. But it shows that the swap file was indeed used, which means (with your minimal swappiness=1, means on DietPi the swapfile is only used in most urgent situations) that 480+150=630M moreless was the overall usage during some peak time, like daily cron job, a heavier software install, where larger files were downloaded to /tmp or compiling happened or such. So to stay on the safe side, I'd keep 512M swap.

danergo, 2020-07-20:
Actually I've tried to compile a pretty hard library on it, and I've even interrupted the compilation process after 20mins, and went cross-compiling on a server machine where it took ~1minute. So it could pump up the swap usage.

To sum things up, after a clean boot, SWAP shall be 0, and if on a long run it stays 0, I can safely disable it, because the system can work healthy from its physical RAM?

MichaIng, 2020-07-20:
> So it could pump up the swap usage.

Jep, very likely, additionally because compiling usually writes data to `/tmp` as well (RAMdisk). If you killed the compiling and did not reboot, it might be still there 😉.

> To sum things up, after a clean boot, SWAP shall be 0, and if on a long run it stays 0, I can safely disable it, because the system can work healthy from its physical RAM?

Yes indeed that is a good strategy, leave it running at least for 24h to cover the daily cron jobs as well (happening 0:17 by default). And if you face any strange behaviour, crashed services or such, keep that in mind, check `dmesg` or `journalctl` it the OOM killer was responsible.

And note that `dietpi-set_swapfile` adjusts the `/tmp` size to be 50% of total memory, hence it will be 240M then. This is usually enough but there are `dietpi-software` installs which temporarily require more (compiling or simply download + extract a larger archive), hence when you see errors about missing free space, you know what to do.

danergo, 2020-07-20:
Perfect, thank you very much!

closed 2020-07-20 by danergo