MichaIng/DietPi #3685: Wifi not working on a Pi 4

CactiChameleon9, 2020-07-20:
#### Required Information
- DietPi version | CORE=6 SUB=31 RC=2 master MichaIng
- Distro version | buster
- Kernel version | Linux DietPi 5.4.51-v71+ #1326 SPM Fri Jul 17 10:51:17 BST 2020 arm71 GNU/Linux
- SBC model | RPi4
- Power supply used | A random USB port that provides enough current for the pi 4 not to shut down under load
- SDcard used | SanDisk ultra 64GB

#### Steps to reproduce
Install dietpi on a pi 4 and following usual steps with connecting to wifi, reboot and wifi will not work

#### Expected behaviour
Wifi should work as normal

#### Actual behaviour
Wifi "Not Found" in dietpi-config network after first reboot

#### Extra details
A lot of updates to do with firmware occurred during first update
Same issue with 64bit dietpi

CactiChameleon9, 2020-07-20:
Oh... so I probably did something wrong then :D. Ill have a look and get back to you.

CactiChameleon9, 2020-07-20:
On boot it states that a valid connection was found but systemd-timesyncd never completes.

CactiChameleon9, 2020-07-20:
Yeah... wifi works now. Thanks. I was wondering what onboard WiFi was, that makes sense now that it is the wifi on the board. Maybe inbuilt would be a better word?

MichaIng, 2020-07-20:
> Maybe inbuilt would be a better word?

"onboard" is actually the right word to say that is it baked into the (main)board/PCB, you say onboard audio and onboard Ethernet as well. Probably I find a way to auto-detect when the onboard adapter is configured and automatically enable/keep it enabled then.

CactiChameleon9, 2020-07-20:
Fair enough - onboard it is, that would be cool if that could get implemented.

Joulinar, 2020-07-20:
Hi guys,

I guess we can close the topic?

MichaIng, 2020-07-20:
From my end yes, I'll have it in mind when coding the new menus.

CactiChameleon9, 2020-07-21:
Yep, you can close this. Thanks for all of your help

closed 2020-07-21 by Joulinar