MichaIng/DietPi #3686: Manually expand the second partition to fill up all space

Sopor, 2020-07-20:
If i want to change micro SD card on my RPi to a larger one and to do that i make an image of my 4GB card and i write it to my 16GB card. To use it i assume i have to do something if i want to use all the 16GB, right?

DietPi will automatically expand the second partition to fill up all space when i boot DietPi for the first time, but is there a way to run the same script DietPi do, but manually to increase the second partition?

Maybe DietPi detect it even if it is not the first run? Will it then ask to increase the partition before it do it?

Joulinar, 2020-07-20:
Hi,

many thanks for your report. You can use `dietpi-drive_manager` to expand your RootFS once your switched SD card.

MichaIng, 2020-07-20:
Easiest is probably to enable the resize service before shutting down the system to clone/copy content to the new SD card:
```
systemctl enable dietpi-fs_partition_resize
```
That way it is expanded automatically on next boot. It also does not hurt when booting the same SD card.

Sopor, 2020-07-22:
Now i know it is quite easy. Thanks for all your help.

closed 2020-07-22 by Sopor