MichaIng/DietPi #3688: Nextcloud Internal Server Error

CactiChameleon9, 2020-07-21:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | buster
- Kernel version | `uname -a`
- SBC model | RPi4
- Power supply used | Random USB port which is tested to work fine under load
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
Freshly installed nextcloud, gives error when accessing webpage

#### Steps to reproduce
Install nextcloud from fresh install

#### Expected behaviour
I should be able to access from x.x.x.x/nextcloud

#### Actual behaviour
The webpage opens with the error:
```
Internal Server Error

The server encountered an internal error and was unable to complete your request.
Please contact the server administrator if this error reappears multiple times, please include the technical details below in your report.
More details can be found in the webserver log
```.

#### Extra details
I an running pi hole (web interface works fine). I am also using the 64 bit image

Joulinar, 2020-07-21:
Hi,

Thanks for your report. Pls can you check if all you services are running. Especially the database.

`dietpi-services status`

CactiChameleon9, 2020-07-21:
All are listed as active

CactiChameleon9, 2020-07-21:
```
[  OK  ] DietPi-Services | proftpd		active (running) since Tue 2020-07-21 08:34:27 BST; 32min ago
[  OK  ] DietPi-Services | redis-server		active (running) since Tue 2020-07-21 08:34:27 BST; 32min ago
[  OK  ] DietPi-Services | mariadb		active (running) since Tue 2020-07-21 08:34:29 BST; 32min ago
[  OK  ] DietPi-Services | php7.3-fpm		active (running) since Tue 2020-07-21 08:34:30 BST; 32min ago
[  OK  ] DietPi-Services | lighttpd		active (running) since Tue 2020-07-21 08:34:30 BST; 32min ago
[  OK  ] DietPi-Services | cron			active (running) since Tue 2020-07-21 08:34:30 BST; 32min ago
[  OK  ] DietPi-Services | dropbear		active (running) since Tue 2020-07-21 08:34:26 BST; 32min ago
[  OK  ] DietPi-Services | pihole-FTL		active (running) since Tue 2020-07-21 08:40:56 BST; 25min ago
[  OK  ] DietPi-Services | dietpi-ramlog	active (exited) since Tue 2020-07-21 08:33:39 BST; 33min ago
[  OK  ] DietPi-Services | dietpi-preboot	active (exited) since Tue 2020-07-21 08:33:39 BST; 33min ago
[  OK  ] DietPi-Services | dietpi-boot		active (exited) since Tue 2020-07-21 08:34:26 BST; 32min ago
[  OK  ] DietPi-Services | dietpi-postboot	active (exited) since Tue 2020-07-21 08:34:26 BST; 32min ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor	inactive (dead
```

CactiChameleon9, 2020-07-21:
`cat /var/log/lighttpd/error.log `
`2020-07-21 09:45:59: (mod_fastcgi.c.421) FastCGI-stderr: PHP message: PHP Parse error:  syntax error, unexpected ''memcache.local'' (T_CONSTANT_ENCAPSED_STRING), expecting ')' in /var/www/nextcloud/config/config.php on line 14`

CactiChameleon9, 2020-07-21:
OK... the issue is with that config file listed, and I think (just guessing) that line 13 is wrong:
`'hashingThreads' => 4 `
to 
`'hashingThreads' => 4,`

CactiChameleon9, 2020-07-21:
Yep... I changed the file, rebooted, and nextcloud now works

Joulinar, 2020-07-21:
pls can you post your `/var/www/nextcloud/config/config.php` ? I did a test installation on my RPi3B+ and it's working ootb. I checked my file and don't have such an entry at all

```
root@DietPi3:~# cat /var/www/nextcloud/config/config.php
<?php
$CONFIG = array (
  'passwordsalt' => 'P1a9FDjNAGsH7ovilwIAMCImSRT5Zi',
  'secret' => 's2/RN2z5pIQ44eH46A3r50bSNlNLgbUEQkQSeZRT9lzmXWJr',
  'trusted_domains' =>
  array (
    0 => 'localhost',
    1 => '*',
  ),
  'datadirectory' => '/mnt/dietpi_userdata/nextcloud_data',
  'dbtype' => 'mysql',
  'version' => '19.0.1.1',
  'memcache.local' => '\\OC\\Memcache\\APCu',
  'filelocking.enabled' => true,
  'memcache.locking' => '\\OC\\Memcache\\Redis',
  'redis' =>
  array (
    'host' => '/run/redis/redis-server.sock',
    'port' => 0,
  ),
  'overwrite.cli.url' => 'http://localhost/nextcloud',
  'dbname' => 'nextcloud',
  'dbhost' => 'localhost',
  'dbport' => '',
  'dbtableprefix' => 'oc_',
  'mysql.utf8mb4' => true,
  'dbuser' => 'oc_admin',
  'dbpassword' => 'xxx',
  'installed' => true,
  'instanceid' => 'ocsir3ueh50r',
  'maintenance' => false,
);
root@DietPi3:~#
```

MichaIng, 2020-07-21:
Ah its the WIP change from dev branch where I started to add Argon2 hash settings to reduce hash times throughout Nextcloud. Many thanks for reporting.

Joulinar, 2020-07-21:
@CactiChameleon9 
keep in mind that the DietPi RPi 64Bit image is still in Alpha phase. As well Raspberry OS 64bit themselves is still Beta as well. There might be a chance that thinks are not working as expected. 😉

CactiChameleon9, 2020-07-21:
@Joulinar I am aware of this, thanks!

MichaIng, 2020-07-21:
The new kernel 5.4 packages have been just released, including official 64-bit packages. I'm currently creating a new image based on this and DietPi stable branch, hence it can be assumed more stable in two layers then 😃.

CactiChameleon9, 2020-07-21:
That sounds great!

MichaIng, 2020-07-21:
New image up: https://dietpi.com/downloads/images/testing/DietPi_RPi-ARMv8-Buster.7z

closed 2020-09-09 by MichaIng