MichaIng/DietPi #3689: NanoPi NEO3

MichaIng, 2020-07-21:
### Creating an image request

#### Formal device information
- Device name | NanoPi NEO3
- Official product URL | https://www.friendlyarm.com/index.php?route=product/product&product_id=279&search=NEO3&description=true&category_id=0&sub_category=true
- GitHub resource URL (if available) | 
- Image download URL | https://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO3

#### Is the SBC officially supported by the Debian installer?
- No

#### If not, is a reliable 3rd party Debian image available for this SBC?
- https://www.armbian.com/nanopineo3/

#### Vote for this image on FeatHub: https://feathub.com/MichaIng/DietPi/+172

StephanStS, 2020-07-31:
@MichaIng: I have generated a NEO3 image. You can download it from my cloud like before (or provide me an upload link).
I generated it as "Generic Device (aarch64)" based on Armbian with linux kernel 5.4.51, an own device entry is appreciated. Branch was 'master', because 'dev' gave an error quite late by mounting /tmp.
First short test with mopidy, Webmin, phpsysinfo and node-red (only install and start, no deep application test): 

- Phpsysinfo: Works.
- Webmin: Problem with Webmin installation (will be investigated later)
- Mopidy: Works.
- Node-RED: Works.

[image]

MichaIng, 2020-07-31:
Nice, I'll get a developer sample next month so will add native code support before that.

> Branch was 'master', because 'dev' gave an error quite late by mounting /tmp.

You mean DietPi-PREP script branch or what you selected during process for the target image?
AFAIK I made some other changes to DietPi-PREP dev branch that are not compatible for creating master branch images, so either both must be dev or (and I'd recommend that) use master for both. To apply three quick and dirty fixes I used the following (current state):
```
sed -i 's/taaem/MichaIng/' /boot/dietpi/dietpi-software
sed -i 's|$temp_bytes_result / $temp_time_result|$temp_bytes_result * 1000000 / $(printf '%.0f' "${temp_time_result}e6")|' /boot/dietpi/func/dietpi-benchmark
sed -i "s/G_WHIP_MENU_ARRAY+=('ARM Idle Frequency'/#G_WHIP_MENU_ARRAY+=('ARM Idle Frequency'/" /boot/dietpi/dietpi-config
rm -Rfv /{root,home/*}/.{bash_history,nano_history,local,config,cache,gnupg,dbus,gconf,nano,vim} /var/{cache/debconf,lib/dpkg}/*-old /etc/resolv.conf_bak
```
- The last only cleanup if I do some last checks or changes.

StephanStS, 2020-08-01:
> You mean DietPi-PREP script branch or what you selected during process for the target image?
> AFAIK I made some other changes to DietPi-PREP dev branch that are not compatible for creating master [...]
I meant: DietPi-PREP script I always take dev branch, the selected target image I changed from dev to master.

Additionally I tested image generation (based on Armbian NEO3 image) with both master and both dev:

1. Both master: 
```
#### Details:
- Date           | Sat Aug  1 21:02:13 UTC 2020
- Bug report     |
- DietPi version | v6.31.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       |  (ID=22)
- Kernel version | Linux nanopineo3 5.4.51-rockchip64 #20.05.9 SMP PREEMPT Sun Jul 12 11:28:47 UTC 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -qq --allow-change-held-packages install apt-utils bash-completion bc bzip2 ca-certificates console-setup cron curl debconf dirmngr ethtool fake-hwclock gnupg htop ifupdown iputils-ping isc-dhcp-client kmod locales nano p7zip parted procps psmisc sudo systemd-sysv tzdata udev unzip usbutils wget whiptail dropbear-run fdisk initramfs-tools haveged hdparm linux-dtb-current-rockchip64 linux-u-boot-nanopineo3-current linux-image-current-rockchip64 linux-buster-root-current-nanopineo3 sunxi-tools armbian-firmware
- Exit code      | 100
- Software title | DietPi-PREP
#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. Install Armbian NEO3 image
2. Run DietPi-PREP script, branch "master" and target image "branch master"
#### Expected behaviour:
<!-- What SHOULD happen? -->
- Image generation should run without errors
#### Actual behaviour:
<!-- What IS happening? -->
- See error message below
#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- Manually changing the command by removing the "linux-buster-root-current-nanopineo3" from the apt-get command and rerun the command (option within the PREP script) works. 
- But then another problem occurs:
    Command: mount -o remount /tmp
    - Exit code: 32
    - DietPi version: v6.31.2 (MichaIng/master) | HW_MODEL: 22 | HW_ARCH: 3 | DISTRO: 5
    - Error log:
         mount: /tmp: mount point not mounted or bad option.
    Then: Changing the command to "mount /tmp" works and the PREP script finishes without further errors.

#### Additional logs:
E: Package 'linux-buster-root-current-nanopineo3' has no installation candidate
```

2. Both dev: 
Only the second problem given above within "Extra details" with "mount: /tmp: mount point not mounted or bad option." occurs and can be 'workarounded' with "mount /tmp".

MichaIng, 2020-08-02:
> E: Package 'linux-buster-root-current-nanopineo3' has no installation candidate

Okay it looks like the image/repo is not yet complete from Armbian side. DietPi-PREP only marks the already installed packages for install/upgrade, hence `linux-buster-root-current-nanopineo3` is installed but missing in the (installed) APT repo.

Just checked, indeed its missing in the stable repo but available on beta: http://beta.armbian.com/pool/main/l/linux-buster-root-current-nanopineo3/
Strange that `/etc/apt/sources.list.d/*armbian*.list` doesn't use/include the beta repo if it's currently required.

> mount: /tmp: mount point not mounted or bad option.

Ah I think that's coming from the `dietpi-set_swapfile` script, right? It adjusts the tmpfs size to match 50% of available memory but it expects it to be a tmpfs already which seems to be then not true on this Armbian image. You can try to solve it by mounting the tmpfs first: `mount -t tmpfs none /tmp`
Makes sense to do that within the script as well if `findmnt /tmp` doesn't give a result.

MichaIng, 2020-08-02:
DietPi-PREP fix: https://github.com/MichaIng/DietPi/commit/401a70f89c6450586b72aca238e6512b9cb969f9
but larger dietpi-set_swapfile enhancement which includes skipping the remount attempt if /tmp is not a mount: https://github.com/MichaIng/DietPi/commit/646eba8bd9cb1bf3923164ec55a7a0cb051625b8

goney3, 2020-08-05:
I just received my Nanopi Neo3 and I am excited to see DietPi being worked on to run on it. Thank you guys so much for looking into this. I really appreciate it as I am a noob.

StephanStS, 2020-08-10:
@MichaIng: Executing `mount -t tmpfs none /tmp` manually in the subshell and doing a retry of the DietPi-PREP `mount -o remount /tmp` worked. I used dev branch of DietPi-PREP. 
Should there have been a fix via 401a70f which should avoid the mount problem?

See how it worked:
```
[ INFO ] DietPi-PREP | Removing swapfile from image
[ SUB1 ] DietPi-Set_swapfile > Applying 0 /var/swap
[ INFO ] DietPi-Set_swapfile | Disabling and deleting all existing swap files
[  OK  ] DietPi-Set_swapfile | swapoff -a
[  OK  ] DietPi-Set_swapfile | Setting in /boot/dietpi.txt adjusted: AUTO_SETUP_SWAPFILE_SIZE=0
[  OK  ] DietPi-Set_swapfile | Desired setting in /boot/dietpi.txt was already set: AUTO_SETUP_SWAPFILE_LOCATION=/var/swap
[ INFO ] DietPi-Set_swapfile | Setting /tmp tmpfs size: 992 MiB
[FAILED] DietPi-Set_swapfile | mount -o remount /tmp
root@nanopineo3:/tmp/DietPi-Set_swapfile# mount -t tmpfs none /tmp
root@nanopineo3:/tmp/DietPi-Set_swapfile# exit
[  OK  ] DietPi-Set_swapfile | mount -o remount /tmp
[  OK  ] DietPi-PREP | Setting in /boot/dietpi.txt adjusted: AUTO_SETUP_SWAPFILE_SIZE=1
```

StephanStS, 2020-08-10:
@goney3: I have an image for the NEO3 ready. Actually it is not downloadable via the dietpi download page. 
I could support you with a download link to my cloud. In this case please post here your email address so that I can send you the download link via email. Afterwards delete the email address posting to not keep your email address within this post.

goney3, 2020-08-10:
Sure, it's just my username at Yahoo, nothing fancy ;)

StephanStS, 2020-08-10:
> 
> 
> Sure, it's just my username at Yahoo, nothing fancy ;)

I just sent out a test email to you.

StephanStS, 2020-08-10:
@MichaIng: Do you want to add an unique ID for the NEO3 to the DietPi-PREP script?

Oleks1y, 2020-08-11:
> I have an image for the NEO3 ready. Actually it is not downloadable via the dietpi download page.

please share the link here

goney3, 2020-08-12:
DietPi installed on the NEO3 fine with StephanStS's image, I set up an LXDE desktop without issue. Though I finally noticed the NEO3 runs very hot. It idles at 63°C and a survey test pushed it to 85°C in mere seconds. I went to examine the heatsink that's included but the smooth white case seems impossible to open? I ordered some thermal paste in the meantime. Are there any tests you'd like me to run and report back on? A huge thank you to StephanStS for the image!

Oleks1y, 2020-08-12:
> DietPi installed on the NEO3 fine with StephanStS's image, I set up an LXDE desktop without issue. Though I finally noticed the NEO3 runs very hot. It idles at 63°C and a survey test pushed it to [...]

I've read that the reason for the high temperature is a Bluetooth service that burns CPU.

goney3, 2020-08-12:
There must be some truth to that, I set it to power saving mode ~400 MHz in dietpi-config and the temp was still around 61°C at idle 😬

I had no idea that it has Bluetooth though, I thought it was LAN only 🤔

MichaIng, 2020-08-12:
Hey guys, jep I'll implement NEO3 support tomorrow with new hardware ID etc. Sorry for delay, I made some excursus into web design, css, js etc to update and freshen our website. Now back on DietPi shell coding 😃.

Oleks1y, 2020-08-12:
Awesome news!!!!

MichaIng, 2020-08-13:
NEO3 ID/detection added: #3713
Not sure about use cases where GPU support might be required, server-side (trans)coding or VNC with server-side rendered desktop? Is it even possible to start an Xserver instance or does it fail as no screen is attached, or RK3328 GPU disabled hardware/firmware/kernel-wise?

StephanStS, 2020-08-13:
@Oleks1y: Same as above: 
I could support you with a download link to my cloud. In this case please post here your email address so that I can send you the download link via email. Afterwards delete the email address posting to not keep your email address within this post.

@MichaIng: Actually generating an image, based on dev with NEO3 ID. On the contrary to the image generation processes before, no error occured during the image generation. I.e. your fixing helped. 👍

MichaIng, 2020-08-13:
Great. Please manually switch `dietpi.txt` > `DEV_GITBRANCH=master` so that systems are migrated to stable branch automatically once released.

In the past we did these migrations in steps via `dietpi-update` but, aside of patch overhead, then the previous branch update is in progress already and stable branch is only applied for the next available update. We could say "`It's testing, hence take care yourself to switch the branch, if wanted.`" but I'd like to keep migration from testing image to stable DietPi code as seamless as possible 🙂. Also this allows users to stay at beta/dev if chosen manually without having `dietpi-update` reverting this choice.

For easy image creation, probably in such cases it is best if `DietPi-PREP` applies master/stable branch automatically, regardless if prior branch choice, so that the branch choice only applies to the initial code base but does not affect the target branch of the final system? For testing of recent beta/dev commits (as well when new SBC image-related) this means that one needs to manually switch. Kinda minor trade-off contributors/devs vs end-user testers 😄.

StephanStS, 2020-08-13:
Regarding NEO3 temperature: I did not see any difference between BT enabled or disabled.

@goney3: I have unmounted the white plastic housing. I made some photos for this but I cannot attach them here. There are small noses in the housing parts where you can open it.
The heatsink is mounted with a thermal pad, so there is no need to use a thermal paste as long as you do not use a different heatsink.
@MichaIng: Can I attach photos (png, gif,...)?

Joulinar, 2020-08-13:
Usually it should be possible on GitHub. Just move the picture into the text field or use add picture button

StephanStS, 2020-08-13:
@MichaIng: OK, I changed the `dietpi.txt` and generated the .7z again. So it is now downloadable from my cloud.

@Joulinar: By pasting a screenshot (-> .png) the editor flashes "![Uploading grafik.png]" but ternimates with `Something went really wrong, and we can't process that file. Try again.`
I found the root cause: My Pihole blocked `github-production-user-asset-6210df.s3.amazonaws.com`. My fault. :-)

StephanStS, 2020-08-13:
@goney3: Here come some pics from my NEO3 opened.
[image]

[image]

[image]

[image]

goney3, 2020-08-13:
I've been throttling the CPU down to 408 MHz with a peak of 600 MHz in an effort to lower the temperature, but its hovering around 59 to 61 C at idle, this must be an issue with the RK3328

I used [Snowflake](https://github.com/subhra74/snowflake) to SSH and install optimized packages, I initially used [SmarTTY](https://sysprogs.com/SmarTTY/) but it didn't like it. 
I've installed LXDE and I use XRDP to connect through Windows Remote Desktop and select Xvnc at login

I apt installed the Arc-theme (Arc-Darker) and set a transparency for the panel. I then made a cheeky wallpaper to match the current state of the NEO3 lol

[image]

MichaIng, 2020-08-13:
Nice to see it in action! I hope our testing sample arrives soon.

Is it possible to start a local X session to attach to via VNC? I.e. `startx`?

goney3, 2020-08-16:
I'd need to read up on startx to try and do what you are suggesting Michalng. I unfortunately don't know enough yet 😭

MichaIng, 2020-08-17:
It's the command that starts the Xserver with whichever default session has been applied. If you installed a desktop (which is done together with the VNC server) then it will start the desktop. Since it cannot run on a pseudo terminal like SSH session, it will be started on the first "real" TTY, i.e. `/dev/tty1`. But what I am not sure about is if the Xserver requires a screen/monitor being attached to this TTY or if this is solved via some internal framebuffer that can be used/manipulated by the Xserver. Then one would be able to start a local X/desktop session and attach to it via `x11vnc` to use it as shared desktop.

So a simple test is to call `startx` from the SSH session and see what happens. Either if fails with an error or it successfully initiates an X session in foreground (you only see console output, no desktop of course) that one can then interrupt via ctrl+c.

Oleks1y, 2020-08-18:
I've got a parcel with neo3 and installed armbian. 
the same hot processor here as well. 
immediately after the start is 85C 

[image]

StephanStS, 2020-08-19:
My NEO also started idle at 65 degree. I then added an additional small heatsink which led to about 52 degree idle temperature.
In my view the standard heatsink is too small, with this hot temperature the lifetime of the system degrades. As a rule of thumb it is said that increasing the chip temperature by 10 degrees will give half of the lifetime. 
In your case: Running the system at 55°C instead of 85°C will give a 8 x longer lifetime of the NEO3.

The plastic housing is too small to use a larger heatsink and also to use a fan. So we have to consider to use a different heatsink and forget the plastic housing. 
Or easier: Add additional heatsinks on top of the actual one (I know that this will give a poorer thermal transmission, but it helps: Use one heatsink with cooling fins upside down going into the actual heatsink cooling fins and put a second additional heatsink on top with cooling fins showing up). The acutal heatsink is mounted very elegantly with screws which I don't want to loose.

MichaIng, 2020-08-19:
Note that Armbian kernel mostly raises clocks to hardware maximum by default. The official manufacturer kernel/images often ship with lower clocks, hence lower temperatures. I see clocks officially range from 400 MHz to 1296 MHz, what is yours? `cpu`
But I see that you guys get high temperatures with manual throttling as well, so finally it seems this larger SoC on a tiny form factor that simply requires a heat sink at least.

Mine should arrive on Friday btw 😃.

Oleks1y, 2020-08-25:
> Okay I give it up. @StephanStS @Oleks1y @goney3 is haveged failing the same way for you? `journalctl -u haveged`

i have the following output

root@nanopineo3:~# journalctl -u haveged
-- Logs begin at Tue 2020-08-25 23:53:00 CEST, end at Wed 2020-08-26 00:45:01 CEST. --
-- No entries --

goney3, 2020-08-25:
[image]

MichaIng, 2020-08-26:
Haveged fixed with: https://github.com/MichaIng/DietPi/commit/87ef9a48d138932381e3cf06ed691e59a29b64e2
On DietPi-PREP for new images as well.

StephanStS, 2020-08-28:
@MichaIng: My proposal is to close this issue and also to close https://feathub.com/MichaIng/DietPi/+172 "Image | NanoPi NEO3".

MichaIng, 2020-08-28:
Jep, I'll add the NEO3 image to our download page later, then close it. Often I keep issues open until the related milestone has been reached, v6.32 release in this case, so during beta phase tester feel more convenient to still report test results, bugs, requests or such, compared to when doing this on a closed issue.

MichaIng, 2020-08-29:
I added NanoPi NEO3, R2S and K2 to the download page now, marked as beta until we have a few more reported users. I mark this as closed as suggested.

closed 2020-08-29 by MichaIng

Oleks1y, 2020-09-19:
[image]

Oleks1y, 2020-10-19:
Nice reading 
https://www.cnx-software.com/2020/10/18/nanopi-r2s-nanopi-neo3-tested-with-armbian-thermal-test-ethernet-and-usb-performance/