MichaIng/DietPi #3695: xserver-xorg-core not installable on DietPi for RPi

Max-Set, 2020-07-25:
Hello, I got the following issue while installing Kodi. Beside a possible solution, what is the reason for this issue?

#### Details:
- Date           | Sat Jul 25 08:21:50 CEST 2020
- Bug report     | aca3b446-4787-478f-b3a8-fe1fd00396b1
- DietPi version | v6.31.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq --allow-change-held-packages install xserver-xorg xinit xcompmgr xterm dbus-x11 xfonts-base x11-xserver-utils x11-utils libgl1-mesa-dri mesa-utils mesa-utils-extra
- Exit code      | 100
- Software title | Kodi from DietPi-Software
#### Steps to reproduce:

1. Trying to install Kodi from dietpi-software
2. retry it
#### Expected behaviour:

- The installation of xserver-xorg-core should be possible
#### Actual behaviour:

- The first time I tried to install Kodi from dietpi-software there was this error:
```
grave bugs of xserver-xorg-core (-> 2:1.20.4-1+rpt2) <Outstanding>
 b1 - #868068 - [xserver-xorg-core] Upgrade from 2:1.16.4-1 to 2:1.16.4-1+deb8u1 loses keyboard and mouse
Summary:
 xserver-xorg-core(1 bug)
xserver-xorg-core pinned by adding Pin preferences in /etc/apt/preferences.d/apt-listbugs. Restart APT session enable
```
#### Extra details:

- ...
#### Additional logs:
```
E: Unable to correct problems, you have held broken packages.
```

Sincerely

Max

Joulinar, 2020-07-25:
Hi,

many thanks for your report. Pls can you try following to check if there are pending apt package installations. 

```
apt update
apt upgrade
```

MichaIng, 2020-07-25:
Many thanks for your report. Please do:
```
apt purge apt-listbugs
```
It tells your about a bug that was reported on Debian Jessie for a particular build and you are two Debian versions further on Buster now. Holding the install for such a completely outdated report that was simply forgotten, doesn't make sense. There are (sadly) plenty of aged unreviewed und unverified bug reports on many packages, hence apt-listbugs will halt your installs all the time for no reason. Furthermore we use all available flags to make the APT install unattended. If apt-listbugs doesn't listen on any of then it is not compatible with DietPi scripts or any scripts that do automated DEB package installs 😞.

Max-Set, 2020-07-25:
Thank you both for your quick reply, I really appreciate it.

Sorry for the german output from the console, I hope it is understandable.

As suggested I run:
```
apt update
apt upgrade
```
but there where no pending installations or updates. After that I run
```
apt purge apt-listbugs
```
and
```
dietpi@DietPi:~$ sudo apt update
OK:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
OK:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
OK:3 https://archive.raspberrypi.org/debian buster InRelease
Paketlisten werden gelesen... Fertig                        
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Alle Pakete sind aktuell.
dietpi@DietPi:~$ sudo apt upgrade
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Paketaktualisierung (Upgrade) wird berechnet... Fertig
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:
  libruby2.5 libyaml-0-2 rake ruby ruby-debian ruby-did-you-mean ruby-gettext ruby-locale ruby-minitest ruby-net-telnet ruby-power-assert ruby-soap4r ruby-test-unit ruby-text ruby-unicode
  ruby-xmlparser ruby-xmlrpc ruby2.5 rubygems-integration
Verwenden Sie »sudo apt autoremove«, um sie zu entfernen.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
```
I also did
```
apt install -f
```
but there was no issue there.

Nonetheless the installation of Kodi did fail again.

If you need further information, I will be happy to help you.

Could it be possible that now an old installation fail of wireguard is causing this trouble?

Thank you very much in advance.

Joulinar, 2020-07-25:
Hehe we understand German very well 😉 

Did you followed the instructions of @MichaIng ?

`apt purge apt-listbugs`

Max-Set, 2020-07-25:
Yes, please see my last comment.

Joulinar, 2020-07-25:
`apt install xserver-xorg-core` is still failing?

Max-Set, 2020-07-25:
I've tried now to install xserver-xorg and xserver-xorg-core and got this:
```
dietpi@DietPi:~$ sudo apt install xserver-xorg
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 xserver-xorg : Hängt ab von: xserver-xorg-core (>= 2:1.17.2-2) ist aber nicht installierbar
                Hängt ab von: xserver-xorg-video-all soll aber nicht installiert werden oder
                               xorg-driver-video
                Hängt ab von: xserver-xorg-input-all soll aber nicht installiert werden oder
                               xorg-driver-input
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.
dietpi@DietPi:~$ sudo apt install xserver-xorg-core
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Paket xserver-xorg-core ist nicht verfügbar, wird aber von einem anderen Paket
referenziert. Das kann heißen, dass das Paket fehlt, dass es abgelöst
wurde oder nur aus einer anderen Quelle verfügbar ist.
Doch die folgenden Pakete ersetzen es:
  xserver-xorg-legacy xserver-common xserver-xorg-input-evdev

E: Für Paket »xserver-xorg-core« existiert kein Installationskandidat.
```

Max-Set, 2020-07-25:
Forget it…I did run
```
apt purge apt-listbugs
```
but the file /etc/apt/preferences.d/apt-listbugs was not deleted so I deleted it manually and now it's working. Sorry for the mistake on my part. The installation went smoothly.

Thank you very much for your quick reply,
have a nice weekend.

Max-Set, 2020-07-25:
Sorry for the inconvenience, but at the end of the installation routine there was a question asking to apply a recommended gpu memory split with the hint to answer with yes if I am unsure. So I chose yes and now my Pi is not booting. I've plugged the micro sd card into my laptop and it shows up, so could there be a problem with the gpu configuration and how can I change it or could it even happen that my Pi is dead?

Joulinar, 2020-07-25:
Do you have a screen attached to your RPi? Any error message showing during boot?

Max-Set, 2020-07-25:
Yes, per HDMI but there is nothing to see. Screen is completely empty, i.e. no console or anything. The TV screen doesn't even light up like it does when there is a device connected.

Joulinar, 2020-07-25:
You can have a look to following forum post on Raspberry Board. 

https://www.raspberrypi.org/forums/viewtopic.php?t=58151

As well you could try to start without SD Card. What is the blinking code of the ACT LED??
Or use a different SD card if available to check if this is working than.

MichaIng, 2020-07-25:
> but the file /etc/apt/preferences.d/apt-listbugs was not deleted so I deleted it manually

Nasty, yeah so it seems to halt all packages via preferences, prints the bugs, and then allows them one-by-one after interactive verification. Since the preferences have been loaded already, no chance to proceed in the same call. There are much better methods, like how `apt-listchanges` does it: It halts the install as well to show the packages changelog, but allows to hit `y` to continue directly. Also it respects APT options `-q/-y` and `DEBIAN_FRONTEND=noninteractive` to allow non-interactive proceeding.
_________
About the boot issue, did you assure that it is related to the GPU memory split? Because there are clear ranges for allowed values that we respect and every value out of the range should simply be ignored and never lead to boot issues. I suspect instead that it is due to some file corruption in relation to the prior CPU bug.

Please try to check the SD cards FAT partition for errors on the external system (Windows can check and repair it, just take care that you do not follow its nasty error message for format the ext4 partition that it cannot read 😄) and you can of course remove/comment the memory split there as well (the three `gpu_mem_*` lines in `config.txt`) there as well, however those should really not play a role.

Max-Set, 2020-07-25:
I've tried the things you've mentioned:

1. Commented the three `gpu_mem_*` lines in `config.txt` out -> not booting, red LED lights up but is not blinking.
2. Downloaded and installed the latest DietPi-version on a new card -> working
    1. Installed Kodi -> working
3. Checked the old card:
```
$ sudo /sbin/fsck -V -C -r /dev/mmcblk0p1
fsck from util-linux 2.33.1
[/usr/sbin/fsck.vfat (1) -- /dev/mmcblk0p1] fsck.vfat /dev/mmcblk0p1 
fsck.fat 4.1 (2017-01-24)
/dev/mmcblk0p1: 304 files, 111785/516191 clusters
/dev/mmcblk0p1: status 0, rss 8224, real 0.138498, user 0.046935, sys 0.016762
$ sudo /sbin/fsck -V -C -r /dev/mmcblk0p2
fsck from util-linux 2.33.1
[/usr/sbin/fsck.ext4 (1) -- /dev/mmcblk0p2] fsck.ext4 -C0 /dev/mmcblk0p2 
e2fsck 1.44.5 (15-Dec-2018)
rootfs: clean, 211662/1799520 files, 4744813/7504896 blocks
/dev/mmcblk0p2: status 0, rss 3220, real 0.027103, user 0.010458, sys 0.005229
```
4. Activated the three `gpu_mem_*` lines on the old card -> not booting

MichaIng, 2020-07-25:
Ah another possible reason: #3690
Does the `config.txt` contain `arm_freq_min` (without leading `#`)? If so, try to comment or remove it to fix boot.

Max-Set, 2020-07-25:
I really appreciate your detailed support, but `arm_freq_min` in `config.txt` is already commented out.

Joulinar, 2020-07-25:
Just for my personal understanding as I'm not sure if I understood right @Max-Set 

You did a `sudo /sbin/fsck -V -C -r /dev/mmcblk0p1` on your old SD card, as stated on your post https://github.com/MichaIng/DietPi/issues/3695#issuecomment-663878054

But how did you do that if the SD Card is not booting? Because `/dev/mmcblk0p1` is the current SD card within your system, that you are using to perform the check. Isn't it? If you plug the SD Card into a nother system (via USB adapter), it should be something like /dev/sda1. Maybe I have been to long at the sun today. So pls correct me if I'm totally wrong.

Max-Set, 2020-07-25:
I've plugged my old SD card into the card slot in my laptop and there it showed up as `/dev/mmcblk0p1` and `/dev/mmcblk0p2`. Sorry for the confusion. I've now plugged it in via a external card reader and now it shows up as `/dev/sdc1` and `/dev/sdc2`.

Joulinar, 2020-07-25:
Ah ok, now it makes sense. Even for me 😆

MichaIng, 2020-07-26:
Hmm, no direct idea now. Of course you can compare the content of `/boot` (`/dev/mmcblk0p1`) for differences or missing files (content of `/boot/dietpi` is not relevant for actual booting), also comparing `config.txt` and `cmdline.txt`. A serial console would provide insights into boot messages before actual video output would happen, but likely not worth it to order one for this case, maybe for the future or when running a larger amount of SBCs, especially headless ones.

Also to check whether there _was_ file or file system corruption you can check for `FSCK00*` prefixed files in the boot/FAT partition and `/lost+found/` for content on the root/ext4 partition. Both is where `fsck` puts files/inodes which lost their file system link. This might habe happened during an earlier boot already so that now there are no errors found, but instead files are missing.

Max-Set, 2020-07-26:
I have found the issue: The three `gpu_mem_*` lines in `config.txt` were the culprit. Commenting them out does *not* solve the problem like mentioned in https://github.com/MichaIng/DietPi/issues/3695#issuecomment-663878054 but overwriting them with old values from a backup brings the pi to life. I have done that twice to ensure that I did not something additional.

This is fine but does lead to the question why those three lines are working fine on a fresh install.

Are you interested in further investigation?

MichaIng, 2020-07-26:
Then I believe the `start.elf` and or `fixup.dat` file was broken, or something in device tree they are loading. When using 16 MiB GPU memory, `start_cd.elf` (cut-down) and `fixup_cd.dat` are loaded instead, which do not load all GPU features since 16 MiB are not sufficient to run those anyway. Kodi and GPU acceleration can also not run with this, so it is totally expected that it fails.

When setting at least 32 MiB GPU memory (76 MiB is the default, 64 MiB on RPi 1+Zero), then the regular startup files are loaded, which enables additional GPU features, required for Kodi to run.

> I think I will start from scratch since we know that it is working in that case and it takes way more time for you and me to solve this with presumably little benefit.

Agree 👍.

closed 2020-08-26 by Max-Set