MichaIng/DietPi #3696: Bugs after update to v6.31.2

Thundo54, 2020-07-25:
### Creating a bug report/issue

#### Required Information
- DietPi version |
`G_DIETPI_VERSION_CORE=6`
`G_DIETPI_VERSION_SUB=31`
`G_DIETPI_VERSION_RC=2`
`G_GITBRANCH='master'`
`G_GITOWNER='MichaIng'`

- Distro version | `10.4 (buster)`
- Kernel version | `Linux 5.4.51-v7+ armv7l`
- SBC model | `RPi 3 Model B+ (armv7l)`
- Power supply used | `5.1V 2.5A`
- SDcard used |  `Goodram M1AA 16GB`

#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated? - Updated
- Can this issue be replicated on a fresh installation of DietPi? - Probably not

#### Steps to reproduce
- Update to DietPi v6.31.2

#### Expected behaviour
- Changeable CPU min_freq and max_freq
- Changeable CPU governor

#### Actual behaviour
- Cannot change cpu freq (min and max) - cpu freq changed automatically (for min - 500Hz to 560Hz and then 700Hz)
- Cannot change cpu governor - after change it to powersave cannot back to ondemand

Joulinar, 2020-07-25:
Hi,

many thanks for your report. Maybe it's related to Raspberry OS Kernel issue. I guess @MichaIng disabled idle frequency setting on all RPi models for the time being. #3690

MichaIng, 2020-07-25:
If you have MOTD in banner enabled, then I hacked the following change inside:
- `dietpi-config` > `Performance Options` > `ARM idle frequency` is not available anymore.
- The `/boot/config.txt` setting `arm_freq_min` that is set by this option is commented if present.

After a reboot this should lead to a stable system. If you had this setting set and do not have MOTD enabled, then you'll face several hang and crash and other major bugs during system operation and you should remove the setting manually from your `config.txt` and reboot to fix them. It might be possible that you cannot change something due to the bug itself. In this case I recommend to poweroff and remove the setting from an external system. Since it is located on the FAT partition, you can do that from Windows as well.

A VERY dangerous bug and I am shocked that this did not lead to a hold on that kernel packages. There even has been another upgrade yesterday but at least no hint that this one has been fixed 😞.

Thundo54, 2020-07-25:
Do you mean that my RPi maybe will be bricked by this bug?

in `/boot/config.txt` I have:
```

#-------Overclock-------
temp_limit=65
initial_turbo=20

#over_voltage=0
#arm_freq=1400
#core_freq=400
#sdram_freq=500

#arm_freq_min=500
#core_freq_min=250
#sdram_freq_min=400
```
Should I unhash these lines?

MichaIng, 2020-07-25:
No it will not be bricked, but the way the bug affects system operation may have lead to file corruption, hence `fsck` may be required to fix them. `#arm_freq_min=500` is commented, so that is pretty fine. Please leave it like that for now, if you did not yet reboot after trying to change CPU settings, please do now. Then you should be able to operate as before, even changing CPU settings, aside of the minimal/idle CPU frequency, which must stay at default 700 MHz for now. Not a big issue since there should be marginal power/head differences only between 500 and 700 MHz 😉.

Just to be sure, run `fsck /dev/mmcblk0p1` to check the boot/FAT partition. You can do the same for the root partition, which is however only possible on reboot, by running `> /forcefsck` and rebooting. Depending on SD card size and speed it might take a while until you can login. LEDs should show you the activity.
__________
Once the bug has been fixed upstream, I'll inform you via those GitHub issues and a MOTD as well, so you know when you're able to safely reduce your idle frequencies again.

Thundo54, 2020-07-25:
No marginal issues was found while running `fsck /dev/mmcblk0p1`, so only way to fixed is to wait until update 

`fsck /dev/mmcblk0p1` return:

```
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
```
```
There are differences between boot sector and its backup.
This is mostly harmless. Differences: (offset:original/backup)
65:01/00
```
```
Reclaimed 270 unused clusters (138240 bytes).
Free cluster summary wrong (387002 vs. really 387272)
```

MichaIng, 2020-07-25:
> 65:01/00

I guess this is the dirty bit, but not sure.

> Reclaimed 270 unused clusters

Please fix those if not yet done, or do alltogether via `dosfsck -a /dev/mmcblk0p1`.
Then check `ls -al /boot` if it contains any `FCK*` or similar named files. Possibly one lost its file system link. If there is one, check its content which should reveal where it belongs to. Then you can rename/move it back in place. While I ran into this bug, it was `/boot/dietpi/.hw_model` which was lost, two times, I guess the bug made the script that creates it hang somehow. Nothing important, it would be recreated (or was already) anyway 😉.

Ah and btw `/lost+found` would contain files from rootfs that lost their fs link/reference.

Thundo54, 2020-07-25:
After doing `fsck /dev/mmcblk0p1` and then checking `ls -al /boot` those files I found:
```
-rwxr-xr-x  1 root root    62464 Jan  1  1980  FSCK0000.REC
-rwxr-xr-x  1 root root    62464 Jan  1  1980  FSCK0001.REC
-rwxr-xr-x  1 root root     3584 Jan  1  1980  FSCK0002.REC
-rwxr-xr-x  1 root root     3584 Jan  1  1980  FSCK0003.REC
-rwxr-xr-x  1 root root    12800 Jan  1  1980  FSCK0004.REC
-rwxr-xr-x  1 root root   529408 Jan  1  1980  FSCK0005.REC
-rwxr-xr-x  1 root root   529408 Jan  1  1980  FSCK0006.REC
-rwxr-xr-x  1 root root   529408 Jan  1  1980  FSCK0007.REC
-rwxr-xr-x  1 root root     6144 Jan  1  1980  FSCK0008.REC
-rwxr-xr-x  1 root root   529408 Jan  1  1980  FSCK0009.REC
-rwxr-xr-x  1 root root     6144 Jan  1  1980  FSCK0010.REC
-rwxr-xr-x  1 root root   529408 Jan  1  1980  FSCK0011.REC
-rwxr-xr-x  1 root root     6144 Jan  1  1980  FSCK0012.REC
-rwxr-xr-x  1 root root   529408 Jan  1  1980  FSCK0013.REC
-rwxr-xr-x  1 root root     6144 Jan  1  1980  FSCK0014.REC
-rwxr-xr-x  1 root root   529408 Jan  1  1980  FSCK0015.REC
-rwxr-xr-x  1 root root     6144 Jan  1  1980  FSCK0016.REC
-rwxr-xr-x  1 root root   529408 Jan  1  1980  FSCK0017.REC
-rwxr-xr-x  1 root root     6144 Jan  1  1980  FSCK0018.REC
-rwxr-xr-x  1 root root   529408 Jan  1  1980  FSCK0019.REC
-rwxr-xr-x  1 root root     6144 Jan  1  1980  FSCK0020.REC
-rwxr-xr-x  1 root root   529408 Jan  1  1980  FSCK0021.REC
-rwxr-xr-x  1 root root     6144 Jan  1  1980  FSCK0022.REC
-rwxr-xr-x  1 root root   529408 Jan  1  1980  FSCK0023.REC
-rwxr-xr-x  1 root root     6144 Jan  1  1980  FSCK0024.REC
-rwxr-xr-x  1 root root   529408 Jan  1  1980  FSCK0025.REC
-rwxr-xr-x  1 root root     6144 Jan  1  1980  FSCK0026.REC
-rwxr-xr-x  1 root root   529408 Jan  1  1980  FSCK0027.REC
-rwxr-xr-x  1 root root     6144 Jan  1  1980  FSCK0028.REC
-rwxr-xr-x  1 root root   529408 Jan  1  1980  FSCK0029.REC
-rwxr-xr-x  1 root root     6144 Jan  1  1980  FSCK0030.REC
-rwxr-xr-x  1 root root   536576 Jan  1  1980  FSCK0031.REC
-rwxr-xr-x  1 root root     1024 Jan  1  1980  FSCK0032.REC
-rwxr-xr-x  1 root root    12800 Jan  1  1980  FSCK0033.REC
-rwxr-xr-x  1 root root   138240 Jan  1  1980  FSCK0034.REC
-rwxr-xr-x  1 root root   138240 Jan  1  1980  FSCK0035.REC
-rwxr-xr-x  1 root root   138240 Jan  1  1980  FSCK0036.REC
-rwxr-xr-x  1 root root   138240 Jan  1  1980  FSCK0037.REC
```
They're contains these 'FCK*' phrase 
I think they're not that ones we're looking for

Thundo54, 2020-07-25:
btw I checked the ` dmesg -l err,crit,alert,emerg`

```
[    1.553275] vc_vchi_sm_init: failed to open VCHI service (-1)
[    1.553279] [vc_sm_connected_init]: failed to initialize shared memory service
[    4.112193] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[    4.112196] [vc_sm_connected_init]: failed to initialize shared memory service
[    4.463658] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[    4.571960] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.787147] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.803023] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[    6.208853] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[28320.656854] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[38240.159493] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
```

Joulinar, 2020-07-25:
Messages should be fine. VCHI can be ignored. Others are about WiFi module I guess.

MichaIng, 2020-07-25:
> I think they're not that ones we're looking for

Those are exactly what we are looking for 😉. All those files have lost their file system link once, so we should go through all of them either, or reinstall RPi kernel to assure all required files are in place. Let me compare the file sizes, probably I can find those quickly then. Luckily many are just copies.

Thundo54, 2020-07-25:
What can I do tho?
for example the last one `FSCK0037.REC` contains:
[FSCK0037.zip](https://github.com/MichaIng/DietPi/files/4975866/FSCK0037.zip)

This file contains at least 4788 lines, wow 😉.

Thundo54, 2020-07-25:
/boot/FSCK0000.REC
```
cat -n /boot/FSCK0034.REC | head -10     1      #!/bin/bash
     2  {
     3          #////////////////////////////////////
     4          # DietPi Function:
     5          # - Enables control and applies settings for specific hardware.
     6          #
     7          #////////////////////////////////////
     8          # Created by Daniel Knight / daniel.knight@dietpi.com / dietpi.com
     9          #
    10          #////////////////////////////////////
```
/boot/FSCK0002.REC
```
     1  #!/bin/bash
     2  {
     3          #////////////////////////////////////
     4          # DietPi Boot Script
     5          #
     6          #////////////////////////////////////
     7          # Created by Daniel Knight / daniel.knight@dietpi.com / dietpi.com
     8          #
     9          #////////////////////////////////////
    10          #
```
/boot/FSCK0004.REC
```
     1  # IMPORTANT:
     2  # - This is intended for advanced users, unless you know what you are doing, do not edit this file. Please use the DietPi programs instead.
     3  # - Do not remove uncommented lines, as the items are scraped by DietPi programs, on demand.
     4
     5  #------------------------------------------------------------------------------------------------------
     6  # D I E T - P I
     7  # DietPi-Automation settings, applied on first boot of DietPi only, ONCE!
     8  #------------------------------------------------------------------------------------------------------
     9  ##### Language/Regional Options #####
    10  # Locale: eg: "en_GB.UTF-8" / "de_DE.UTF-8" | One entry and UTF-8 ONLY!
```
/boot/FSCK0005.REC
```
     1  #!/bin/bash
     2  {
     3          #////////////////////////////////////
     4          # DietPi Software
     5          #
     6          #////////////////////////////////////
     7          # Created by Daniel Knight / daniel.knight@dietpi.com / dietpi.com
     8          #
     9          #////////////////////////////////////
    10          #
```
/boot/FSCK0008.REC
```
     1  #!/bin/bash
     2  {
     3          #////////////////////////////////////
     4          # DietPi-logclear Script
     5          #
     6          #////////////////////////////////////
     7          # Created by Daniel Knight / daniel.knight@dietpi.com / dietpi.com
     8          #
     9          #////////////////////////////////////
    10          # Info:
```
/boot/FSCK0031.REC
```
     1  #!/bin/bash
     2  {
     3          #////////////////////////////////////
     4          # DietPi Software
     5          #
     6          #////////////////////////////////////
     7          # Created by Daniel Knight / daniel.knight@dietpi.com / dietpi.com
     8          #
     9          #////////////////////////////////////
    10          #
```
/boot/FSCK0032.REC
```
     1  # DietPi-Services Include/Exclude configuration
     2
     3  # Include custom service (Use '+ servicename' without the comments to enable DietPi control of that service)
     4  #       Once completed, for DietPi to control the service, please run the following command, without quotes (')
     5  #       'dietpi-services dietpi_controlled'
     6  #+ myservice1
     7  #+ myservice2
     8
     9  # Exclude DietPi from controlling known services (Use '- servicename' without the comments to disable DietPi control for that service)
    10  #       The service will be in disabled form, and, you can start and stop it manually
```
/boot/FSCK0033.REC
```
     1  #!/bin/bash
     2  {
     3          #////////////////////////////////////
     4          # DietPi Function:
     5          # - benchmark
     6          #
     7          #////////////////////////////////////
     8          # Created by Daniel Knight / daniel.knight@dietpi.com / dietpi.com
     9          #
    10          #////////////////////////////////////
```
/boot/FSCK0034.REC
```
     1  #!/bin/bash
     2  {
     3          #////////////////////////////////////
     4          # DietPi-Config Script
     5          #
     6          #////////////////////////////////////
     7          # Created by Daniel Knight / daniel.knight@dietpi.com / dietpi.com
     8          #
     9          #////////////////////////////////////
    10          #
```

MichaIng, 2020-07-25:
Lets automate fixing this:
```
[[ -f '/boot/dietpi/dietpi-config' ]] || mv /boot/FSCK0034.REC /boot/dietpi/dietpi-config
rm -f FSCK003{4,5,6,7}.REC
[[ -f '/boot/dietpi/func/dietpi-benchmark' ]] || mv /boot/FSCK0034.REC /boot/dietpi/func/dietpi-benchmark
rm -f FSCK0033.REC
[[ -f '/boot/dietpi/.dietpi-services_include_exclude' ]] || mv /boot/FSCK0032.REC /boot/dietpi/.dietpi-services_include_exclude
rm -f /boot/FSCK0032.REC
[[ -f '/boot/dietpi/dietpi-software' ]] || mv /boot/FSCK0031.REC /boot/dietpi/dietpi-software
rm -f FSCK00{05,06,07,09,11,13,15,17,19,21,23,25,27,29,31}.REC
[[ -f '/boot/dietpi/func/dietpi-logclear' ]] || mv /boot/FSCK0008.REC /boot/dietpi/func/dietpi-logclear
rm -f FSCK00{08,10,12,14,16,18,20,22,24,26,28}.REC
[[ -f '/boot/dietpi.txt' ]] || mv /boot/FSCK0004.REC /boot/dietpi.txt
rm -f /boot/FSCK0004.REC
[[ -f '/boot/dietpi/boot' ]] || mv /boot/FSCK0002.REC /boot/dietpi/boot
rm -f /boot/FSCK000{2,3}.REC
[[ -f '/boot/dietpi/func/dietpi-set_hardware' ]] || mv /boot/FSCK0000.REC /boot/dietpi/func/dietpi-set_hardware
rm -f FSCK000{0,1}.REC
```

Thundo54, 2020-07-25:
Okey I done this and then reboot the RPi - nothing important show to me,
what could I do next?

MichaIng, 2020-07-25:
That should be it, all fine as long as you don't face any issues.

Thundo54, 2020-07-26:
Okey, I'll close the issue and will be waiting for new DietPi 64Bit ver 😄

closed 2020-07-26 by Thundo54