MichaIng/DietPi #3697: fail2ban not banning IP adresses and not noticing failed logins over SSH

Darwolia, 2020-07-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=31
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `buster`
- Kernel version | `Linux DietPi 5.4.51-v7+`
- SBC model | `RPi 3 Model B (armv7l)`
- Power supply used | 5V 2.5A
- SDcard used | SanDisk Ultra 16 Gb

#### Additional Information (if applicable)
- Software title | fail2ban
- Was the software title installed freshly or updated/migrated? Installed Freshly
- Can this issue be replicated on a fresh installation of DietPi? Unkown - will try once I have physical access to the SBC again

#### Steps to reproduce
1. Install fail2ban with the dietpi-software command
2. Deliberately fail the login with PuTTY (with root user and an extra added user - with a RSA key-pair)

#### Expected behaviour
- IP adress should be banned after three failed attempts (using the default jail)
- failed logins should be shown when using "fail2ban-client status sshd" command

#### Actual behaviour
- Can still login with PuTTY after many failed logins
- "fail2ban-client status sshd" does not show any failed attempts

#### Extra details
Installed software since fresh install:
- iptables
- openSHH (installed on first launch)
- python3
- pytonh3-pip
- chromium-browser

MichaIng, 2020-07-25:
Many thanks for your report.

How exactly did you trigger the failed login? In case of a key pair, if you e.g. use a passphrase and enter it wrong, this is a pre-authentication failure on the client only and the server will not recognise about it at all.

Please check `journalctl -u sshd` if you see any logs about the failed login attempts.

Darwolia, 2020-07-25:
Hello!

I did not know about entering a wrong passphrase beeing a failure on the clients side (I am using a passphrase)

I did some testing with root, an existing user, non existing user, a correct, wrong and no key-file (both with a passphrase) for login. In the sshd_config file I prohibited root login and password authentication.

Results:

User&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;root&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;existing&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;not existing
right key&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;not noticed&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;works&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;noticed
wrong key&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;not noticed&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;not noticed&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;noticed
no key&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;not noticed&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;not noticed&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;noticed

Please excuse the formatting.

I have entered the correct passphrase for each key everytime. Guess my lack of knowledge was the issue and this behaviour is intented?
Executing `journalctl -u sshd` afterwards shows no entries though. But I am not sure if that is a reason to keep this issue open.

-- Logs begin at Thu 2019-02-14 11:11:58 CET, end at Sat 2020-07-25 20:57:45 CEST. --
-- No entries --

MichaIng, 2020-07-25:
Ah it should be `journalctl -u ssh` as the service name is `ssh.service`. Hmm probably the default sshd jail lacks a pattern for failed key authentication. When you enter the right key, it should produce an authentication error log and hence a fail2ban trigger.

Darwolia, 2020-07-26:
For each failed login there are two lines generated:
` error: Received disconnect from IP adress port 58060:14: No supported authentication methods available [pre
auth]`
and
`Disconnected from authenticating user root IP adress port 58059 [preauth]`

The lines are always the same for root (prohibited by the sshd_config file) and my actual user, wether I use the right key file, the wrong one or none at all.

The only time it looks different is when I use a non existing username:

`error: Received disconnect from IP adress port 58067:14: No supported authentication methods available [pre
auth]`
and
`Disconnected from invalid user asd IP adress port 58067 [preauth]`

MichaIng, 2020-07-26:
This is how bots try to break in our server minutely:
```
Jul 26 11:20:11 dietpi.com sshd[2570408]: pam_unix(sshd:auth): check pass; user unknown
Jul 26 11:20:11 dietpi.com sshd[2570408]: pam_unix(sshd:auth): authentication failure; logname= uid=0 euid=0 tty=ssh ruser= rhost=XXX.YY.Z.UUU
Jul 26 11:20:15 dietpi.com sshd[2570408]: error: maximum authentication attempts exceeded for invalid user admin from 141.98.9.157 port 34347 ssh2 [preauth]
```
Not sure how PAM works here and why there is an initial "check pass" if the user does not even exist. The following error message from PAM and the one from the sshd outside of PAM (we allow only a single authentication attempt) are both tracked by fail2ban, hence count as two failure which we must consider when setting the allowed retries before adding a ban.

If the user does exist, the first log line is missing, the second has `user=<username>` appended and third has `invalid user` replaced by the actual user name, which matches your case.

When I use a wrong key, I only see the last log (either with invalid user or with user name) which basically makes sense. Since the key authentication is done within the SSH server only, nothing is directed to PAM until the key authentication succeeded. Password authentication instead is done against the UNIX user password, hence is forwarded to PAM directly. Only after key authentication succeeded, it is forwarded to PAM to apply OS-side checks, which gives me then `pam_unix(sshd:session): session opened for user <name> by (uid=X)`.

The PAM failure is caught by `/etc/fail2ban/filter.d/sshd.conf` regex:
```
^pam_unix\(sshd:auth\):\s+authentication failure;\s*logname=\S*\s*uid=\d*\s*euid=\d*\s*tty=\S*\s*ruser=<F-USER>\S*</F-USER>\s*rhost=<HOST>\s.*%(__suff)s$
```
The sshd error I get is covered by regex:
```
^(error: )?maximum authentication attempts exceeded for <F-USER>.*</F-USER> from <HOST>%(__on_port_opt)s(?: ssh\d*)?%(__suff)s$
```
We seem to always get the same sshd error which obviously overrides all other error messages that should be printed instead, although probably it is different when I prohibit password authentication for all accounts instead of just root. Regardless there is at least one error log with remove IP that is tracked and in case banned by fail2ban.

And now I see the reason why in your case it does not work. There is the regex:
```
^Received <F-MLFFORGET>disconnect</F-MLFFORGET> from <HOST>%(__on_port_opt)s:\s*14: No supported authentication methods available%(__suff)s$
```
But its part of the `mdre-extra` block and while I am no expert at those fail2ban filter, it looks quite clear that for this to be in effect one needs to set `mode = extra`.
Please open `/etc/fail2ban/fail.conf` and below `[sshd]` add `mode = extra`:
```
[sshd]
mode = extra
```
Hmm not sure if this mode makes sense by default. I guess it is seen as too harmless when someone does a single authentication attempt with no or wrong key while key authentication is strictly required, not sure.

Btw `[pre auth]` is actually `[preauth]` with a line break inside, right? Because the first one would not be covered by any regex `%(__suff)s` 😉.

Darwolia, 2020-07-26:
Yes, I copied the line breaks by accident. Adding `mode = extra` or even `mode = aggressive`  beneath `[sshd]` in the `/etc/fail2ban/jail.conf` does not change the behaviour for some reason. But it seems to be a problem with fail2ban itself and not DietPi?

MichaIng, 2020-07-26:
> But it seems to be a problem with fail2ban itself and not DietPi?

Found the issue: Our `jail.conf` does not pass the mode variable: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-software

Darwolia, 2020-07-26:
Alright, with the changes to the `jail.conf` and `sshd.conf` preauth errors are noticed with `mode = extra` and the IP address is eventually banned. I'll close this issue, thanks for the help!

closed 2020-07-26 by Darwolia