MichaIng/DietPi #3698: mirrors.netix.net - a certificate in the chain is expired, apt cant update.

jerryhopper, 2020-07-28:
### Creating a bug report/issue

Hey all,  a certificate error (mirrors.netix.net)  is preventing apt from updating dietpi!

a certificate in the chain is expired, as you can see here  https://www.sslshopper.com/ssl-checker.html#hostname=mirrors.netix.net

Joulinar, 2020-07-28:
Hi,

many thanks for your report. But this seems to be out of DietPi scope as DietPi is not hosting this website.

jerryhopper, 2020-07-28:
Ok.  but somehow a clean dietpi hits mirrors.netix.net - and spits out the error - and installation is stuck.
Is there any way to exclude this mirror?  

I tried editing /etc/hosts  and point that hostname to 0.0.0.0  - but while running update, it doesnt try a different mirror.

Joulinar, 2020-07-28:
I guess you are using a Raspberry Pi. As the DietPi base image for RPi is Raspberry OS, you are going to taget a Raspbian Repository Mirror 😉 

not sure if this is working but you can try to edit `/etc/apt/sources.list.d/raspi.list` and change `HTTPS` into `HTTP`

jerryhopper, 2020-07-28:
I actually have a NanoPiNeo2Black.  i have edited /etc/apr/sources.list.d/raspi.list  - but now its whining about a public key that is missing.

Maybe forcing apt to use insecure mirror for this mirrors.netix.net is the fastest way.
ill try inform peepz at raspbian about this faulty mirror.

MichaIng, 2020-07-28:
Accessing via browser https://mirrors.netix.net/raspbian/ shows that the certificate chain is fully intact. Is it probably only a wrong system time? In case try: `/boot/dietpi/func/run_ntpd 1`

Joulinar, 2020-07-28:
@jerryhopper 
this is fully at Raspbian side. They are managing the mirrors.

MichaIng, 2020-07-28:
Does this show anything? `ls -l /etc/ssl/certs/USERTrust*`
I get:
```
lrwxrwxrwx 1 root root 76 Dec 24  2017 /etc/ssl/certs/USERTrust_ECC_Certification_Authority.pem -> /usr/share/ca-certificates/mozilla/USERTrust_ECC_Certification_Authority.crt
lrwxrwxrwx 1 root root 76 Dec 24  2017 /etc/ssl/certs/USERTrust_RSA_Certification_Authority.pem -> /usr/share/ca-certificates/mozilla/USERTrust_RSA_Certification_Authority.crt
```

jerryhopper, 2020-07-28:
@MichaIng i get exactly what you have.  but like @Joulinar says, the problem is on the armbian side.

Best way is to have that mirror removed at the armbian side....  or hardcode the fix in a dietpi image/update. (which is rather nasty and temporary, and doesnt fix the existing dietpi images.)

Joulinar, 2020-07-28:
well It's quite a big mirror for a lot of destro's including Raspbian and Armbian. So it's not isolated to a single Destro.

http://mirrors.netix.net/

MichaIng, 2020-07-28:
Strange is that I can use this mirror very fine on my RPi and my browser shows that no cert of the chain is expired, but SSLShopper shows the expiry as posted by Joulinar above: https://www.sslshopper.com/ssl-checker.html#hostname=mirrors.netix.net

jerryhopper, 2020-07-28:
in a unintended install, when this specific repo is used - you dont get dietpi installed unless you apply my fix.
dietpi login keeps spawning the update, which obviously fails.

on my nanopineo2black with the dietpi image, apt specificly spits out the error about expired cert.

MichaIng, 2020-07-28:
But you setup this mirror first manually, don't you? Since our RPi images are shipped with raspbian.raspberrypi.org as Raspbian mirror.

jerryhopper, 2020-07-28:
i didnt edit anything.  a clean image straight from the site.

i checked, originally it is set to : apt.armbian.com

im using this image : https://github.com/MichaIng/DietPi/issues/3333 ( the admin edit )

MichaIng, 2020-07-28:
Ah your right, so its not about Raspbian (?) but about Armbian indeed. Jep that one is redirected indeed: https://apt.armbian.com

Joulinar, 2020-07-28:
but this should not make any difference if it's Armbian or Raspbian mirror. They all will land on `mirrors.netix.net` isn't it?

jerryhopper, 2020-07-28:
maybe its selection based on ip/geo ? 
Which is also weird, im in NL - from a NL ip.  there are closer mirrors.

MichaIng, 2020-07-28:
But raspbian.raspberrypi.org is not a mirror director but a dedicated server, so there one must set this mirror manually. apt.armbian.com is (at least now, not sure if this was before) a mirror director that is redirected to netix and sometimes also other mirrors, likely depending on current load and location.

jerryhopper, 2020-07-28:
Well.  in that case im totally clueless why i always get the netix mirror when installing on several devices. ( i have 3 devices, all same model - so same image )

ill just email netix about this, hoping they will fix the cert... which is clearly expired in may.   ( even digicert.com sees that, but says its valid - lol)

Joulinar, 2020-07-28:
well still strange, I was able to install small peace of software from NetIX without issues. 

```
root@DietPi3:/etc/apt# apt install lsof
Reading package lists... Done
Building dependency tree
Reading state information... Done
Suggested packages:
  perl
The following NEW packages will be installed:
  lsof
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 307 kB of archives.
After this operation, 447 kB of additional disk space will be used.
Get:1 https://mirrors.netix.net/raspbian/raspbian buster/main armhf lsof armhf 4.91+dfsg-1 [307 kB]
Fetched 307 kB in 1s (500 kB/s)
Selecting previously unselected package lsof.
(Reading database ... 22174 files and directories currently installed.)
Preparing to unpack .../lsof_4.91+dfsg-1_armhf.deb ...
Unpacking lsof (4.91+dfsg-1) ...
Setting up lsof (4.91+dfsg-1) ...
root@DietPi3:/etc/apt#
```

MichaIng, 2020-07-28:
> ill just email netix about this, hoping they will fix the cert... which is clearly expired in may. ( even digicert.com sees that, but says its valid - lol)

Good idea. However I am not sure that it is really outdated since in my case it all works very fine. Probably there is some caching issue or something else gets mixed up.
_______________
Another idea, at which version is the ca-certificates package? `dpkg -l ca-certificates`
Probably the CA certificate file on the system is outdated.

jerryhopper, 2020-07-28:
yet the ssl cert says its expired in may 2020...   In theory what you have, shouldnt happen with a invalid cert in the chain.

<pre>
root@DietPi:~# dpkg -l ca-certificates
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name            Version          Architecture Description
+++-===============-================-============-=================================
ii  ca-certificates 20200601~deb10u1 all          Common CA certificates
root@DietPi:~#
</pre>

Joulinar, 2020-07-28:
could it be that the `USERTrust RSA Certification Authority` certificate on the system would need to be changed?

https://www.tbs-certificates.co.uk/FAQ/en/USER-Trust-RSA-Certification-Authority.html

jerryhopper, 2020-07-28:
Yes - that is the one!   - but, do I need to install that?  naaah... right?    i assume they need to add that to the chain @ netix.net ?

jerryhopper, 2020-07-28:
netix replied & confirmed...   they will fix.

Did i just saved the internet?  :)

Joulinar, 2020-07-28:
let's say it this way: That's one small step for a man - one giant leap for mankind

MichaIng, 2020-07-28:
SSL Labs: => Expired: https://www.ssllabs.com/ssltest/analyze.html?d=mirrors.netix.net
But my Windows:
[image]

```
openssl s_client -showcerts -connect mirrors.netix.net:443 | openssl x509 -noout -dates
notBefore=Jan 17 00:00:00 2020 GMT
notAfter=Feb 15 23:59:59 2022 GMT
```
- But this is only the first certificate I guess. How to check the through the chain?

jerryhopper, 2020-07-28:
im using online services.

https://www.sslshopper.com/ssl-checker.html#hostname=mirrors.netix.net

https://www.digicert.com/help/

https://whatsmychaincert.com/?mirrors.netix.net

https://www.ssllabs.com/ssltest/analyze.html?d=mirrors.netix.net&s=87.121.121.2

MichaIng, 2020-07-28:
Now the here string verifies it:
```
2020-07-29 00:19:44 root@dietpi:~# openssl x509 -noout -dates <<< "-----BEGIN CERTIFICATE-----
> MIIFdzCCBF+gAwIBAgIQE+oocFv07O0MNmMJgGFDNjANBgkqhkiG9w0BAQwFADBv
> MQswCQYDVQQGEwJTRTEUMBIGA1UEChMLQWRkVHJ1c3QgQUIxJjAkBgNVBAsTHUFk
> ZFRydXN0IEV4dGVybmFsIFRUUCBOZXR3b3JrMSIwIAYDVQQDExlBZGRUcnVzdCBF
> eHRlcm5hbCBDQSBSb290MB4XDTAwMDUzMDEwNDgzOFoXDTIwMDUzMDEwNDgzOFow
> gYgxCzAJBgNVBAYTAlVTMRMwEQYDVQQIEwpOZXcgSmVyc2V5MRQwEgYDVQQHEwtK
> ZXJzZXkgQ2l0eTEeMBwGA1UEChMVVGhlIFVTRVJUUlVTVCBOZXR3b3JrMS4wLAYD
> VQQDEyVVU0VSVHJ1c3QgUlNBIENlcnRpZmljYXRpb24gQXV0aG9yaXR5MIICIjAN
> BgkqhkiG9w0BAQEFAAOCAg8AMIICCgKCAgEAgBJlFzYOw9sIs9CsVw127c0n00yt
> UINh4qogTQktZAnczomfzD2p7PbPwdzx07HWezcoEStH2jnGvDoZtF+mvX2do2NC
> tnbyqTsrkfjib9DsFiCQCT7i6HTJGLSR1GJk23+jBvGIGGqQIjy8/hPwhxR79uQf
> jtTkUcYRZ0YIUcuGFFQ/vDP+fmyc/xadGL1RjjWmp2bIcmfbIWax1Jt4A8BQOujM
> 8Ny8nkz+rwWWNR9XWrf/zvk9tyy29lTdyOcSOk2uTIq3XJq0tyA9yn8iNK5+O2hm
> AUTnAU5GU5szYPeUvlM3kHND8zLDU+/bqv50TmnHa4xgk97Exwzf4TKuzJM7UXiV
> Z4vuPVb+DNBpDxsP8yUmazNt925H+nND5X4OpWaxKXwyhGNVicQNwZNUMBkTrNN9
> N6frXTpsNVzbQdcS2qlJC9/YgIoJk2KOtWbPJYjNhLixP6Q5D9kCnusSTJV882sF
> qV4Wg8y4Z+LoE53MW4LTTLPtW//e5XOsIzstAL81VXQJSdhJWBp/kjbmUZIO8yZ9
> HE0XvMnsQybQv0FfQKlERPSZ51eHnlAfV1SoPv10Yy+xUGUJ5lhCLkMaTLTwJUdZ
> +gQek9QmRkpQgbLevni3/GcV4clXhB4PY9bpYrrWX1Uu6lzGKAgEJTm4Diup8kyX
> HAc/DVL17e8vgg8CAwEAAaOB9DCB8TAfBgNVHSMEGDAWgBStvZh6NLQm9/rEJlTv
> A73gJMtUGjAdBgNVHQ4EFgQUU3m/WqorSs9UgOHYm8Cd8rIDZsswDgYDVR0PAQH/
> BAQDAgGGMA8GA1UdEwEB/wQFMAMBAf8wEQYDVR0gBAowCDAGBgRVHSAAMEQGA1Ud
> HwQ9MDswOaA3oDWGM2h0dHA6Ly9jcmwudXNlcnRydXN0LmNvbS9BZGRUcnVzdEV4
> dGVybmFsQ0FSb290LmNybDA1BggrBgEFBQcBAQQpMCcwJQYIKwYBBQUHMAGGGWh0
> dHA6Ly9vY3NwLnVzZXJ0cnVzdC5jb20wDQYJKoZIhvcNAQEMBQADggEBAJNl9jeD
> lQ9ew4IcH9Z35zyKwKoJ8OkLJvHgwmp1ocd5yblSYMgpEg7wrQPWCcR23+WmgZWn
> RtqCV6mVksW2jwMibDN3wXsyF24HzloUQToFJBv2FAY7qCUkDrvMKnXduXBBP3zQ
> YzYhBx9G/2CkkeFnvN4ffhkUyWNnkepnB2u0j4vAbkN9w6GAbLIevFOFfdyQoaS8
> Le9Gclc1Bb+7RrtubTeZtv8jkpHGbkD4jylW6l/VXxRTrPBPYer3IsynVgviuDQf
> Jtl7GQVoP7o81DgGotPmjw7jtHFtQELFhLRAlSv0ZaBIefYdgWOWnU914Ph85I6p
> 0fKtirOMxyHNwu8=
> -----END CERTIFICATE-----
> "
notBefore=May 30 10:48:38 2000 GMT
notAfter=May 30 10:48:38 2020 GMT
```
Strange that curl connects without complaining and my APT instance as well. And why don't I see this expired CA cert in Windows? It looks like `mirrors.netix.net` ships an outdated root CA cert on requests while there is a valid one installed on the system that is used for validation instead? I'm a bid confused. However yeah either way `mirrors.netix.net` should know and do something about it 😄.

jerryhopper, 2020-07-29:
certificate is fixed by netix support.   case closed, problem solved!

closed 2020-07-29 by jerryhopper