MichaIng/DietPi #3699: Nzbget failing to start after intalling Netdata

beerbandito, 2020-07-29:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.31
- Distro version | 10.4
- Kernel version | Linux DietPi 5.4.51-v7l+ #1327
- SBC model | RPi 4 Model B (armv7l)
- Power supply used | 5.1V / 3.0A 
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | NZBget and/or Netdata
- Bug report ID | ec7cbf41-8f15-4f47-ae8c-93da6b39a8ef

#### Actual behaviour
-NZBget wont start after I installed Netdata. 

#### Extra details
I've tried manually starting NZBget in dietpi-services and tried reinstalling both NZBget and Netdata but it always fails.

Joulinar, 2020-07-29:
Hi,

many thanks for your report. Pls can you share following `journalctl -u nzbget.service`

beerbandito, 2020-07-30:
-- Logs begin at Thu 2019-02-14 11:11:58 CET, end at Thu 2020-07-30 06:20:01 CES
T. --
Jul 28 19:27:03 DietPi systemd[1]: Starting NZBget (DietPi)...
Jul 28 19:27:04 DietPi nzbget[774]: nzbget.conf(0): Could not open file /mnt/myb
ook/dietpi_userdata/nzbget/nzbget.conf
Jul 28 19:27:04 DietPi systemd[1]: nzbget.service: Contr
ol process exited, code=exited, status=1/FAILURE
Jul 28 19:27:04 DietPi systemd[1]: nzbget.service: Faile
d with result 'exit-code'.
Jul 28 19:27:04 DietPi systemd[1]: Failed to start NZBge
t (DietPi).
Jul 28 19:35:23 DietPi systemd[1]: Starting NZBget (DietPi)...
Jul 28 19:35:23 DietPi nzbget[2044]: nzbget.conf(0): Could not open file /mnt/my
book/dietpi_userdata/nzbget/nzbget.conf
Jul 28 19:35:23 DietPi systemd[1]: nzbget.service: Contr
ol process exited, code=exited, status=1/FAILURE
Jul 28 19:35:23 DietPi systemd[1]: nzbget.service: Faile
d with result 'exit-code'.
Jul 28 19:35:23 DietPi systemd[1]: Failed to start NZBge
t (DietPi).
Jul 28 19:35:26 DietPi systemd[1]: Starting NZBget (DietPi)...
Jul 28 19:35:26 DietPi nzbget[2068]: nzbget.conf(0): Could not open file /mnt/my
book/dietpi_userdata/nzbget/nzbget.conf

Joulinar, 2020-07-30:
There you go

```
nzbget.conf(0): Could not open file /mnt/my
book/dietpi_userdata/nzbget/nzbget.conf
```

Did you moved DietPi user data to an external device?

beerbandito, 2020-07-30:
Not recently. I moved it when I installed dietpi about 6 months ago

Joulinar, 2020-07-30:
can you past `ls -la /mnt/mybook/dietpi_userdata/nzbget/`

Joulinar, 2020-07-30:
ok you habe the config file, but it doesn't have any permissions. Therefore Nzbget is not able to read it.

`---------- 1 nzbget dietpi   68047 Jul 28 19:10 nzbget.conf`

Pls do following 
```
chmod 644 /mnt/mybook/dietpi_userdata/nzbget/nzbget.conf
systemctl restart nzbget.service
```

check status
systemctl status nzbget.service

if still not working, pls post `journalctl -u nzbget.service` again.

beerbandito, 2020-07-31:
It worked for a few minutes then failed. I repeated it after rebooting and got the same result.

-- Logs begin at Thu 2019-02-14 11:11:58 CET, end at Fri 2020-07-31 05:43:27 CES
T. --
Jul 31 05:37:28 DietPi systemd[1]: Starting NZBget (DietPi)...
Jul 31 05:37:29 DietPi systemd[1]: Started NZBget (DietPi).
Jul 31 05:39:54 DietPi systemd[1]: nzbget.service: Main
process exited, code=killed, status=11/SEGV
Jul 31 05:39:54 DietPi systemd[1]: nzbget.service: Faile
d with result 'signal'.
Jul 31 05:41:02 DietPi systemd[1]: Starting NZBget (DietPi)...
Jul 31 05:41:02 DietPi systemd[1]: Started NZBget (DietPi).
Jul 31 05:43:27 DietPi systemd[1]: nzbget.service: Main
process exited, code=killed, status=11/SEGV
Jul 31 05:43:27 DietPi systemd[1]: nzbget.service: Faile
d with result 'signal'.

MichaIng, 2020-07-31:
Does the binary log file exist and has correct permissions?
```
ls -l /var/log/nzbget.log
tail -20 /var/log/nzbget.log
```

beerbandito, 2020-07-31:
root@DietPi:~# ls -l /var/log/nzbget.log
-rw-rw-r-- 1 nzbget dietpi 13566 Jul 31 15:35 /var/log/nzbget.log

root@DietPi:~# tail -20 /var/log/nzbget.log
Fri Jul 31 15:32:50 2020        INFO    Unrar: Extracting from 436dd3a0a97a4241a142c7115384562e.part05.rar
Fri Jul 31 15:32:54 2020        INFO    Unrar: Extracting from 436dd3a0a97a4241a142c7115384562e.part06.rar
Fri Jul 31 15:32:59 2020        INFO    Unrar: Extracting from 436dd3a0a97a4241a142c7115384562e.part07.rar
Fri Jul 31 15:33:02 2020        INFO    Unrar: Extracting from 436dd3a0a97a4241a142c7115384562e.part08.rar
Fri Jul 31 15:33:06 2020        INFO    Unrar: Extracting from 436dd3a0a97a4241a142c7115384562e.part09.rar
Fri Jul 31 15:33:07 2020        INFO    Unrar: Extracting from 436dd3a0a97a4241a142c7115384562e.part10.rar
Fri Jul 31 15:35:11 2020        ERROR   Error saving diskstate: Could not rename file /mnt/dietpi_userdata/downloads/queue/progress.new to /mnt/dietpi_userdata/downloads/queue/progress:
Fri Jul 31 15:35:18 2020        ERROR   Error saving diskstate: Could not rename file /mnt/dietpi_userdata/downloads/queue/progress.new to /mnt/dietpi_userdata/downloads/queue/progress:
Fri Jul 31 15:35:20 2020        ERROR   Error saving diskstate: Could not rename file /mnt/dietpi_userdata/downloads/queue/progress.new to /mnt/dietpi_userdata/downloads/queue/progress:
Fri Jul 31 15:35:22 2020        ERROR   Error saving diskstate: Could not rename file /mnt/dietpi_userdata/downloads/queue/progress.new to /mnt/dietpi_userdata/downloads/queue/progress:
Fri Jul 31 15:35:24 2020        ERROR   Error saving diskstate: Could not rename file /mnt/dietpi_userdata/downloads/queue/progress.new to /mnt/dietpi_userdata/downloads/queue/progress:
Fri Jul 31 15:35:26 2020        ERROR   Error saving diskstate: Could not rename file /mnt/dietpi_userdata/downloads/queue/progress.new to /mnt/dietpi_userdata/downloads/queue/progress:
Fri Jul 31 15:35:27 2020        INFO    Partially downloaded Amovie.part11.rar
Fri Jul 31 15:35:28 2020        ERROR   Error saving diskstate: Could not rename file /mnt/dietpi_userdata/downloads/queue/progress.new to /mnt/dietpi_userdata/downloads/queue/progress:
Fri Jul 31 15:35:28 2020        INFO    Unrar: Extracting from 436dd3a0a97a4241a142c7115384562e.part11.rar
Fri Jul 31 15:35:29 2020        ERROR   Error saving diskstate: Could not rename file /mnt/dietpi_userdata/downloads/queue/progress.new to /mnt/dietpi_userdata/downloads/queue/progress:
Fri Jul 31 15:35:32 2020        ERROR   Error saving diskstate: Could not rename file /mnt/dietpi_userdata/downloads/queue/progress.new to /mnt/dietpi_userdata/downloads/queue/progress:
Fri Jul 31 15:35:33 2020        ERROR   Error saving diskstate: Could not rename file /mnt/dietpi_userdata/downloads/queue/progress.new to /mnt/dietpi_userdata/downloads/queue/progress:
Fri Jul 31 15:35:36 2020        ERROR   Error saving diskstate: Could not rename file /mnt/dietpi_userdata/downloads/queue/progress.new to /mnt/dietpi_userdata/downloads/queue/progress:
Fri Jul 31 15:35:37 2020        ERROR   Error saving diskstate: Could not rename file /mnt/dietpi_userdata/downloads/queue/progress.new to /mnt/dietpi_userdata/downloads/queue/progress:

beerbandito, 2020-07-31:
Sorry I forgot to mention that NZBGet is running without failing (with downloads paused) but I'm getting heaps of errors messages as you can see.

MichaIng, 2020-07-31:
Could you please paste the following to check dir and file permissions? Mask any concrete file names in case:
```
ls -al /mnt/dietpi_userdata/downloads/{,queue/{,progress*}}
```

beerbandito, 2020-07-31:
root@DietPi:~# ls -al /mnt/dietpi_userdata/downloads/{,queue/{,progress*}}
ls: cannot access '/mnt/dietpi_userdata/downloads/queue/progress': Input/output error
-rwxrwxrwx 1 nzbget dietpi    2425 Jul 31 15:35 /mnt/dietpi_userdata/downloads/queue/progress.new

/mnt/dietpi_userdata/downloads/:
total 1703077
drwxrwxrwx 1 dietpi      dietpi      4096 Jul 27 18:30  .
drwxrwxrwx 1 dietpi      dietpi      4096 Apr  9 13:20  ..
drwxrwxrwx 1 nzbget      dietpi         0 Apr  3 18:56  complete
drwxrwxrwx 1 qbittorrent dietpi      4096 Apr 10 06:34  justanotherfile
drwxrwxrwx 1 qbittorrent dietpi      4096 May 14 10:40  justanotherfile
drwxrwxrwx 1 nzbget      dietpi      4096 Jul 31 15:11  intermediate
drwxrwxrwx 1 nzbget      dietpi    417792 Jul 31 16:01  nzb
-rwxrwxrwx 1 nzbget      dietpi         6 Jul 31 15:09  nzbget.lock
drwxrwxrwx 1 nzbget      dietpi   6160384 Jul 31 16:01  queue
-rwxrwxrwx 1 qbittorrent dietpi 929623142 Apr  9 22:53 'justanotherfile.mkv'
-rwxrwxrwx 1 qbittorrent dietpi 807713036 Apr 10 04:23 'justanotherfile.mkv'
drwxrwxrwx 1 qbittorrent dietpi      4096 Apr 10 04:29  justanotherfile
drwxrwxrwx 1 qbittorrent dietpi      4096 May 14 10:58  justanotherfile
drwxrwxrwx 1 nzbget      dietpi         0 Jul 31 16:01  tmp
drwxrwxrwx 1 qbittorrent dietpi      4096 Apr  9 13:23 'justanotherfile'

/mnt/dietpi_userdata/downloads/queue/:
ls: cannot access '/mnt/dietpi_userdata/downloads/queue/progress': Input/output error
ls: cannot access '/mnt/dietpi_userdata/downloads/queue/4594s': No such file or directory
total 19447
drwxrwxrwx 1 nzbget dietpi 6160384 Jul 31 16:01 .
drwxrwxrwx 1 dietpi dietpi    4096 Jul 27 18:30 ..
-rw-rw-r-- 1 nzbget dietpi     239 Jul 31 16:01 1637
-rw-rw-r-- 1 nzbget dietpi    8184 Jul 31 16:01 1638
-rw-rw-r-- 1 nzbget dietpi    8186 Jul 31 16:01 1639
-rw-rw-r-- 1 nzbget dietpi    8187 Jul 31 16:01 1640
-rw-rw-r-- 1 nzbget dietpi    8184 Jul 31 16:01 1641
-rw-rw-r-- 1 nzbget dietpi    8184 Jul 31 16:01 1642
-rw-rw-r-- 1 nzbget dietpi    8187 Jul 31 16:01 1643
-rw-rw-r-- 1 nzbget dietpi    8188 Jul 31 16:01 1644
-rw-rw-r-- 1 nzbget dietpi    8186 Jul 31 16:01 1645
-rw-rw-r-- 1 nzbget dietpi    8188 Jul 31 16:01 1646
-rw-rw-r-- 1 nzbget dietpi    8186 Jul 31 16:01 1647
-rw-rw-r-- 1 nzbget dietpi    8184 Jul 31 16:01 1648
-rw-rw-r-- 1 nzbget dietpi    8188 Jul 31 16:01 1649
-rw-rw-r-- 1 nzbget dietpi    8184 Jul 31 16:01 1650
-rw-rw-r-- 1 nzbget dietpi    8186 Jul 31 16:01 1651
-rw-rw-r-- 1 nzbget dietpi    8184 Jul 31 16:01 1652
-rw-rw-r-- 1 nzbget dietpi    8187 Jul 31 16:01 1653
-rw-rw-r-- 1 nzbget dietpi    8188 Jul 31 16:01 1654
-rw-rw-r-- 1 nzbget dietpi    8188 Jul 31 16:01 1655
-rw-rw-r-- 1 nzbget dietpi    8186 Jul 31 16:01 1656
-rw-rw-r-- 1 nzbget dietpi    8187 Jul 31 16:01 1657
-rw-rw-r-- 1 nzbget dietpi    8187 Jul 31 16:01 1658
-rw-rw-r-- 1 nzbget dietpi    8186 Jul 31 16:01 1659
-rw-rw-r-- 1 nzbget dietpi    8186 Jul 31 16:01 1660
-rw-rw-r-- 1 nzbget dietpi    8184 Jul 31 16:01 1661
-rw-rw-r-- 1 nzbget dietpi    8186 Jul 31 16:01 1662
-rw-rw-r-- 1 nzbget dietpi    8187 Jul 31 16:01 1663
-rw-rw-r-- 1 nzbget dietpi    8184 Jul 31 16:01 1664
-rw-rw-r-- 1 nzbget dietpi    8184 Jul 31 16:01 1665
-rw-rw-r-- 1 nzbget dietpi    8184 Jul 31 16:01 1666
-rw-rw-r-- 1 nzbget dietpi    8184 Jul 31 16:01 1667
-rw-rw-r-- 1 nzbget dietpi    8184 Jul 31 16:01 1668
-rw-rw-r-- 1 nzbget dietpi    8184 Jul 31 16:01 1669
-rw-rw-r-- 1 nzbget dietpi    8188 Jul 31 16:01 1670
-rw-rw-r-- 1 nzbget dietpi    8186 Jul 31 16:01 1671
-rw-rw-r-- 1 nzbget dietpi    8184 Jul 31 16:01 1672
-rw-rw-r-- 1 nzbget dietpi    8186 Jul 31 16:01 1673
-rw-rw-r-- 1 nzbget dietpi    8188 Jul 31 16:01 1674
-rw-rw-r-- 1 nzbget dietpi    8188 Jul 31 16:01 1675
-rw-rw-r-- 1 nzbget dietpi    8187 Jul 31 16:01 1676
-rw-rw-r-- 1 nzbget dietpi    8187 Jul 31 16:01 1677
-rw-rw-r-- 1 nzbget dietpi    8187 Jul 31 16:01 1678
[... 1044 lines cut ...]
-rwxrwxrwx 1 nzbget dietpi   82170 Jul 26 09:04 n2138.log
-rwxrwxrwx 1 nzbget dietpi   34290 Jul 25 09:36 n2139.log
-rwxrwxrwx 1 nzbget dietpi   19967 Jul 25 09:47 n2140.log
-rwxrwxrwx 1 nzbget dietpi   14723 Jul 26 10:02 n2141.log
-rwxrwxrwx 1 nzbget dietpi   18793 Jul 25 09:55 n2145.log
-rwxrwxrwx 1 nzbget dietpi   23147 Jul 25 10:06 n2149.log
-rwxrwxrwx 1 nzbget dietpi   29454 Jul 25 10:32 n2162.log
-rwxrwxrwx 1 nzbget dietpi   14649 Jul 25 10:40 n2165.log
-rwxrwxrwx 1 nzbget dietpi    7263 Jul 25 10:40 n2167.log
-rwxrwxrwx 1 nzbget dietpi   56405 Jul 25 15:53 n2172.log
-rwxrwxrwx 1 nzbget dietpi    6836 Jul 25 15:56 n2179.log
-rwxrwxrwx 1 nzbget dietpi   17658 Jul 25 16:09 n2183.log
-rwxrwxrwx 1 nzbget dietpi     281 Jul 25 15:32 n2185.log
-rwxrwxrwx 1 nzbget dietpi   11973 Jul 25 16:15 n2188.log
-rwxrwxrwx 1 nzbget dietpi   34373 Jul 25 16:21 n2191.log
-rwxrwxrwx 1 nzbget dietpi   44995 Jul 25 16:27 n2192.log
-rwxrwxrwx 1 nzbget dietpi   19124 Jul 25 16:39 n2193.log
-rwxrwxrwx 1 nzbget dietpi    7505 Jul 25 16:40 n2196.log
-rwxrwxrwx 1 nzbget dietpi   35929 Jul 26 07:59 n2214.log
-rwxrwxrwx 1 nzbget dietpi   31455 Jul 26 08:08 n2219.log
-rw-rw-r-- 1 nzbget dietpi     241 Jul 31 16:01 n25.log
-????????? ? ?      ?            ?            ? progress
-rwxrwxrwx 1 nzbget dietpi    2425 Jul 31 15:35 progress.new
-rw-rw-r-- 1 nzbget dietpi   18963 Jul 31 16:01 queue
-rw-rw-r-- 1 nzbget dietpi    6533 Jul 31 15:35 stats

beerbandito, 2020-07-31:
I'm also getting the following error message from both Sonarr and Radarr:

Unable to communicate with Nzbget. Unable to connect to NzbGet. Error: ConnectFailure (Connection refused): 'http://localhost:6789/jsonrpc'

MichaIng, 2020-07-31:
```
-????????? ? ? ? ? ? 4594s
-????????? ? ? ? ? ? progress
```
These two files are corrupted. Please stop NZBget and try to remove them:
```
systemctl stop nzbget
rm /mnt/dietpi_userdata/downloads/queue/{progress,4594s}
```
If this still fails, please try to reset their permissions and owner first:
```
chattr -i /mnt/dietpi_userdata/downloads/queue/{progress,4594s}
chattr -a /mnt/dietpi_userdata/downloads/queue/{progress,4594s}
chmod 0777 /mnt/dietpi_userdata/downloads/queue/{progress,4594s}
chown nzbget:dietpi /mnt/dietpi_userdata/downloads/queue/{progress,4594s}
rm /mnt/dietpi_userdata/downloads/queue/{progress,4594s}
```
If all this fails still, then I guess you need to do a fsck to clean those errors. You can do this via `dietpi-drive_manager` or manually via e.g. `umount /dev/sdb1 && fsck -f /dev/sdb1` if this is the external drive (`df` to check).

beerbandito, 2020-08-01:
root@DietPi:~# systemctl stop nzbget
root@DietPi:~# rm /mnt/dietpi_userdata/downloads/queue/{progress,4594s}
rm: cannot remove '/mnt/dietpi_userdata/downloads/queue/progress': No such file or directory
rm: cannot remove '/mnt/dietpi_userdata/downloads/queue/4594s': Input/output error

root@DietPi:~# chattr -i /mnt/dietpi_userdata/downloads/queue/{progress,4594s}
chattr: Invalid argument while reading flags on /mnt/dietpi_userdata/downloads/queue/progress
chattr: Input/output error while trying to stat /mnt/dietpi_userdata/downloads/queue/4594s

root@DietPi:~# chattr -a /mnt/dietpi_userdata/downloads/queue/{progress,4594s}
chattr: Invalid argument while reading flags on /mnt/dietpi_userdata/downloads/queue/progress
chattr: Input/output error while trying to stat /mnt/dietpi_userdata/downloads/queue/4594s

root@DietPi:~# chmod 0777 /mnt/dietpi_userdata/downloads/queue/{progress,4594s}
chmod: cannot access '/mnt/dietpi_userdata/downloads/queue/4594s': Input/output error

root@DietPi:~# chown nzbget:dietpi /mnt/dietpi_userdata/downloads/queue/{progress,4594s}
chown: cannot access '/mnt/dietpi_userdata/downloads/queue/4594s': Input/output error

root@DietPi:~# rm /mnt/dietpi_userdata/downloads/queue/{progress,4594s}
rm: cannot remove '/mnt/dietpi_userdata/downloads/queue/progress': No such file or directory
rm: cannot remove '/mnt/dietpi_userdata/downloads/queue/4594s': Input/output error

I also did the Check & Repair in dietpi-drive_manager but it is no different. I deleted everything from my Nzbget queue and tried changing permissions and tried  ls -al /mnt/dietpi_userdata/downloads/{,queue/{,progress*}} again. 

root@DietPi:~# ls -al /mnt/dietpi_userdata/downloads/{,queue/{,progress*}}
-rwxrwxrwx 1 nzbget dietpi     691 Aug  1 07:21 /mnt/dietpi_userdata/downloads/queue/progress
-rwxrwxr-x 1 nzbget dietpi    1143 Aug  1 07:56 /mnt/dietpi_userdata/downloads/queue/progress.new

/mnt/dietpi_userdata/downloads/:
total 1703077
drwxrwxr-x 1 dietpi      dietpi      4096 Aug  1 07:52  .
drwxrwxr-x 1 dietpi      dietpi      4096 Apr  9 13:20  ..
drwxrwxr-x 1 nzbget      dietpi         0 Apr  3 18:56  complete
drwxrwxr-x 1 qbittorrent dietpi      4096 Apr 10 06:34  maskedfilename
drwxrwxr-x 1 qbittorrent dietpi      4096 May 14 10:40  maskedfilename
drwxrwxr-x 1 nzbget      dietpi      4096 Aug  1 07:33  intermediate
drwxrwxr-x 1 nzbget      dietpi    417792 Aug  1 07:33  nzb
-rw-r----- 1 nzbget      dietpi         5 Aug  1 07:56  nzbget.lock
drwxrwxr-x 1 nzbget      dietpi   6160384 Aug  1 07:56  queue
-rwxrwxr-x 1 qbittorrent dietpi 929623142 Apr  9 22:53 'maskedfilename.mkv'
-rwxrwxr-x 1 qbittorrent dietpi 807713036 Apr 10 04:23 'maskedfilename.mkv'
drwxrwxr-x 1 qbittorrent dietpi      4096 Apr 10 04:29  maskedfilename
drwxrwxr-x 1 qbittorrent dietpi      4096 May 14 10:58  maskedfilename
drwxrwxr-x 1 nzbget      dietpi         0 Aug  1 07:33  tmp
drwxrwxr-x 1 qbittorrent dietpi      4096 Apr  9 13:23 'maskedfilename'

/mnt/dietpi_userdata/downloads/queue/:
ls: cannot access '/mnt/dietpi_userdata/downloads/queue/4594s': Input/output error
ls: cannot access '/mnt/dietpi_userdata/downloads/queue/4595': Input/output error
total 15313
drwxrwxr-x 1 nzbget dietpi 6160384 Aug  1 07:56 .
drwxrwxr-x 1 dietpi dietpi    4096 Aug  1 07:52 ..
-rwxrwxr-x 1 nzbget dietpi    3572 Aug  1 07:20 100
-rwxrwxr-x 1 nzbget dietpi    2009 Aug  1 07:21 1000
-rwxrwxr-x 1 nzbget dietpi    3644 Aug  1 07:21 1001
-rwxrwxr-x 1 nzbget dietpi    6735 Aug  1 07:21 1002
-rwxrwxr-x 1 nzbget dietpi    2090 Aug  1 07:21 1003
-rwxrwxr-x 1 nzbget dietpi    3740 Aug  1 07:21 1004
-rwxrwxr-x 1 nzbget dietpi    3669 Aug  1 07:21 1005
-rwxrwxr-x 1 nzbget dietpi    3749 Aug  1 07:21 1006
-rwxrwxr-x 1 nzbget dietpi    3778 Aug  1 07:21 1007
-rwxrwxr-x 1 nzbget dietpi     266 Aug  1 07:21 1008
-rwxrwxr-x 1 nzbget dietpi    3629 Aug  1 07:21 1009
-rw-rw-r-- 1 nzbget dietpi     643 Aug  1 07:56 100s
-rwxrwxr-x 1 nzbget dietpi    3575 Aug  1 07:20 101
-rwxrwxr-x 1 nzbget dietpi    3573 Aug  1 07:21 1010
-rwxrwxr-x 1 nzbget dietpi    3555 Aug  1 07:21 1011
-rwxrwxr-x 1 nzbget dietpi    3606 Aug  1 07:21 1012
-rwxrwxr-x 1 nzbget dietpi    3625 Aug  1 07:21 1013
[... 1392 lines cut ...]
-rwxrwxr-x 1 nzbget dietpi   29454 Jul 25 10:32 n2162.log
-rwxrwxr-x 1 nzbget dietpi   14649 Jul 25 10:40 n2165.log
-rwxrwxr-x 1 nzbget dietpi    7263 Jul 25 10:40 n2167.log
-rwxrwxr-x 1 nzbget dietpi   56405 Jul 25 15:53 n2172.log
-rwxrwxr-x 1 nzbget dietpi    6836 Jul 25 15:56 n2179.log
-rwxrwxr-x 1 nzbget dietpi   17658 Jul 25 16:09 n2183.log
-rwxrwxr-x 1 nzbget dietpi   11973 Jul 25 16:15 n2188.log
-rwxrwxr-x 1 nzbget dietpi   34373 Jul 25 16:21 n2191.log
-rwxrwxr-x 1 nzbget dietpi   44995 Jul 25 16:27 n2192.log
-rwxrwxr-x 1 nzbget dietpi   19124 Jul 25 16:39 n2193.log
-rwxrwxr-x 1 nzbget dietpi    7505 Jul 25 16:40 n2196.log
-rwxrwxr-x 1 nzbget dietpi   35929 Jul 26 07:59 n2214.log
-rwxrwxr-x 1 nzbget dietpi   31455 Jul 26 08:08 n2219.log
-rwxrwxr-x 1 nzbget dietpi     263 Aug  1 07:20 n2.log
-rwxrwxr-x 1 nzbget dietpi     263 Aug  1 07:20 n3.log
-rwxrwxr-x 1 nzbget dietpi     247 Aug  1 07:21 n4.log
-rwxrwxr-x 1 nzbget dietpi     247 Aug  1 07:21 n5.log
-rwxrwxr-x 1 nzbget dietpi     263 Aug  1 07:21 n6.log
-rwxrwxr-x 1 nzbget dietpi     263 Aug  1 07:21 n7.log
-rwxrwxr-x 1 nzbget dietpi     263 Aug  1 07:21 n8.log
-rwxrwxr-x 1 nzbget dietpi     263 Aug  1 07:21 n9.log
-rwxrwxrwx 1 nzbget dietpi     691 Aug  1 07:21 progress
-rwxrwxr-x 1 nzbget dietpi    1143 Aug  1 07:56 progress.new
-rw-rw-r-- 1 nzbget dietpi   21267 Aug  1 07:33 queue
-rw-rw-r-- 1 nzbget dietpi    6440 Aug  1 07:56 stats

MichaIng, 2020-08-01:
Okay the progress file has been cleared, but the other one not + there as a next corrupted 🤔. In case try to run fsck repeatedly and the command series I posted above on those two: https://github.com/MichaIng/DietPi/issues/3699#issuecomment-667342657

Is it normal that NZBget stores downloads while processing in a huge amount of such tiny ~3000 byte files? This is highly inefficient due to the fixed 4k blocks usually. Larger chunks would be reasonable.

beerbandito, 2020-08-01:
>Okay the progress file has been cleared, but the other one not + there as a next corrupted 🤔. In case try to run fsck repeatedly and the command series I posted above on those two: #3699 (comment)

I will try this

> Is it normal that NZBget stores downloads while processing in a huge amount of such tiny ~3000 byte files? This is highly inefficient due to the fixed 4k blocks usually. Larger chunks would be reasonable.

No idea. I am a total noob to all this stuff. Maybe some settings have changed in Nzbget so I will check to see if anything seems obviously wrong.

beerbandito, 2020-08-02:
>Okay the progress file has been cleared, but the other one not + there as a next corrupted 🤔. In case try to run fsck repeatedly and the command series I posted above on those two: #3699 (comment)

I've tried to clear them but nothing is happening. They're always showing up.

MichaIng, 2020-08-02:
Is it a NTFS drive, right? If so, I see people recommending to check and repair on Windows since the Linux ntfsfix/fsck.ntfs seems to have not full abilities to do that.

If its ext4 then fsck should be able to fix it when calling it interactively, at least as there is not bad block on the physical disk itself (known from SDcards which start to die). See if this looks familiar: https://www.claudiokuenzler.com/blog/325/rm-cannot-remove-input-output-error-ext4-fs

beerbandito, 2020-08-02:
>Is it a NTFS drive, right? If so, I see people recommending to check and repair on Windows since the Linux ntfsfix/fsck.ntfs seems to have not full abilities to do that.

It is NTFS but for some reason I can't assign a drive letter to it on my windows machine. I think I'll reformat and do a fresh install

MichaIng, 2020-08-02:
> It is NTFS but for some reason I can't assign a drive letter to it on my windows machine. I think I'll format it and do a fresh install

It seems to be possible to run chkdsk without drive letter: https://www.tecknowledgebase.com/85/how-to-run-chkdsk-on-a-disk-without-a-drive-letter/
But of course a fresh flash will do as well.

closed 2020-09-09 by MichaIng