MichaIng/DietPi #3702: dietpi-software :- Updating or reinstalling - reports No Internet

MDAR, 2020-07-31:
#### Details:
- Date           | Fri Jul 31 15:45:34 BST 2020
- DietPi version | v6.31.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid C2 (aarch64) (ID=12)
- Kernel version | Linux DietPi 3.16.57+ #1 SMP PREEMPT Sun Aug 19 15:27:56 CEST 2018 aarch64 GNU/Linux
- Distro         | stretch (ID=4)
- Command        | ./node-install.sh
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

I can run 
apt-get update 
apt-get upgrade
happily from the command line and perform all other tasks.

But the dietpi-update fails and reports no internet

As does trying to upgrade Node-RED

#### Expected behaviour:

#### Actual behaviour:

Scripts fail and report "No Internet"

#### Extra details:
is supect it is failing to download a particular file, rather than not getting access to the internet

#### Additional logs:
```
Node Linux Installer by www.github.com/taaem
Running as Root User
Get Latest Version Number...
ERROR: No Internet Connection
```

MDAR, 2020-07-31:
HOWEVER

Node-RED hasn't been installed / updated.

Joulinar, 2020-07-31:
Hi,

Many thanks for your report. This is already known and should be similar to #3671 

Usually a quick fix should have been provided by MOTD.

MichaIng, 2020-07-31:
Quick fix:
```
sed -i 's/taaem/MichaIng/' /boot/dietpi/dietpi-software
```

closed 2020-07-31 by MichaIng

MDAR, 2020-07-31:
Thank you so much 😃 

This has fixed it perfectly 😄