MichaIng/DietPi #3703: Grub Rescue

aspinks, 2020-07-31:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ... after installing unrar I get the attached error after reboot. I have tried fresh installs on other hard drives and get the same error.  I have attached an image of the error. 
2. ...

#### Expected behaviour

- ... it should not to the normal command line. 

#### Actual behaviour

- ... grub error

#### Extra details
 
- ... happened after I instants unrar and it does this on any drive I do a fresh install on.

Joulinar, 2020-07-31:
Hi,

Many thanks for your report.  Probably this is impacting you as well #3700

MichaIng, 2020-07-31:
Jep, before you install or update anything, do the following to proactively fix grub:
```
debconf-set-selections <<< 'grub-pc grub-pc/install_devices multiselect /dev/sda'
apt update
apt install --reinstall grub-pc
```
The VirtualBox image on our download page has been updated with the fix (and in general refreshed from scratch).

Joulinar, 2020-07-31:
@MichaIng 
It's a PC, not a VM. 

Possible solution if you can use a live USB stick 

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=966575#35

aspinks, 2020-07-31:
Do I just run that code on the screen with the error at boot?

MichaIng, 2020-07-31:
Dammit, I'll check our BIOS mage as well then and recreate it. Makes sense as usually physical drives have such an ID which is the culprit here.

I just added what we got to the bug tracker as well.

@aspinks 
Ah sorry nope you cannot flash grub from the grub console, AFAIK. Hmm not sure how to do what, theoretically it should be possible from a second machine, attaching the drive there and flashing grub manually or via interactive debconf selection to the attached drive. In case you have a backup (e.g. dietpi-backup on dist or external one) it would be probably easier to restore the previous `/boot/grub` content which should restore bootability.

aspinks, 2020-07-31:
How do I restore to the previous boot/grub? Sorry still learning

MichaIng, 2020-08-01:
Please do the following:
1. Flash the new image.
2. Boot it.
3. When logging in, cancel the running firstrun scripts (hitting ctrl+c) and apply the above fix manually.
4. Then reboot to quickly check if it works and firstrun update/setup will proceed.

aspinks, 2020-08-01:
That got me going for now, thank you! .  How will I know when the new image is ready?

MichaIng, 2020-08-01:
I'll send a post here, will be done later today.

MichaIng, 2020-09-27:
I mark this as closed in favour of #3700.

closed 2020-09-27 by MichaIng