MichaIng/DietPi #3704: Blynk Update and Version services

Phil1988, 2020-08-02:
I am using a Blynk Server and just saw, that the new version supports some sort of theming and wanted to give it a try. 
(https://community.blynk.cc/t/change-the-green-app-color/41622/11)

But I had a bit of work to archive this
The Blynk server itself is started (as far as I know) from /mnt/dietpi_userdata/blynk/blynkserver.jar.

I know its important to have generic names for such a project as DietPi, but normally the *.jar file is containing the version number of the server itself.
Is there a way (systemd service) to report back the version number?

There is also no update command for this (as I know... like for pihole the **pohole -up**).
Is there?
Would it be enough to only replace the blynkserver.jar with the latest from https://github.com/blynkkk/blynk-server/releases ?

If not, then I made a short tutorial how I successfully made it - for others that might have the same goal as I had.

Maybe someone can post it like here: https://dietpi.com/phpbb/viewtopic.php?p=5901#p5901
for others to find it better:

Updating the Blynk server:
______________________________________

First, backup the entire folder /mnt/dietpi_userdata/blynk with all subdirectorys
Then uninstall blynk with dietpi-software and install it after this with dietpi-software :)

For the **users**:
restore all *.Blynk.user files in /mnt/dietpi_userdata/blynk/data/ with owner 'blynk' [998] and rights 0644

For cloned **projects**:
restore all files in /mnt/dietpi_userdata/blynkBackup/data/clone as owner 'root' [0] and rights 0755

For the **server setup**:
restore /mnt/dietpi_userdata/blynk/server.properties with owner 'blynk' [998] and rights 0644

If **Datastreams (Diagrams/Graphs)** are used, Blynk creates an additional data directory at /mnt/dietpi_userdata/blynk/data/ linked to the users@email.com.
So you also have to restore the folder /mnt/dietpi_userdata/blynk/data/data with owner 'blynk' [998] and rights 0755
And all its subdirectorys named as the users email adresses (eg. users@email.com) with owner 'blynk' [998] and rights 0755
And all containing *.bin files with owner 'blynk' [998] and rights 0644

After restoring everything, dont forget to restart the blynk server with:
systemctl restart blynkserver
______________________________________

Joulinar, 2020-08-02:
Hi,

as far as I can see, DietPi is pulling the *.jar file. It should be latest version.

https://github.com/MichaIng/DietPi/blob/f1cfa3c843ebfa28b8edd8ba46dccd35742de52c/dietpi/dietpi-software#L4184

So maybe you could do a backup of your system and simple `dietpi-software reinstall 131`

Phil1988, 2020-08-02:
Which is basically the same that I did but as dietpi-software reinstall 131 instead of using uninstall + install right?

So is there a way to get the current installed version?
And is it enough to just replace the *.jar file instead of doing all the backup and restore stuff described above?

Joulinar, 2020-08-02:
`dietpi-software reinstall 131` will replace the *.jar. Usually user data should be stay untouched. However it's recomend to perform a `dietpi-backup` before. Just in case thinks go wrong.

MichaIng, 2020-08-02:
I added the update info to our online docs: https://dietpi.com/phpbb/viewtopic.php?p=5901#p5901
You can as well manually download and replace the .jar, but it is not something handy for the docs since one must take care to take the correct one for the own Java version + the reinstall will update the Blynk JS libary as well.

Not sure if the current version can be estimated from the command, usually it can. Just try it out, as a start with:
```
java -jar $G_FP_DIETPI_USERDATA/blynk/blynkserver.jar --help
```

Phil1988, 2020-08-03:
Thanks for your answers.
I will keep this in mind.

```java -jar $G_FP_DIETPI_USERDATA/blynk/blynkserver.jar --help```
returns
```
Exception in thread "main" cc.blynk.cli.UnrecognizedOptionException: Unrecognize                                                                                d option: --help
        at cc.blynk.cli.DefaultParser.handleUnknownToken(DefaultParser.java:217)
        at cc.blynk.cli.DefaultParser.handleLongOptionWithoutEqual(DefaultParser                                                                                .java:256)
        at cc.blynk.cli.DefaultParser.handleLongOption(DefaultParser.java:237)
        at cc.blynk.cli.DefaultParser.handleToken(DefaultParser.java:130)
        at cc.blynk.cli.DefaultParser.parse(DefaultParser.java:80)
        at cc.blynk.server.launcher.ArgumentsParser.parse(ArgumentsParser.java:5                                                                                7)
        at cc.blynk.server.launcher.ServerLauncher.main(ServerLauncher.java:53)
```

MichaIng, 2020-08-03:
Hmm, if this is the full list of command line options, then we have no chance: https://github.com/blynkkk/blynk-server/blob/master/server/launcher/src/main/java/cc/blynk/server/launcher/ArgumentsParser.java
Probably one of the var created files contain the version string. Otherwise go with mdate/cdate and compare with release date at https://api.github.com/repos/blynkkk/blynk-server/releases/latest?
Probably someone at https://github.com/blynkkk/blynk-server/issues has a good idea. Not sure about the HTTP API e.g.

MichaIng, 2020-09-27:
I'll mark this as closed. Feel free to reopen if required.

closed 2020-09-27 by MichaIng

ErikV7-official, 2021-10-31:
> Hmm, if this is the full list of command line options, then we have no chance: [...]

Here is a copy of the file https://github.com/Peterkn2001/blynk-server/blob/master/server/launcher/src/main/java/cc/blynk/server/launcher/ArgumentsParser.java for anyone who reads this post and needs it