MichaIng/DietPi #3706: The ability to see the contents of an install script for an app? Maybe in the help section?

CactiChameleon9, 2020-08-06:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- ... I want to update owncloud, but I have moved the directory, and was wondering if I could just adapt the installer script or use it to find the download location, but alas I have no idea where the scripts are stored.

#### Describe the solution you'd like:

- ... A link to each script in the "Help!" menu of dietpi software

#### Describe alternatives you've considered:

- ... Maybe a webpage on the dietpi website instead?

#### Additional context

- ...

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

Joulinar, 2020-08-06:
Hi,

basically all script are available on GitHub and you could check the code https://github.com/MichaIng/DietPi/tree/master/dietpi
on OS layer, they are store at `/boot/dietpi`

General overview of available software title https://dietpi.com/phpbb/viewtopic.php?f=8&t=5
Some OwnCloud info https://dietpi.com/phpbb/viewtopic.php?f=8&t=5#p47

CactiChameleon9, 2020-08-06:
I had actually moved the non data directory (the dir in /var/www) I also cannot find the install script for owncloud, is it inbuilt in dietpi-software script?

Joulinar, 2020-08-06:
Yes `dietpi-software` is responsible for all activities around install, reinstall and removal. But I guess you would need to change quite some lines if you like to change `/var/www/owncloud`.

Anyway, updating owncloud should be possible from outside the GUI

CactiChameleon9, 2020-08-06:
What binary is used for owncloud (armhl)?

Joulinar, 2020-08-06:
OwnCloud is a web application and installed from following source

https://github.com/MichaIng/DietPi/blob/f1cfa3c843ebfa28b8edd8ba46dccd35742de52c/dietpi/dietpi-software#L3417

CactiChameleon9, 2020-08-06:
Ah... that makes senses, I assumed it was compiled for a specific architecture - I don't know much about php :D . Thank you so much for you help! Very interesting how all of the install scripts are part of the dietpi-software script, I'd just assume they were separate and then executed when chosen.

Joulinar, 2020-08-06:
Well, I guess a design decision from the first days of DietPi to put everything in a single script. Not sure if @MichaIng has any thoughts about splitting it into smaller peaces 😃

MichaIng, 2020-08-07:
Whenever a software title has an internal updater, you use it instead of reinstalling via `dietpi-software` 😃. Also the "reinstall" will not touch the `/var/www/owncloud` content (only refresh dependencies and settings) with the same hint. That is as well true for ownCloud: https://doc.owncloud.com/server/admin_manual/maintenance/update.html

There have been thoughts to split the code, even a start, but in the end we didn't do it: #1163
Large effort with not only benefits but downsides as well. I always wanted to merge and sort some parts of the code to make it cleaner and better readable which then as well is a better basis for further coding changes, but cannot say when I find time for it.

CactiChameleon9, 2020-08-07:
OK, makes sense. Thanks for the comment, its really interesting how you were mentioning block size and how the scripts are stored in ramfs, it just shows how much thought has gone into this amazing project! In the end I just reverted back the changes I made, updated and then used a symbolic link instead to move the dir to where I want it.

MichaIng, 2020-08-07:
> OK, makes sense. Thanks for the comment, its really interesting how you were mentioning how block sizes effect individual file sizes and how the scripts are stored in ramfs, it just shows how much thought has gone into this amazing project!

Many thanks, sometimes things might be even overthought, but I'm perfectionist 😄.

As of moving the Nextcloud data dir: Jep, using a symlink is the easiest solution. Cleaner is really moving it and adjusting config file + database, but see my HowTo on Nextcloud forum for details: https://help.nextcloud.com/t/howto-change-move-data-directory-after-installation/17170?u=michaing
On DietPi also adjust SOFTWARE_OWNCLOUD_DATADIR in dietpi.txt then, just to be sure that reinstalls will always have the right info. The script does not check config.php entry first, if present, before using SOFTWARE_OWNCLOUD_DATADIR, but that would be probably a bid safer in case it got changed manually.

CactiChameleon9, 2020-08-07:
My data dir is actually still in the same place currently its the install dir that I changed(the one in /var/www/), but thanks a lot I may bother soon to move that dir off the sdcard :D

And I am a bit of a perfectionist too... I often end up tweaking ui when there is actual code to be done :D.

MichaIng, 2020-08-07:
Ah sorry, I misunderstood, the website directory you mean. Yes that is natively not possible to change through DietPi, we place all websites into /var/www. A symlink is then really the best as it will even survive dietpi-software reinstalls.

CactiChameleon9, 2020-08-07:
Yeah. I like mine in /var/www/html to keep things organised I also renamed the admin from pi hole to pihole

MichaIng, 2020-08-08:
Ah yes Pi-hole strictly requires to be placed at `/var/www/html`, that makes it a bid unclean when everything else is fine in `/var/www`.

closed 2020-09-27 by MichaIng