MichaIng/DietPi #3707: Lighttpd vhost problem

PanosssD, 2020-08-07:
Hello.
Dietpi on PC.
I use it for wordpress web server with Lets Encrypt etc.
I just tried to add a vhost (i already have 3 vhosts working ok) but after setup (db,vhost on conf-enabled,ssl with certbot) when i visit the site it says that ssl is invalid because it reads ssl from letsencrypt.conf and not from e.g example.com.conf.
Anyone knows why is it happening.
And i want to test if issue happens because of mods in lighttpd.conf
Did anyone have the default lighttpd.conf file for dietpi

MichaIng, 2020-08-08:
Many thanks for your report.

Your run all vhosts on different ports? `/etc/lighttpd/conf-enabled/letsencrypt.conf` is the 443 vhost, hence will always be in charge when a request to port 443 is incoming, in no other case.

If you enabled it, there is `/etc/lighttpd/conf-enabled/redirect.conf` which redirects HTTP to HTTPS for all vhosts/ports but it should preserve the vhost/port. But I never tested it myself to be true, hence you might want to test if the issue appears as well if you explicitly connect via `https://`.

PanosssD, 2020-08-08:
All vhosts conf files start with **$HTTP["host"] == "example.com" {**
they are on the same (443) port but it goes by hostmane e.g you look for  

- (public IP) = you get nothing
- example1.com = you get example1.com page
- example2.com = you get example2.com page
- etc

in each conf file there is

- server.document-root = "/var/www/.../.../wordpress" # Document-root of the webserver
- accesslog.filename = "/var/log/lighttpd/example1_access.log" # Web server Access log file
- server.errorlog = "/var/log/lighttpd/example1_error.log" # Web server Error log file
- ssl.pemfile = "/etc/letsencrypt/live/example1.com/combined.pem"
- ssl.ca-file = "/etc/letsencrypt/live/example1.com/fullchain.pem"

example1,2,3 .conf files are placed inside conf-enabled directory 

Is there any place to find default files that dietpi uses for:

- /etc/lighttpd/conf-enabled/redirect.conf
- /etc/lighttpd/conf-enabled/letsencrypt.conf
- /etc/lighttpd/lighttpd.conf
???
I want to test if the problem is happening e.g. from bad configuration that i 've made, limitation of lighttpd or dietpi
And i want to remind you that i 've already setup 3 vhosts working properly. On 4th and 5th problem starts to appear.

MichaIng, 2020-08-08:
Okay so since all your vhosts listen on 443, `letsencrypt.conf` is used for all of them. I'm not 100% sure which directive overrides which one (`$SERVER["socket"]` vs `$HTTP["host"]`) or if its only a question of alphabetical order.

If you have HTTPS redirect active, easiest is probably to:
- Comment those two lines in `letsencrypt.conf`: https://github.com/MichaIng/DietPi/blob/3f63e4d7497fd4dcae62a530bd2b2e3425a9db81/dietpi/dietpi-letsencrypt#L142-L143
So it contains only the SSL config that shall be used across all vhosts.
- Wrap your vhosts into `$SERVER["socket"] == ":443"`, so they're HTTPS-only, while `redirect.conf` redirects all plain HTTP requests. Although you say "`they are on the same (443) port`", so this seems to be already the case?

PanosssD, 2020-08-08:
I make some test before your answer.
I read the so obvious README file inside /etc/lighttpd/conf-available/
which says: 

`       If you want to create your own files they names should be
        build as nn-name.conf where "nn" is two digit number (number
	is used to find order for loading files)`

So i 've renamed all files for vhosts including redirect & letsencrypt.conf with **"nn-"** at start **and thats it!!!**
And to mention: the problem doesn`t exist when 3 vhost enabled but after 4th vhost it reads ssl (only for vhost 4, the others are working ok) from letsencrypt.conf causing invalid certificate error.
BTW thank you for your help!

MichaIng, 2020-08-08:
Okay so this means the alphabetical order of your configs was like:
1. example.conf
2. letsencrypt.conf
3. next.example1.conf
4. next.example2.conf
5. next.example3.conf

`next.exampleN.conf` vhosts were parsed after `letsencrypt.conf`, hence their `ssl.pemfile` + `ssl.ca-file` were effective.
`example.conf` vhost was parsed before `letsencrypt.conf`, hence its `ssl.pemfile` + `ssl.ca-file` was overwritten and ineffective?
_______
Those file names were never touched since the initial `dietpi-letsencrypt` creation while meanwhile we follow the `nn-` priority-naming convention. It makes sense to give this global `letsencrypt.conf` a pretty low priority (i.e. `00-letsencrypt.conf` if no module ships related directives with higher priority) so that any custom vhost/config overrides it.
`redirect.conf` on the other hand needs to have highest priority (i.e. `98-redirect.conf`, to still allow admin overrides via `99-`) to effectively redirect each and every non-HTTPS request.

PanosssD, 2020-08-08:
Exactly!!! The names were:

1. blog.example (not working)
2. eshop.example (not working)
3. letsencrypt
4. mail.example (working)
5. (naked domain starting with s) (working)
6. etc (working)

You can use `50-redirect.conf` if anyone wants an ascending order for vhosts there is some space before and after, and if someone don't bother 50 or 99 he goes either way after redirect.conf

MichaIng, 2020-08-25:
Done: https://github.com/MichaIng/DietPi/pull/3734
Renaming those on running systems, accidentally merged into dev directly 😄: https://github.com/MichaIng/DietPi/commit/5e993ecabf056d2fcde57ab0f62547c50021a340

closed 2020-08-31 by MichaIng