MichaIng/DietPi #3708: Unifi Controller

Gill-Bates, 2020-08-09:
The Unifi Controller is a very popular tool on the Raspberry. I found a guide on the net how to install it:

https://smarthome-training.com/de/unifi-controller-raspberry-pi/

Does anyone have the time and inclination to add the Unifi controller to the software archive so that it can be easily installed via `dietpi-software`?

taduo, 2020-09-12:
Also looking for this to be available :)

Gill-Bates, 2020-09-17:
I would to bring this topic up again. Here comes the Steps to Install Unifi on Dietpi. To add this to the Software Catalouge should be not a big deal:

# Install correct openJDK
apt install openjdk-8-jdk-headless
echo "JAVA_HOME=/usr/lib/jvm/java-8-openjdk-armhf" | sudo tee -a /etc/default/unifi

# Add Repo
echo 'deb http://www.ubnt.com/downloads/unifi/debian stable ubiquiti' | sudo tee /etc/apt/sources.list.d/100-ubnt-unifi.list
sudo wget -O /etc/apt/trusted.gpg.d/unifi-repo.gpg https://dl.ubnt.com/unifi/unifi-repo.gpg

# Install Unifi
# Install correct openJDK
sudo apt install openjdk-8-jdk-headless

# Add Repo
echo 'deb http://www.ubnt.com/downloads/unifi/debian stable ubiquiti' | sudo tee /etc/apt/sources.list.d/100-ubnt-unifi.list
sudo wget -O /etc/apt/trusted.gpg.d/unifi-repo.gpg https://dl.ubnt.com/unifi/unifi-repo.gpg

# Install Unifi
sudo apt update; sudo apt install unifi

Joulinar, 2020-09-17:
Hi,

many thanks for your request. Just one comment from my side. I don't think it will be implemented if a dependency is `openjdk-8` because `openjdk-8` is not available on standard Debian repository for Debian Buster systems. Yes it's part of Raspbian Repository but than it would be dedicated to one set of SBC device. Even I'm not sure how it is on Raspberry OS 64bit, as the Foundation will switch to standard Debian repository with it. But it's up to @MichaIng

rhkean, 2020-11-17:
there is a work around for this.  I am running the unifi controller on my dietpi raspberry pi 4, but it has to be done very early in the dietpi setup.

I found the reference to this install script on the Ubiqui forums here:  [https://community.ui.com/questions/UniFi-Installation-Scripts-or-UniFi-Easy-Update-Script-or-UniFi-Lets-Encrypt-or-UniFi-Easy-Encrypt-/ccbc7530-dd61-40a7-82ec-22b17f027776](https://community.ui.com/questions/UniFi-Installation-Scripts-or-UniFi-Easy-Update-Script-or-UniFi-Lets-Encrypt-or-UniFi-Easy-Encrypt-/ccbc7530-dd61-40a7-82ec-22b17f027776)
The author's main website is here:  [https://glennr.nl/](https://glennr.nl/)

1. perform a basic install of dietpi.  I booted a fresh install of dietpi, ran dietpi-software to configure the basics, like network setting and such, but did not select any software to install

2.  then do the following:  (this will download and run an installation script that will download and configure everything needed to install and run the unifi controller.
`wget "https://get.glennr.nl/unifi/install/unifi-6.0.27.sh" -O install-unifi-6.0.27.sh && chmod +x install-unifi-6.0.27.sh && ./install-unifi-6.0.27.sh`

3.  NOTE:  if you run the script a second time, it will detect the installation and give you an option to update if there is a newer version available.  When I did this a couple weeks ago, only the 6.0.27 version was available for download, but running the script a 2nd time upgraded me to 6.0.28.  (there may be a newer version at this time)

4.  after the script completes, then you can run dietpi-software to install other applications as needed.  **NOTE:  I do not run any other java-based applications on my dietpi server, so I can not say whether this will cause any version compatibility issues.**

Gill-Bates, 2020-11-17:
@rhkean Thanks for sharing your Script. I would prefer to using the standard-way with Board-Instruments of Linux as desribed above. Then with this solution, an `apt dist-upgrade` will also auto-upgrade the Unifi-Controller - without additional Scripts.

Joulinar, 2020-11-17:
well the script is still trying to install JAVA8. Not sure if this is an option for nonRPi systems

rhkean, 2020-11-17:
I would prefer that, too. Unfortunately, Ubiquiti's controller code
requires java 8 and there are several other packages that I ran into
problems with... This script worked, though.

To be clear, this is not my script. I found it on Ubiquiti's users forum.

-rob

rhkean, 2020-11-17:
I'm not certain about what boards are supported, tbh.

You would have to read the scripts to find out. I do recall from my quick
glance a while back, that it does as much 'apt-get' as it can without
breaking anything.

Joulinar, 2020-11-17:
> To be clear, this is not my script. I found it on Ubiquiti's users forum.

I know. I just tried to highlight that the script will not change the requirement of JAVA8. And this limit the possible use case.

Currently I'm trying to run the script on one of my VM's but it seems to be stuck somewhere during all the apt install steps 🤔

MichaIng, 2020-11-18:
Good find. That is somehow an issue as we already dropped several other software titles from Buster on, which depend on Java 8. Yes Raspbian contains it, but with upcoming Raspberry Pi OS 64-bit (and my in production testing Debian-based ARMv7 image), there will be mostly only RPi 1 and Zero left (+existing systems of course) which use Raspbian. And using binaries from old-old-stable (from summer on) repositories is not something I'm keen to do, although we did in other cases (e.g. libssl1.0.0).

Would be interesting if there are any plans to make it Java 11 compatible?

rhkean, 2020-11-18:
Agreed.. for as popular as their devices are, I'm a bit dumbfounded that
they haven't brought it current, especially since they've brought other
dependencies current like mangodb.

Gill-Bates, 2020-11-18:
Or only applications that are compatible for the installed platform appear in the software catalog. :-)

MichaIng, 2020-11-18:
> Or only applications that are compatible for the installed platform appear in the software catalog. :-)

Jep, that is how it is currently. However, to implement software for a very limited set of platforms is less of a pleasure of course, specially when it's about outdated dependencies which will break them on even the small left platforms earlier or later 😄.

So for now it would be all Stretch platforms and Raspbian. Since it is famous for RPi which in combination which Raspbian still covers ~50% of our users, that should be worth enough to implement.

Fireman1224, 2021-08-12:
Does anyone know if the Unifi Controller has been resolved to work with the latest version of DietPi?

Also, would it be possible to run PiHole and PiVPN/WireGuard on the same unit (RPi 3 B+) with the Controller or should I keep the controller separate?

Thanks in advance!

Joulinar, 2021-08-12:
Well if I'm not mistaken, issue was the requirement of JAVA8 which is quite old and outdated.

Fireman1224, 2021-08-13:
So if I install Openjdk8, I should be able to run all 3 on one device with DietPi?  I'm a noob, so just trying to understand...

Joulinar, 2021-08-13:
PiHole + WireGuard shouldn't have any influence on Unifi Controller software. Means for you to mange is the installation of Unifi Controller software 😉

Gill-Bates, 2021-08-13:
Here my installation steps:

# Source: https://smarthome-training.com/de/unifi-controller-raspberry-pi/

# Install correct openJDK
sudo apt install openjdk-8-jdk-headless

# Add Repo
echo 'deb http://www.ubnt.com/downloads/unifi/debian stable ubiquiti' | sudo tee /etc/apt/sources.list.d/100-ubnt-unifi.list
sudo wget -O /etc/apt/trusted.gpg.d/unifi-repo.gpg https://dl.ubnt.com/unifi/unifi-repo.gpg

# Install Unifi
sudo apt update; sudo apt install unifi

# Change Max Heap (only Raspberry 3 and below)
nano /usr/lib/unifi/bin/unifi.init
JVM_MAX_HEAP_SIZE=512M # Raspberry 3+

systemctl restart unifi.service

Joulinar, 2021-08-13:
AGH is available via DietPi software catalouge. There is no need to install it manually. As well why not using PiHole? What is the relation to Unifi??

Gill-Bates, 2021-08-13:
> AGH is available via DietPi software catalogue. There is no need to install it manually. As well why not using PiHole? What is the relation to Unifi??

AGH is much better than Pihole. Here is a compensation matrix: https://github.com/AdguardTeam/AdGuardHome

Joulinar, 2021-08-13:
well there is no relation to Unifi so to say. You probably would need to make to clear. Otherwise it could confuse people. Next to this you are referring a comparison done by AGH themselves. I guess they never will come to a point where PiHole wins. Both AdBlocker have their pros and cons. It depends of your personal opinion at the end.

MichaIng, 2021-08-13:
Of course AGH likes AGH more than Pi-hole. I had closer looks into both and clearly prefer Pi-hole for its use/inclusion of well maintained separate open software solutions and related higher customisation possibilities. AGH is an all-in-one single binary, which may be preferred and looks more streamlined for less experienced users, but is means that more experienced users or custom needs are more difficult to achieve. Also they partly need to re-invent the wheel, which means potentially higher risk of security issues related to that. The matrix mainly contains features which are extremely easy to add to Pi-hole, via well-known trusted long-existing software:
- DoT/DoH via Unbound
- DNSCrypt via DNSCrypt-proxy
- Parental blocking, phishing and malware domain blocking is also in AGH only done via upstream blocklists or upstream DNS, so you can simply use `1.1.1.3` as upstream DNS (Cloudflare with adult + fishing blocking) or AdGuard itself (`94.140.14.14`) or any other that fits better to the personal needs: https://en.wikipedia.org/wiki/Public_recursive_name_server
- Access settings and running without root privileges is possible/done by Pi-hole as well, there the matrix is simply wrong.

I can assure anyone who is reading this, that there is no objective/substantial preference possible, the core features are identical. It is finally a matter of taste, and whether you have specific network/webserver setup needs (then use Pi-hole) or not. And do not trust developers own comparison tables: They don't aim to lie of course, but naturally such tables are one-sided and not objective.

Joulinar, 2021-08-13:
I guess we are moving away from original topic. Let's close the discussion on AGH and PiHole at this point.

Fireman1224, 2021-08-13:
I have PiHole set up currently, so migrating to another PiHole would be easier especially since I have Samsung TVs that were a nightmare to set up and get its TVplus working correctly.  

My main goal is to get as few devices running on my network as possible (I'm currently running four Pi 3B+'s). 

My main questions are is it possible to get all 3 on one device (which it appears it it) and is it SMART to have the Unifi Controller, Wireguard, and network-wide ad blocking on the same device...

Joulinar, 2021-08-13:
yes that should be possible to run all on the same RPi.

MichaIng, 2021-08-13:
Yes, they should not interfere at all and Pi-hole is extremely lightweight (AGH as well, for completeness, DNS generally is pretty light, compared to HTTP/PHP web applications e.g.), so does not add relevantly to system load, even on RPi Zero.

Btw, the Java runtime environment should be sufficient:
```sh
 apt install openjdk-8-jre-headless
```
This works on RPi 32-bit/Raspbian images only, since Debian Buster and above ships only Java 11 and above. Would be interesting to test it with Java 11 btw, probably they shipped some update regarding this?

Fireman1224, 2021-08-13:
I guess the nagging question for me now is should I run all 3 on one device or should it be broken into multiple devices from a network stability/security standpoint.

I really appreciate everyone's assistance!!

Fireman1224, 2021-08-13:
@MichaIng Good point on the Java version as I was planning on using the new bullseye release from you when available, so if it only works on the 32bit version that might be a problem???

Joulinar, 2021-08-13:
Bullseye will be available as 32bit version as well. The challenge is with the outdated JAVA8, since Debian Buster and above ships only Java 11 and above.

Fireman1224, 2021-08-13:
@Joulinar I figured they would release it in 32bit as well, I had just planned on using the 64bit version...

Joulinar, 2021-08-13:
There isn't a real benefit using 64bit. Currently there are more downside as Raspberry OS 64bit has not jet been released as stable. But in case of PiHole/WireGuard, it doesn't matter much. Or do you have a specific reason to use 64bit?

MichaIng, 2021-08-13:
If Java 8 is still required for Unifi Controller, then it can only work on a 32-bit ARMv6 Raspberry Pi image, also on Bullseye. New image is available (not yet published): https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Bullseye.7z

But I would first test it with Java 11 or Java 17 (when using the Bullseye image), as the info about Java 8 being required is from a pretty old guide:
```sh
# On Bullseye
 apt install openjdk-17-jre-headless
# On Buster, or if Java 17 does not work:
 apt install openjdk-11-jre-headless
# else
apt install openjdk-8-jre-headless
```
And another note on newer Java versions: The ARMv6 RPi models (RPi 1 and Zero) do not support Java 8, at least when using the Debian builds with the VM engine. We'll test 3rd party OpenJDK builds to hopefully find newer Java 11 and/or 17 builds running on ARMv6 as well.

Fireman1224, 2021-08-13:
@Joulinar  Not really, just figured it would be better in the long term.

@MichaIng According to the website, Unifi Controller 6.2.25 requires Java 8, Java 9 and later are not yet supported.

I'm currently 2 debian versions behind, so I was going to upgrade to the latest once it is released (I was told release date is 08/14/2021).  I may just migrate PiHole/Wireguard to a new DietPi once bullseye is released and upgrade the controller once they support java 9+.  Maybe this will give me time to do a little more research.

I honestly thought 64bit was a better long term plan for the 64bit hardware, is that not correct?

MichaIng, 2021-08-13:
> I honestly thought 64bit was a better long term plan for the 64bit hardware, is that not correct?

Hard to say, currently it has overall more downsides than upsides, mainly because of still missing software and also (userland) firmware packages/maintenance from Raspberry Pi repository/maintainer side. AFAIK also from our software options, there are more which support armhf/32-bit while not supporting arm64/64-bit then the other way round. How things develop, especially how fast, towards better aarch64 support in general of for Raspberry Pi explicitly is a good question 🤔.

Joulinar, 2021-08-13:
Do you like to perform an in-place upgrade or a new installation? Theoretically you could already install Bullseye as we are days before the official release. Usually there are no surprises anymore as Bullseye software catalogue is already frozen since a while.

Fireman1224, 2021-08-13:
It sounds like the 32bit is better for noobs like myself as there is more support for when things may go wrong...

As for the upgrade vs new install, I was just looking to clean up trial and error installs with a clean install for dedicated use, as well as moving from a microSD card to a USB flash drive for stability.

Joulinar, 2021-08-13:
in this case, do a fresh install directly on the USB drive. If you like, you could use our brand new ARMv7 Bullseye image https://dietpi.com/downloads/images/DietPi_RPi-ARMv7-Bullseye.7z

Migrating WireGuard should be easy. Just do an installation on the new system and copy all config files from `/etc/wireguard/`.

Fireman1224, 2021-08-13:
Can you point me to the instructions to install directly on a USB drive?  I thought it had to be installed/configured on a microSD and then migrated to the USB...

Fireman1224, 2021-08-13:
Very nice.  I will try to get this going this weekend!  Thanks!!

Fireman1224, 2021-08-14:
> on the tool you use the flash the image, simply select the USB drive as
> target. Usually Raspberry Pi 3B+ supports USB mass storage boot out of the
> box.
> https://www.raspberrypi.org/documentation/computers/raspberry-pi.html#usb-mass-storage-boot

So... I have installed the new version that you guys forwarded me to and I
have bricked 2 USB thumb drives.  Everything works as intended until it
reboots.  After rebooting, the drive is no longer recoverable as it is not
seen as having any information on it.  It's almost like the partition
information is corrupt... Any ideas?

Thanks,

Dee McKeel
***@***.***

MichaIng, 2021-08-14:
You mean our RPi Bullseye image? It does not come up on reboot? Is there any output on the screen? And what happens if you attach the drive to another Linux system and try to mount the partitions there?

This is quite unexpected since there is AFAIK currently only a single APT package upgraded. On very early boot stage, the root partition and filesystem are expanded, but that should not break anything, especially when bootup continues afterwards and first run setup goes through.

To check, on first boot session, whether everything went fine, the log files in `/var/tmp/dietpi/logs` can be checked, which also includes the partition and filesystem expansion.

Fireman1224, 2021-08-14:
Yes. Installed it on a 32gb nand flash drive, used ssh to complete installation with Fail2Ban, PiVPN, and Pi-Hole. Had static IP set on device and on the router. I was able to reboot after the Pi-Hole install and continue with the rest of the apps, but after completing the install and the command prompt going back to root, if I reboot the drive is toast. Is not recognized if I boot from the microSD card and insert the flash drive. 

I’m going to try to install to another microSD and test that, but I don’t want to keep bricking flash drives. It literally says that it’s a drive of 0b on a windows machine and doesn’t show at all on the rasbian device. 

Thanks,

D L McKeel, Jr.

MichaIng, 2021-08-15:
> Is not recognized if I boot from the microSD card and insert the flash drive.

What do you mean by not recognised? Is the drive itself recognised, i.e. shown by `lsblk`, are the contained partitions recognised (same command), are the filesystems seen at least, so that you can run `fsck` on them? And then of course are there any kernel errors shown?
```sh
dmesg -l emerg,alert,crit.err
```

If one reboot is possible, after installing Fail2Ban PiVPN and Pi-hole, then it obviously is one of the "rest of the apps" which causes the issue, or e.g. an issue with voltage when due to additional services the CPU load and disk writes raise.

Fireman1224, 2021-08-15:
Here is the result of ````lsblk```` after booting from the microSD with the
USB drive installed.

***@***.***:~# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 14.9G  0 disk
├─mmcblk0p1 179:1    0 41.8M  0 part /boot
└─mmcblk0p2 179:2    0 14.8G  0 part /
***@***.***:~#````

The USB drive is not listed, only the 16gb microSD.

Should we move the conversation to a different thread?

Thanks
Dee McKeel
***@***.***

Joulinar, 2021-08-15:
Can you share `dmesg` as well. It might be a longer output. Probably you could copy it into a file and attached it manually to the GitHub post.

MichaIng, 2021-08-15:
And yes, it makes sense to open a new issue for this 👍.

Fireman1224, 2021-08-28:
Ok, I have tried installing the Unifi controller on a new Pi4 8gb with dietpi bullseye on a 32gb USB 3.1 SanDisk Ultra Fit drive.  When trying to run:
```
apt install openjdk-8-jre-headless
```
I get this:
```
root@DietPi:~# apt install openjdk-8-jre-headless
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package openjdk-8-jre-headless
```

Any idea how to correct this?  I get the same E: error when I try:
```
root@DietPi:~# echo 'deb http://www.ubnt.com/downloads/unifi/debian stable ubiquiti' | sudo tee /etc/apt/sources.list.d/100-ubnt-unifi.list
sudo wget -O /etc/apt/trusted.gpg.d/unifi-repo.gpg https://dl.ubnt.com/unifi/unifi-repo.gpg

deb http://www.ubnt.com/downloads/unifi/debian stable ubiquiti
--2021-08-29 00:15:50--  https://dl.ubnt.com/unifi/unifi-repo.gpg
Resolving dl.ubnt.com (dl.ubnt.com)... 99.86.224.96
Connecting to dl.ubnt.com (dl.ubnt.com)|99.86.224.96|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1196 (1.2K)
Saving to: ‘/etc/apt/trusted.gpg.d/unifi-repo.gpg’

/etc/apt/trusted.gpg.d/unif 100%[========================================>]   1.17K  --.-KB/s    in 0.003s

2021-08-29 00:15:50 (386 KB/s) - ‘/etc/apt/trusted.gpg.d/unifi-repo.gpg’ saved [1196/1196]

root@DietPi:~# sudo apt update; sudo apt install unifi
Hit:2 https://deb.debian.org/debian bullseye InRelease
Get:3 https://deb.debian.org/debian bullseye-updates InRelease [36.8 kB]
Get:4 https://deb.debian.org/debian-security bullseye-security InRelease [44.1 kB]
Get:5 https://deb.debian.org/debian bullseye-backports InRelease [39.3 kB]
Hit:6 https://archive.raspberrypi.org/debian bullseye InRelease
Get:1 https://dl.ubnt.com/unifi/debian stable InRelease [3023 B]
Get:7 https://dl.ubnt.com/unifi/debian stable/ubiquiti armhf Packages [712 B]
Fetched 124 kB in 3s (37.6 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 unifi : Depends: mongodb-server (>= 2.4.10) but it is not installable or
                  mongodb-10gen (>= 2.4.14) but it is not installable or
                  mongodb-org-server (>= 2.6.0) but it is not installable
         Depends: mongodb-server (< 1:4.0.0) but it is not installable or
                  mongodb-10gen (< 4.0.0) but it is not installable or
                  mongodb-org-server (< 4.0.0) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Please help and thanks in advance!!

Joulinar, 2021-08-29:
I guess MongoDB is not available on Debian at all and JAVA8 is not available on Debian Bullseye.

Fireman1224, 2021-08-29:
Ugh.... Does this mean that the Unifi Controller can't be installed on Bullseye?

Joulinar, 2021-08-29:
Main issue is MongoDB as you are not able to install it on Raspberry Pi at all. Doesn't matter if it is Buster or Bullseye. Because the MongoDB repository does not support ARM platform.

Fireman1224, 2021-08-29:
How can I fix? What do I need to do?

Joulinar, 2021-08-29:
Maybe I was not clear enough. You can't install MongoDB on ARM as it is not supported.

Maybe you can use a search engine to find a guide describing a way around.

Fireman1224, 2021-08-29:
OK, no reason to be rude, I know very little about Linux...  I guess installing a version other than bullseye?  Just trying to figure all this out...  There must be a way since I'm running a controller on stretch on a pi 3.  simply trying to upgrade...  And by the way, I have been using a search engine.  I'd much rather do that than ask for help...

Joulinar, 2021-08-29:
Well Stretch is quite old and maybe condition changed in meantime. Means, maybe MongoDB was not required before but is it now.

Fireman1224, 2021-08-29:
I'm pretty sure MongoDB has always been required, kinda like Java 8... All the tutorials I've found reference the same install procedures that now don't work... Just trying to understand.  Maybe I need Raspian Lite vs dietpi?  not sure, just grasping at straws...  I really like the performance from dietpi...

Joulinar, 2021-08-29:
DietPi is not an own OS, it is a set of shell scripts on top of base image. In case of Raspberry Pi, base image used is Raspberry OS.

If you have a running Stretch system, you should be able to update to Buster as a first step.

Gill-Bates, 2021-08-29:
I run it with no issues …

[image]

Joulinar, 2021-08-29:
I read thru the internet and it seems older version of MongoDB are able to run on Raspberry OS but somewhere in 2000 this changed. 

There are a couple of entries on MongoDB support forum requesting support for ARM64. found this as well https://www.mongodb.com/community/forums/t/mongod-4-4-on-rpi4-aarch64-arch-for-unifi/10378

Gill-Bates, 2021-08-29:
Maybe this could help?
`
wget http://ftp.br.debian.org/debian/pool/main/o/openjdk-8/openjdk-8-jre-headless_8u252-b09-1~deb9u1_arm64.deb`

Joulinar, 2021-08-29:
This is a Debian 9 (Stretch) package, something I would not recommend on Buster or Bullseye. To install Java 8 you could use `adoptopenjdk` repository

https://adoptopenjdk.net/installation.html?variant=openjdk8&jvmVariant=hotspot#linux-pkg

[14 more replies]