MichaIng/DietPi #3709: Cuberite missing

symbios24, 2020-08-10:
### Creating a bug report/issue
Hallo Cuberite software is missing from the latest dietpi image in an Rpi4 i havent try though in my Rpi3 .
#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2020-08-10:
Hi,

I checked on my system and there it's still available 

```
root@DietPi4:~# dietpi-software list | grep cub
id 52 | =0 | cuberite: minecraft server with web interface (c++) | | https://dietpi.com/phpbb/viewtopic.php?p=2068#p2068
root@DietPi4:~#
```
Are you using the normal 32bit image or the 64bit Raspberry OS beta?

symbios24, 2020-08-10:
32 bit strange

Joulinar, 2020-08-10:
what does `dietpi-software list | grep cub` gives you back?

symbios24, 2020-08-10:
Im using the image released in 20-06-14
Tried to install cuberite manual with the dietpi-software install 52 and it says that cuberite is not supported in aarch64 
Is it possible with the dietpi-update i did in the beginning converted the image to 64 bit??

Joulinar, 2020-08-10:
probably you download the 64bit Beta image? Do you still have the file you download? if yes, whats the name of the file?
It's not realy possible to fully convert a 32bit system into a 64bit. However on a 32bit system, you could force kernel to be 64bit by setting `arm_64bit=1` within `/boot/config.txt`. If you system is a 32bit one, just remove the entry and do a reboot.

Joulinar, 2020-08-10:
pls remove `arm_64bit=1` within `/boot/config.txt` and do a reboot

symbios24, 2020-08-11:
Upon installing  Cuberite in my Rpi3 i got an 404 not found error for the Cuberite.tar.gz file 
Can you check it?

Joulinar, 2020-08-11:
Cuberite installation source address changed. You could adjust it during `dietpi-software` installation. Once you hit by the error, you will get an error handle to deal with the issue. Pls select `change command`. Now you can adjust the download URL. This would need to be done twice. First error is for checking, second for the actual download. Pls remove always the entire line and replace as follow:

1. `wget --spider -t 1 -T 10 https://download.cuberite.org/linux-armhf-raspbian/Cuberite.tar.gz`
2. `wget https://download.cuberite.org/linux-armhf-raspbian/Cuberite.tar.gz -O 52.tar`

it's already being fixed on DietPi dev branch and will be implemented with next release

https://github.com/MichaIng/DietPi/blob/8d8cfabbbb988a7a3f3dff3f9d09453369011754/dietpi/dietpi-software#L4990

symbios24, 2020-08-11:
Thanks
Is there going to be an 64 bit version of cuberite?

Joulinar, 2020-08-11:
best to my knowledge it's not yet available by cuberite themselves. However you could try to compile yourselves.  

https://github.com/MichaIng/DietPi/issues/3664#issuecomment-657031119

MichaIng, 2020-08-11:
Jep checkout their download page: https://cuberite.org/
x86 and RPi packages only, the latter works on all 32-bit ARMs

But in theory, since it's quite self-contained, you could try to run the armhf package on 64-bit ARM as well. This only causes issues when it would depend on 64-bit libraries installed in the system.

Joulinar, 2020-08-11:
I guess the initial issue has been fixed? Right?

symbios24, 2020-08-12:
Yes thank you

closed 2020-08-12 by symbios24