MichaIng/DietPi #3710: Pi-Hole & Resolv.conf: Bad Nameserver

bccarbone, 2020-08-11:
**Environment:**

OS = DietPi v6.31.2
Pi-Hole Version = 5.1.1

**Error:**

I have been running Pi-hole for the past month, and as of yesterday, I have a problem.

From the command line, I did a pihole -g and get the error "DNS resolution is currently unavailable"

**Troubleshooting:**

After searching on the forums, it looks like it has to do with /etc/resolv.conf

There's a SINGLE line that says nameserver 192.168.1.1 (which references my router).

I know if I change it to 127.0.0.1 and save it works fine.

**Help Needed:**

The problem is when I REBOOT, the nameserver reverts BACK to 192.168.1.1. How do I fix this so it always stays to 127.0.0.1 ? This used to work FINE until yesterday. Unsure as to what changed....

**Additional Note:**  I reached out to Pi-Hole's support FIRST and they recommended I run it by over here.

Thank you for any help, it is greatly appreciated!

bccarbone, 2020-08-11:
Reference to my Pi-Hole open issue is here:  https://discourse.pi-hole.net/t/pihole-g-dns-resolution-is-currently-unavailable/37046

Joulinar, 2020-08-11:
Hi,

Many thanks for your report. What is the settings in `dietpi-config` for your network adapter? Usually DNS settings should be done there.

MichaIng, 2020-08-11:
Yep, change it via dietpi-config or in /etc/network/interfaces directly. /etc/resolv.conf is configures on startup based on those, either the static one or via DHCP (which you should not use in combination with Pi-hole).

bccarbone, 2020-08-11:
Thank you very much, really appreciate the responses.  This did the trick.

bccarbone, 2020-08-11:
I posted this on the forum by accident first, would you mind deleting the thread please?  https://dietpi.com/phpbb/viewtopic.php?f=11&t=7971

Many thanks.

Joulinar, 2020-08-11:
just to confirm, it's working now? If yes, I'm gonna close the ticket :)

bccarbone, 2020-08-11:
Yes it is working now, many thanks.  Also see comment above, I also posted this on your forum and realized I should have done so here.  Would you mind deleting my forum thread please?  Thank you!

Joulinar, 2020-08-11:
well doesn't matter where you post it. We will support both ways.

closed 2020-08-11 by Joulinar