MichaIng/DietPi #3711: RPi-zero-w: Iterating previous commands using up/down arrow keys has become very slow and unresponsive

maartenlangeveld, 2020-08-11:
I already posted in raspberrypi/firmware (https://github.com/raspberrypi/firmware/issues/1450) but I am not sure what causes the erratic behaviour. It started around the time that the raspberry kernel (v5+) was installed. Just to check if has nothing to do with DietPi. Does anyone else recently experience problems iterating commands in shell ?

OS/system
```
Linux M-STREAMER 5.4.51+ #1327 Thu Jul 23 10:53:06 BST 2020 armv6l GNU/Linux
DietPi v6.31.2
```

hardware
`Raspberry Pi Zero W Rev 1.1`

configuration:
`Headless. Access via xterm (utf-8): RPi: Dropbear SSH Server, windows PC: Bitvise SSH Client`

Joulinar, 2020-08-11:
Hi,

I don't have issues with using up/down arrow keys on my RPi3B+ as well as RPi4B. Both running kernel 5.4.51-v7. Unfortunately I don't have a Pi Zero to check.

MichaIng, 2020-08-13:
Is it probably #3690?

MichaIng, 2020-08-29:
Marking as closed for now. We were not able to find out the reason, probably the combination of SSH client + server + network connection, or indeed kernel-related. Hopefully it solves itself after the still relatively new RPi 5.4.y kernel was been settled a bid.

closed 2020-08-29 by MichaIng