MichaIng/DietPi #3712: Can't change bazarr user name

Dr0bac, 2020-08-13:
Hi, i have problem like it is described in this issue:

https://github.com/morpheus65535/bazarr/issues/1061

in bazarr wiki is says after installation that i need to change ownership of files where bazarr is installed, in my case in `/opt/bazarr`

it also says that my user must have home directory.

so i created new user bazarr `adduser username` and added that user to dietpi group `usermod -g dietpi bazarr`

tried with `sudo chown -R $bazarr:$dietpi /opt/bazarr` but nothing changed, it is still `root:root`

````
root@DietPi:~# cd /opt/
root@DietPi:/opt# ls -l
total 28
drwxr-xr-x 12 root   root    4096 Aug 13 06:06 bazarr
drwxr-xr-x  7 root   root    4096 Dec  8  2019 emby-server
drwxr-xr-x  3 sonarr dietpi 12288 Mar 27 23:51 NzbDrone
drwxr-xr-x  3 radarr dietpi  4096 Mar  9 22:04 Radarr
drwxr-xr-x  4 root   root    4096 Jul 14  2019 vc
root@DietPi:/opt# python3 bazarr/bazarr.py
Bazarr starting...
2020-08-13 06:28:04,318 - root                             (76fb0ad0) :  INFO (init:57) - BAZARR installing requirements...
2020-08-13 06:28:10,226 - root                             (76fb0ad0) :  INFO (init:61) - BAZARR requirements installed.
2020-08-13 06:28:10,228 - root                             (76fb0ad0) :  INFO (init:67) - Bazarr is being restarted...
Bazarr is restarting...
2020-08-13 06:28:13,928 - root                             (76fa0ad0) :  INFO (init:57) - BAZARR installing requirements...
2020-08-13 06:28:19,838 - root                             (76fa0ad0) :  INFO (init:61) - BAZARR requirements installed.
2020-08-13 06:28:19,839 - root                             (76fa0ad0) :  INFO (init:67) - Bazarr is being restarted...
Bazarr is restarting...
2020-08-13 06:28:23,555 - root                             (76fb1ad0) :  INFO (init:57) - BAZARR installing requirements...
2020-08-13 06:28:29,467 - root                             (76fb1ad0) :  INFO (init:61) - BAZARR requirements installed.
2020-08-13 06:28:29,468 - root                             (76fb1ad0) :  INFO (init:67) - Bazarr is being restarted...
Bazarr is restarting...
2020-08-13 06:28:33,168 - root                             (76efead0) :  INFO (init:57) - BAZARR installing requirements...
^CBazarr exited.
root@DietPi:/opt# sudo chown -R $bazarr:$dietpi /opt/bazarr
root@DietPi:/opt# ls -l
total 28
drwxr-xr-x 12 root   root    4096 Aug 13 06:06 bazarr
drwxr-xr-x  7 root   root    4096 Dec  8  2019 emby-server
drwxr-xr-x  3 sonarr dietpi 12288 Mar 27 23:51 NzbDrone
drwxr-xr-x  3 radarr dietpi  4096 Mar  9 22:04 Radarr
drwxr-xr-x  4 root   root    4096 Jul 14  2019 vc
````

Joulinar, 2020-08-13:
Hi,

many thanks for your request. Command to change directory ownership is as follow

`chown -R bazarr:dietpi /opt/bazarr`

Dr0bac, 2020-08-13:
Thank you. I can now change user/group names. So dollar signs were the main problem here?

closed 2020-08-13 by Dr0bac

Joulinar, 2020-08-13:
Yes

MichaIng, 2020-08-14:
Dollar signs are variable calls in shell. Means `$bazarr` will be replaced with the value of the variable named "bazarr" and since it most likely does not exist, it will be an empty string. Try out this:
```
# Print the empty variable named "bazarr"
echo $bazarr
# Assign a value to this variable
bazarr=someContent
# Print the variable which now has a value
echo $bazarr
```
In your case you wanted to change owner of the files to the literal user names "bazarr" and "dietpi", not to the names assigned as values to the variables "$bazarr" and "$dietpi". I remember my first days of messing with the shell 🤣.