MichaIng/DietPi #3716: AUTO_SETUP_ACCEPT_LICENSE conflicts with AUTO_SETUP_GLOBAL_PASSWORD

mrbluecoat, 2020-08-15:
### Creating a bug report/issue

#### Required Information
- DietPi version

```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=31
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```

- Distro version | 10.5
- Kernel version | Linux DietPi 5.7.8-sunxi64 #trunk SMP Fri Jul 24 17:12:25 MST 2020 aarch64 GNU/Linux
- SBC model | Orange Pi One Plus
- Power supply used | 5V 3A
- SDcard used | Toshiba Exceria Pro

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi?  Yes

#### Steps to reproduce

```
# works:
sudo su
sed -i "s/AUTO_SETUP_AUTOMATED=.*/AUTO_SETUP_AUTOMATED=1/" boot/dietpi.txt
sed -i "s/AUTO_SETUP_ACCEPT_LICENSE=.*/AUTO_SETUP_ACCEPT_LICENSE=1/" boot/dietpi.txt

# doesn't work:
sudo su
sed -i "s/AUTO_SETUP_AUTOMATED=.*/AUTO_SETUP_AUTOMATED=1/" boot/dietpi.txt
sed -i "s/AUTO_SETUP_GLOBAL_PASSWORD=.*/AUTO_SETUP_GLOBAL_PASSWORD=Mypswd2/" boot/dietpi.txt
sed -i "s/AUTO_SETUP_ACCEPT_LICENSE=.*/AUTO_SETUP_ACCEPT_LICENSE=1/" boot/dietpi.txt

# works:
sudo su
sed -i "s/AUTO_SETUP_AUTOMATED=.*/AUTO_SETUP_AUTOMATED=1/" boot/dietpi.txt
sed -i "s/AUTO_SETUP_GLOBAL_PASSWORD=.*/AUTO_SETUP_GLOBAL_PASSWORD=Mypswd2/" boot/dietpi.txt
echo '#!/bin/bash' > boot/Automation_Custom_Script.sh
echo "test -f /var/lib/dietpi/license.txt && mv /var/lib/dietpi/license.txt /var/lib/dietpi/license.accepted" >> boot/Automation_Custom_Script.sh
```

#### Expected behaviour
DietPi autoconfigures and presents console upon SSH login

#### Actual behaviour

Waited 5 minutes on vanilla setup then SSH into device and got:

```
[INFO] DietPi first run setup: Currently running on another screen (PID=1206).
Please resume setup on the active screen.

# htop
/bin/login -f
	-bash
		/bin/bash /boot/dietpi/dietpi-login
			whiptail --title DietPi-Login --backtitle Generic Device ...
```

MichaIng, 2020-08-16:
Many thanks for your report.

Mysterious, both settings have very isolated specific effects on totally different stages of the first boot. I think license and password don't play a role here but instead there happened an error during either firstun update or installs on the local tty1. In this case it switches to interactive to avoid an error loop. If you check back the whole `whiptail` line, there should be something like `[FAILED] Unknown install state/First run setup failed`.

But of course for automated install on headless or without monitor attached this switch to interactive prompt is on the wrong tty. We need to set an error flag then to pickup by subsequent logins and allow repeating the update and/or install interactively on the new login.
_________
Actually it is interesting that the autologin on first boot worked. This never did the first time I tried. I see now also that this can fail, if AUTO_SETUP_AUTOSTART_LOGIN_USER is non-root. However firstrun setup on autologin would just exit without blocking anything, so it would be regularly started on subsequent login.
_________
So two things we need to do better, but I'll simply try to replicate your steps to reproduce as well, probably I've overseen something.

MichaIng, 2020-08-19:
First step: https://github.com/MichaIng/DietPi/commit/d750f2cf711e7a3ca0c27b384fa83aeeb11d070a
- Autologin on first boot with AUTO_SETUP_AUTOMATED will now always be as root user.

closed 2020-08-31 by MichaIng