MichaIng/DietPi #3717: Upgrade failed

Zebulon67fr, 2020-08-16:
#### Details:
- Date           | Sat 15 Aug 19:49:26 CEST 2020
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 2 Model B (armv7l) (index=2)
- Kernel version | #1253 SMP Thu Aug 15 11:49:46 BST 2019
- Distro         | buster (index=5)
- Command        | /DietPi/dietpi/func/dietpi-ramdisk 1
- Exit code      | 127
- Software title | DietPi-Update

#### Steps to reproduce

1. ...Just launched the automatic upgrade
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...terminated without finishing due to a ramdisk issue?

#### Extra details

- ...

Joulinar, 2020-08-16:
Hi,

Many thanks for your report. Pls can you provide some more information issue? Maybe an exact error message. Or the last lines from update output

MichaIng, 2020-08-16:
A pre-RAMdisk system and it seems the parent update process has not been killed as expected. The old updater then calls `dietpi-ramdisk 1` to write new RAMdisk content to disk. But that is of course neither required, nor does the script even exist anymore, hence the error (127 "command not found").

However it does not cause any issue, the update has fully finished: https://github.com/MichaIng/DietPi/blob/7cb3d00f8fe85d8ff1dbe824f392021ca4ff0740/dietpi/dietpi-update#L488
@Zebulon67fr 
Please simply reboot and check banner that should show that everything has been fully updated. I think I saw your case one or two times already, not sure how the kill of the parent process can fail, but it causes a wrong visual output only. If you still have the terminal scrollback buffer/history, you should see `Everything done! Terminating the obsolete DietPi-Update parent instance...`, that is where nothing else should have proceeded because everything has successfully finished.

MichaIng, 2020-09-09:
I mark this as closed due to missing reply. Feel free to reopen if required.

closed 2020-09-09 by MichaIng