MichaIng/DietPi #3718: Uninstall failed

Aakhroot, 2020-08-16:
#### Details:
- Date           | Sun Aug 16 20:00:50 BST 2020
- DietPi version | v6.31.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq --purge autoremove
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

Joulinar, 2020-08-16:
Hi,

Many thanks for your message. Seems something went wrong during `apt uninstall`. Pls can you executed `apt --fix-broken install`

MichaIng, 2020-09-09:
I mark this as closed due to missing reply. Feel free to reopen if required.

closed 2020-09-09 by MichaIng

Aakhroot, 2020-09-09:
Thanks. I had to reimage so could not reproduce.

MichaIng, 2020-09-09:
Okay, no problem 🙂.