MichaIng/DietPi #3724: [Software] bitwarden_rc server

CactiChameleon9, 2020-08-19:
### Creating a software request

#### Formal software information
- bitwarden_rc
- Unofficial Bitwarden password manager compatible server written in Rust
- https://github.com/dani-garcia/bitwarden_rs (server)
- https://bitwarden.com/ (client)
- https://github.com/dani-garcia/bitwarden_rs/wiki/Building-binary (the non-docker option)

#### Are there similar/alternative software titles available with DietPi-Software?

- no alternatives that I can see

#### What makes your requested software better than the above solutions, if available?

- Passwords need to be somewhere you trust, so a self hosted convenient solution is important
- Has a web UI, but maybe you could provide an option to go without to reduce resource usage.

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- They could provide the bineries or
- Install build depenencies and install in the correct place
- Add to boot/autostart

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
1. Install dependencies listed, including rust nightly
2. download and build
3. install in /opt or somewhere permanent
4. add to systemd to allow autostart

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- I could try, but I am not great at best practices while installing things.

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2020-08-19:
Many thanks for your request. Looks quite doable. What do you think, SQLite or MariaDB database backend? For small single user or family instances MariaDB is probably overkill (database size very small anyway) on the other hand MariaDB is already used by many other software title and when the MariaDB server is running anyway, it would be a shame to not use it 😄.

CactiChameleon9, 2020-08-19:
Gosh, that's a hard choice. I barely know the pros and cons of either, so you'll be more able to choose. If most software on dietpi uses MariaDB anyway, then sure why not. Since its just a compiler flag you may even be able to provide an option, however it depends on if you want a one click install script or not.

MichaIng, 2020-08-19:
Yes an option would be best, however a larger plan to add install options (as well post-install) for many software titles, probably we'll go for a non-choice for now. Do you have an instance running to check the database size? Performance should be better with MariaDB (due to good caching implementation and table-based file writes instead of all data in a single "large" file), but it is itself an own running server that takes disk space and certain stock RAM space for the server process. For a 1 MiB database alone its definitely an overkill 😄.

CactiChameleon9, 2020-08-19:
No.. I do not have an instance running I'm afraid... but yes standalone it would be overkill. Maybe just go with sqlite to be on the safe side. Is it really be that bad to run mariadb for just it though? I have not actually needed to use mariadb on my server yet its all been sqlite, so that seems to be the one used by most from my findings - I may be wrong though.

Joulinar, 2020-08-19:
just an unqualified comment from my side regarding size. I'm using SafeInCloud PW Manager where "database" file is stored on my Synology. I have stored around 150 passwords including one icons. File size is around 2 MB. So defnetly MariaDB would be an overkill. Probably trying to detect what DB is running. If nothing, go for SQLite. Otherwise go for already installed? Yeah probably to much work 😄

CactiChameleon9, 2020-08-19:
Thanks @Joulinar, that's really helpful. I wonder how hard it is to detect a mariadb. You could probably check the status of the systemd service and then check that that doesn't return service not found, pipe to a diff command maybe?

MichaIng, 2020-08-19:
I think we'll go for SQLite for now. In theory detection or option is easy but finally one has to think though a bunch of details to not cause unexpected behaviour of the script. I think we need to find a generic approach for a database preference/selection first, similar to webserver preference, but with the difference that multiple database systems can run nicely beside each other and that not all software can use all database systems.

CactiChameleon9, 2020-09-19:
Any advancement with this? Is there any way I could contribute at all?

MichaIng, 2020-09-19:
As always: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title 😉
My priorities are:
1. Fixing bugs and important updates/maintenance of the software titles we already support (currently in process).
2. Following the votes, especially for software titles which are already on the top for a long time: https://feathub.com/MichaIng/DietPi (I'll implement Bazarr definitely with v6.33)

Of course there are other considerations playing a role as well, e.g. that we have no single standalone password manager in portfolio as well, or Bazarr because it fits and complements the Sonarr/Radarr/Lidarr set perfectly find and setup can mostly be adopted.

So especially for new software requests there need to be someone else starting with the implementation to have it realistically implemented short- or mid-term, I'm afraid.

Btw: https://feathub.com/MichaIng/DietPi/+126
Actually, what are the benefits of the Rust version vs the "original"? https://github.com/bitwarden/server
Okay VS, .NET, I can imagine a Windows server in the first place and heavy-weight 😄.
I added the FeatHub link to the first post, even that I am not sure if the Rust version was in mind, but it should not play a large role as long as the clients work with it.

anubis-genix, 2020-09-22:
Besides taking less resources than the official version this one seems to have all premium features unlocked

CactiChameleon9, 2020-10-09:
I am actually working on this now, but I am struggling with an error during installation.

[image]

Can I have some help please?

CactiChameleon9, 2020-10-09:
My code for installing is here, I copied a bit from spotify to check if it was the issue
`
                        Banner_Installing

                        # Dependencies: https://github.com/dani-garcia/bitwarden_rs/wiki/Building-binary#dependencies
                        DEPS_LIST='pkg-config libssl-dev'
                        # Rust from https://rustup.rs/
                        curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh

                        # Download & Install
                        INSTALL_URL_ADDRESS='https://github.com/dani-garcia/bitwarden_rs/releases' # Full path fails wget spider test...
                        G_CHECK_URL "$INSTALL_URL_ADDRESS"

                        LATEST_URL=$(curl -s https://api.github.com/repos/dani-garcia/bitwarden_rs/releases/latest | grep 'zipball_url' | cut -d\" -f4) # Get download url using api
                        no_check_url=1 Download_Install "$LATEST_URL"
                        [[ -d '/opt/bitwarden_rs' ]] && G_EXEC rm -R /opt/bitwarden_rs
                        G_EXEC mv dani-garcia-bitwarden_rs-* /opt/bitwarden_rs

                        # Build
                        G_EXEC cd /opt/bitwarden_rs
                        G_EXEC_OUTPUT=1 G_EXEC cargo build --features sqlite --release
                        G_EXEC cd /tmp/$G_PROGRAM_NAME

                        # Uninstall rust after compiling
                        G_EXEC rustup self uninstall
`

MichaIng, 2020-10-09:
When you use `Download_Install` function for a non deb/zip/7z/tar file, it will be simply downloaded but requires a target path/name then, like:
```
Download_Install 'https://example.org/example.jar' /path/to/executable.jar
```
This is optional for DEB packages since those are simply installed and obsolete for known archives as those are extracted to current directory if no target is present.

CactiChameleon9, 2020-10-09:
It is actually a zip file I am downloading

MichaIng, 2020-10-09:
Ah I see, but the URL does not represent that: `https://api.github.com/repos/dani-garcia/bitwarden_rs/tarball/1.16.3`
Let me see, there is another case where we scrape the version via `tag_name` and then download this particular version archive.

CactiChameleon9, 2020-10-09:
The url which the api returns is here: https://api.github.com/repos/dani-garcia/bitwarden_rs/zipball/1.16.3

CactiChameleon9, 2020-10-09:
Just a thought, shouldn't you check the file type based on the `file` command instead of on name extension?

MichaIng, 2020-10-09:
Ah like here:
```
INSTALL_URL_ADDRESS='https://api.github.com/repos/dani-garcia/bitwarden_rs/releases/latest'
G_CHECK_URL "$INSTALL_URL_ADDRESS"
local version=$(curl -sf "$INSTALL_URL_ADDRESS" | grep -m1 '^[[:blank:]]*"tag_name":' | cut -d \" -f 4)
local fallback_url='https://github.com/dani-garcia/bitwarden_rs/archive/1.16.3.tar.gz'
Download_Install "https://github.com/dani-garcia/bitwarden_rs/archive/$version.tar.gz"
```

CactiChameleon9, 2020-10-09:
OK, thanks a lot. Yeah getting just the version would solve that

MichaIng, 2020-10-09:
Ah forget about `fallback_url`, as this is impossible to be not available as every release has a `tag_name`. We use this when scraping the final download URL directly via `browser_download_url` entry, where available, and in cases there are multiple releases for multiple platforms we need to scrape for identifying patterns that define its for Linux and/or the devices architecture and these naming conventions can change.

CactiChameleon9, 2020-10-09:
A small issue, the rust installer uses path variable changing as it installs locally in the home folder (I remove at the end don't worry). It seems that it is not using that change, hence command not found error for compiling

CactiChameleon9, 2020-10-09:
OK, fixed. I have to run 'source $HOME/.cargo/env' instead of "logging out" and back in as I cannot do so in script

CactiChameleon9, 2020-10-09:
Is it OK to compile as part of the installation process and should I provide a warning saying this may take a while?

MichaIng, 2020-10-09:
You can export the variables to have the Rust installed into the correct directory. I just ran into the same issue when trying to install the new Rust-based Firefox Sync Server. Solved it by running rustup as the final run user: https://github.com/MichaIng/DietPi/issues/3774#issuecomment-703254902
That is probably the cleanest method, as it does not depend on multiple variables and all files/dir have correct permissions directly:
- Make the install dir to run users home dir.
- Start rustup as run user, so the whole Rust environment is installed into the install dir.

CactiChameleon9, 2020-11-06:
OK... so basically I lost my code in the update :+1: I am redoing now and I need to now test

CactiChameleon9, 2020-11-06:
In theory, it now works!!
I have done some tweaking/additions to the steps as I fixed the original first attempt I did awhile ago so It should all work

I still need to write an uninstall script, but if someone would like to test/check what I have done is OK the code is here:
```
                #------------------
                software_id=181

                aSOFTWARE_NAME[$software_id]='Bitwarden RS'
                aSOFTWARE_DESC[$software_id]='Unoffical Bitwarden password manager server written in rust'
                aSOFTWARE_TYPE[$software_id]=0
                aSOFTWARE_CATEGORY_INDEX[$software_id]=4
                aSOFTWARE_ONLINEDOC_URL[$software_id]='p=24799#p24799'
                aSOFTWARE_REQUIRES_BUILDESSENTIAL[$software_id]=1
                aSOFTWARE_REQUIRES_SQLITE[$software_id]=1

                
                
                
                

                software_id=181 # Bitwarden RS
                if (( ${aSOFTWARE_INSTALL_STATE[$software_id]} == 1 )); then
                    Banner_Installing

                    # Dependencies: https://github.com/dani-garcia/bitwarden_rs/wiki/Building-binary#dependencies
                    DEPS_LIST='pkg-config libssl-dev'
                    # Rust from https://rustup.rs/
                    G_EXEC curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- -y
                    source $HOME/.cargo/env

                    # Download & Install
                    INSTALL_URL_ADDRESS='https://api.github.com/repos/dani-garcia/bitwarden_rs/releases/latest'
                    G_CHECK_URL "$INSTALL_URL_ADDRESS"
                    local version=$(curl -sf "$INSTALL_URL_ADDRESS" | grep -m1 '^[[:blank:]]*"tag_name":' | cut -d \" -f 4)
                    Download_Install "https://github.com/dani-garcia/bitwarden_rs/archive/$version.tar.gz"

                    [[ -d '/opt/bitwarden_rs' ]] && G_EXEC rm -R /opt/bitwarden_rs
                    G_EXEC mv bitwarden_rs-* /opt/bitwarden_rs

                    # Build
                    G_EXEC cd /opt/bitwarden_rs
                    G_EXEC_OUTPUT=1 G_EXEC cargo build --features sqlite --release
                    G_EXEC cd /tmp/$G_PROGRAM_NAME

                    # Uninstall rust after compiling
                    G_EXEC rustup self uninstall -y
                fi
                    
                    
                    
                    
                    
                    
                    
                software_id=181 # Bitwarden RS
                if (( ${aSOFTWARE_INSTALL_STATE[$software_id]} == 1 )); then

                    Banner_Configuration

                    Create_User -d /mnt/dietpi_userdata/bitwarden_rs bitwarden_rs
                    
                    # make dir and data dir needed to run
                    G_EXEC mkdir -p /mnt/dietpi_userdata/bitwarden_rs/data
                                        
                    # make and edit the template config to have no web vault
                    G_EXEC cp /opt/bitwarden_rs/.env.template /mnt/dietpi_userdata/bitwarden_rs/bitwarden_rs.env
                    G_EXEC sed -i 's/# WEB_VAULT_ENABLED=true/WEB_VAULT_ENABLED=false/g' /mnt/dietpi_userdata/bitwarden_rs/bitwarden_rs.env

                    # permissions
                    G_EXEC chown -R bitwarden_rs:bitwarden_rs /mnt/dietpi_userdata/bitwarden_rs
                    G_EXEC chmod +x /opt/bitwarden_rs/target/release/bitwarden_rs
                    
                    # systemd service
                    cat << _EOF_ > /etc/systemd/system/bitwarden_rs.service
[Unit]
Description=Bitwarden Server (Rust Edition)
Documentation=https://github.com/dani-garcia/bitwarden_rs

After=dietpi-boot.service network.target

[Service]
# The user/group bitwarden_rs is run under. the working directory (see below) should allow write and read access to this user/group
User=bitwarden_rs
Group=bitwarden_rs
# The location of the .env file for configuration
EnvironmentFile=/mnt/dietpi_userdata/bitwarden_rs/bitwarden_rs.env
# The location of the compiled binary
ExecStart=/opt/bitwarden_rs/target/release/bitwarden_rs
# Set reasonable connection and process limits
LimitNOFILE=1048576
LimitNPROC=64
# Isolate bitwarden_rs from the rest of the system
PrivateTmp=true
PrivateDevices=true
ProtectHome=true
ProtectSystem=strict
# Only allow writes to the following directory and set it to the working directory (user and password data are stored here)
WorkingDirectory=/mnt/dietpi_userdata/bitwarden_rs
ReadWriteDirectories=/mnt/dietpi_userdata/bitwarden_rs
# Allow bitwarden_rs to bind ports in the range of 0-1024
AmbientCapabilities=CAP_NET_BIND_SERVICE

[Install]
WantedBy=multi-user.target
_EOF_

                fi
```

CactiChameleon9, 2020-11-07:
OK, that is the uninstall script done.
```
                software_id=177 # Firefox Sync Server
                if (( ${aSOFTWARE_INSTALL_STATE[$software_id]} == -1 )); then

                        Banner_Uninstalling
                        if [[ -f '/etc/systemd/system/bitwarden_rs.service' ]]; then

                                systemctl disable --now bitwarden_rs
                                rm -R /etc/systemd/system/bitwarden_rs.service*

                        fi
                        [[ -d '/etc/systemd/system/bitwarden_rs.service.d' ]] && rm -R /etc/systemd/system/bitwarden_rs.service.d
                        
                        getent passwd bitwarden_rs > /dev/null && userdel ffsync
                        getent group bitwarden_rs > /dev/null && groupdel ffsync
                        
                        [[ -d '/opt/bitwarden_rs' ]] && rm -R /opt/bitwarden_rs
                        
                        [[ -d '/mnt/dietpi_userdata/bitwarden_rs' ]] && rm -R /mnt/dietpi_userdata/bitwarden_rs

                fi
```
I will now attempt an uninstall and reinstall. If it all seems to work I will make a pr

CactiChameleon9, 2020-11-07:
Also, I am not very sure on what to do with the online doc url, as I have not made one?

MichaIng, 2020-11-07:
If you can provide some short description, initial setup instructions, if useful/required, a screenshot if applicable, that would be awesome. I'll create a docs page and add the online docs URL then. Just to give you an idea how this may look like: https://dietpi.com/phpbb/viewtopic.php?p=27577#p27577
I kept them mostly short, especially when I don't use the software myself, a few more details that are interesting/useful for users are always welcome of course 😄.

CactiChameleon9, 2020-11-07:
> If you can provide some short description, initial setup instructions, if useful/required, a screenshot if applicable, that would be awesome. I'll create a docs page and add the online docs URL then. Just to give you an idea how this may look like: https://dietpi.com/phpbb/viewtopic.php?p=27577#p27577
> I kept them mostly short, especially when I don't use the software myself, a few more details that are interesting/useful for users are always welcome of course smile.

Thanks, I will do this in the morning

CactiChameleon9, 2020-11-08:
**Description**
Bitwarden RS is a server for the open source password manager Bitwarden - a place where you can store and access all of your passwords safely and securely on your own hardware.
**Usage**
To use, just install the Bitwarden browser extension or/and install the app on your Phone.
Then press the cog icon change the server address to [your dietpi device ip address]:8000
Return to the login screen, make an account, and login. Then use as normal
**Installing the Web Vault**
Go to https://github.com/dani-garcia/bw_web_builds/releases/latest
Download the tar.gz to dietpi server 
Extract it to /mnt/dietpi_userdata/bitwarden_rs/ (should be a folder called web-vault)
Then enable the webvalt in the config file (/mnt/dietpi_userdata/bitwarden_rs/bitwarden_rs.env)
And finally reboot - the webvalt will be at [your dietpi device ip address]:8000 in your webrowser
Optionally setup a reverse proxy to access though a subdomain if you choose

MichaIng, 2020-11-28:
Meta info: https://github.com/MichaIng/DietPi/pull/3934

id4vip, 2020-12-03:
Thanks @Joulinar,
with your instruction i am able to install bitwarden & it is working perfectly on my raspberrypi b2 on dietpi operating system. I am able to access the webUI of bitwarden through browser. so everything is working as per intended from this software.

but one side note that i am not able to access this host bitwarden through android bitwarden app. it shows me some certificate error. so some expert can check it & find the solution for it.

one thing more if i want to backup/sync my vault data. which folder in dietpi i have to store some where. also want to clarify . if some thing happen to bitwarden this host. if i have those vault backup files. can i restore from those backup stored files. those files are independent to version of bit-warden version or operating system.

CactiChameleon9, 2020-12-03:
> but one side note that i am not able to access this host bitwarden through android bitwarden app. it shows me some certificate error. so some expert can check it & find the solution for it.

That's probably because we are using self-signed a https cert so the web-vault is accessible.
@id4vip could you try doing http://[your dietpi device ip address]:8001 to see if that works?

Joulinar, 2020-12-03:
It's not related to `bitwarden` but I know some tools where you need to set explicitly to allow not trusted certificates. Not sure if there is such a possibility inside the app.

id4vip, 2020-12-03:
No, Android app is not pick https or http address. this problem specially for android users. when i try to find solution i found some forums as below. but i am not much that expert so no path to achieve this.  https://github.com/bitwarden/mobile/issues/474

CactiChameleon9, 2020-12-03:
@id4vip OK - apparently you can install the certificate directly into android and then it will work - however thats a bit of a process. I can try an come up with a method if you want to however

MichaIng, 2020-12-03:
Yes exactly, the official clients require to have the certificate added to the OS trusted CA store: https://dietpi.com/test/software/cloud/#bitwarden_rs

For Windows 10 I made screenshot-supported instructions, for other OSes this is outstanding. I could do that for Android 8 Samsung UI (old phone stuck on old version, custom OS builds lack features) but probably it would be beneficial to have that for Android 10 native UI.

CactiChameleon9, 2020-12-04:
> Without HTTPS, the web UI denies to work, probably some clients as well, so that is no option.

Actually, I think most of the clients are happier with no certificate - I used that with android for a while.

BTW - nice revamped docs.

MichaIng, 2020-12-04:
> Actually, I think most of the clients are happier with no certificate

The official docs at least say that HTTPS is required, so I would not count on that being true for long, at least I hope so. Sending a master password, even only connecting to an interface that hold potentially EVERYTHING of your most sensitive access credentials and data should never be done unencrypted, even within the home network. So if that little extra effort to get it running makes people aware of HTTPS, that it's not only about encryption, but as well about authentication, which both are essential parts of internet security, then it is totally worth it 🙂. We'll do our best to make it as easy as possible for end users.

closed 2020-12-20 by MichaIng

anubis-genix, 2020-12-21:
I'm trying to get bitwarden_rs to use the certs I've already issued for my nextcloud instance, without configuring an additional reverse proxy.
So I took the bitwarden_rs.env and changed the ROCKET_TLS line to ROCKET_TLS={"/etc/letsencrypt/live/domain.tld/fullchain.pem","key="/etc/letsencrypt/live/domain.tld/privkey.pem"}
Unfortunately this broke bitwarden and I got the message "Failed to start Bitwarden Server (Rust Edition)", so I reverted back to selfsigned certs.
Is there a logical error on my side?

MichaIng, 2020-12-21:
Most likely this is due to missing permissions to read the private key. Best practice would be to copy/overwrite the self-signed key+cert with the ones from Certbot and `chown bitwarden_rs:bitwarden_rs /mnt/dietpi_userdata/bitwarden_rs/{privkey,cert}.pem`. You can use a script in /etc/letsencrypt/ post-deploy hook directory to have this done automatically when the original certificate is renewed.

anubis-genix, 2020-12-21:
Thanks @MichaIng, that did the trick!
Will mess around with the post-deploy hook script you mentioned. Maybe this could also be automated in a later release?

Joulinar, 2020-12-23:
Hi guys, 

Did someone of you install iOS client? We have a user on our forum complaint about not trusted certifications, while is logic as it's self signed. 

https://dietpi.com/phpbb/viewtopic.php?f=9&t=8430

MichaIng, 2020-12-23:
I can figure is out on macOS, but have not iOS device here. Yeah, OS'es really don't make it easy. Would be great if we could add instructions for all major OS'es here: https://dietpi.com/docs/software/cloud/#bitwarden_rs to the `How do I add a self-signed certificate to the OS’ Trusted Root Certification Authorities store?` info box.

Joulinar, 2020-12-23:
well I have an old iPhone SE flying round. Will reactive it as I don't use any Apple devices normally.

CactiChameleon9, 2020-12-23:
It seems, that people say that you just email the certificate to yourself and open it, and IOS 13+ has a raised standard in certificates meaning it that RSA keys must be at least 2048 bits [(source)](https://stackoverflow.com/questions/58314788/self-signed-ca-not-trusted-in-ios-13-anymore)

MichaIng, 2020-12-23:
We create a 4096 bit key, so that shouldn't be an issue then 😎.

Joulinar, 2020-12-23:
currently installing bitwarden on my RPi4 64bit. But it's taking time :/

Joulinar, 2020-12-23:
ok I'm giving up on these iPhone stuff. I tried to upload the certificate but it creates a profile. I don't think this is correct. As well Bitwarden is still complaining about not trusted certificates

CactiChameleon9, 2020-12-24:
https://old.reddit.com/r/Bitwarden/comments/ep9qyz/self_signed_certs_iosmacos_issue_solved/ This person here seemed to have managed, could you try those steps @Joulinar ?

Joulinar, 2020-12-24:
Hi, 

for W10 I did a small guide on our forum. It's pretty strait forward https://dietpi.com/phpbb/viewtopic.php?p=29810#p29810

On iOS, I have the issue at point 4 `Goto Settings -> General -> About -> Certificate Trust Settings` because It's empty. there is nothing to enable, even if I managed to install profile on point 3. Not sure what I'm doing wrong. Maybe it's the certificate as it has a lifetime of 20 years. As well I'm not sure what files are needed on iOS. On Windows a `.crt` file is enough.

BTW: should we create the `.crt` file by default? It's needed on W10 at least. Or make a statement on our online docs
`openssl x509 -outform der -in your-cert.pem -out your-cert.crt`

CactiChameleon9, 2020-12-24:
> for W10 I did a small guide on our forum. It's pretty strait forward https://dietpi.com/phpbb/viewtopic.php?p=29810#p29810

Great! Hopefully this can be added to the docs?

> On iOS, I have the issue at point 4 `Goto Settings -> General -> About -> Certificate Trust Settings` because It's empty. there is nothing to enable, even if I managed to install profile on point 3. [...]

Then I am clueless.... I really don't know much about iOS... Maybe it is the 20 years being too much, I am not sure.

> BTW: should we create the `.crt` file by default? It's needed on W10 at least. Or make a statement on our online docs
> `openssl x509 -outform der -in your-cert.pem -out your-cert.crt`

I say we add it to our online docs for the time being and then make the change to the bitwarden configuration code later - then again its not up to me and I don't know much about certificates either.

MichaIng, 2020-12-24:
Many thanks, however for Windows 10 we have a guide already, which is starting with downloading the certificate from within the browser when accessing the web vault. But from opening the certificate file on it matches yours: https://dietpi.com/docs/software/cloud/#bitwarden_rs

Downloading the file via browser should be the easiest method usually. The conversion in Windows 10 is done when storing it to file.

> Maybe it is the 20 years being too much, I am not sure.

Please not. If we make it too short, users might start complaining when it expires suddenly, so I thought that a moreless unlimited lifetime is easiest here. But at least to worth to give it a try, if so far import on iOS did not succeed for anyone.

_It is a pain that OS'es make it so difficult to manually trust a certificate, starting with the support for all widely used formats. There is no alternative if you do not have a public domain or need it to access local network hosts via local hostnames/IPs. The alternative for such cases would be plain unauthenticated + unencrypted HTTP, which of course is much worse than trusting even an attackers fake certificate._

Joulinar, 2020-12-24:
even with a smaller lifetime, I don't see the certificate inside `Certificate Trust Settings`.

Joulinar, 2020-12-24:
Success on iOS14 😄 

Adding following into `/etc/ssl/openssl.cnf` fixed it

```
[SAN]
basicConstraints                = CA:TRUE,pathlen:0
```

I checked it with 700 days atm. Will do more test later. Next to this I need to figure out how to add it to the command line to avoid changing `/etc/ssl/openssl.cnf` 🤔 

[image]

MichaIng, 2020-12-24:
Just to assure, this was required on the server, right? Did you find out what is the background of it?

It seems to be a flag to set that this certificate can be used as CA, which is what is done when importing it into trusted store: https://manpages.debian.org/buster/openssl/x509.1ssl.en.html#CERTIFICATE_EXTENSIONS
But I can't find something about the `pathlen` option.

We can add it via command line option, which is currently:
```
-config <(cat /etc/ssl/openssl.cnf <(echo -ne "[SAN]\nsubjectAltName=DNS:$(</etc/hostname),IP:$(mawk 'NR==4' /run/dietpi/.network)"))
```
SAN is required for most modern browsers but not defined by default in openssl.cnf and it requires the hostname and IPs added that shall be used to access. So command line option uses the default openssl.cnf and adds SAN options to the end with best guess we can do: local IP and hostname. This alone can btw cause issues already with some clients browsers when accessing via public hostname/IP, in which case a Let's Encrypt cert (or similar) makes sense at latest. However, the setting can hence be added easily:
```
-config <(cat /etc/ssl/openssl.cnf <(echo -ne "[SAN]\nsubjectAltName=DNS:$(</etc/hostname),IP:$(mawk 'NR==4' /run/dietpi/.network)\nbasicConstraints=CA:TRUE,pathlen:0"))
```

Joulinar, 2020-12-24:
After Santa Claus is gone again and finished eating, I did another test. My iPhone SE also accepts the certificate with a term of 20 years. so we just have to add `nbasicConstraints=CA:TRUE,pathlen:0`

MichaIng, 2020-12-25:
Fixed: https://github.com/MichaIng/DietPi/commit/8a5d0931fcfb764511f941f3e8782573240ed991 (whoops, this is for v6.35 of course 😄)
Changelog: https://github.com/MichaIng/DietPi/commit/4973b5202eeb86271126ba1f3e0d56549e023c9b

anubis-genix, 2020-12-27:
Can we change the /admin page to /bwadmin so it doesn't collide with other services like pihole?

[10 more replies]