MichaIng/DietPi #3726: Dietpi-software version of motion eye on doesn't support RTSP (workaround)

ickam, 2020-08-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=31
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | 10.4
- Kernel version | Linux DietPi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
- SBC model | RPi3
- Power supply used | Original RPi3 PSU
- SDcard used |  SanDisk High Endurance

#### Additional Information (if applicable)
- Software title | (motion / motioneye)
- Was the software title installed freshly or updated/migrated? Freshly installed
- Can this issue be replicated on a fresh installation of DietPi? Probably

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ... Install motion eye
2. ...

#### Expected behaviour

- Be able to add RTSP stream of a network camera

#### Actual behaviour

- Error message in WebUI: "Protocol "rtsp" not supported or disabled in libcurl"

#### Extra details

- 
```bash
apt remove motion
apt install  pkg-config libtool libavcodec-dev libavdevice-dev libavformat-dev libswscale-dev libpq-dev libsqlite3-dev debhelper dh-autoreconf libwebp-dev libmicrohttpd-dev gettext libmariadbclient-dev
wget https://raw.githubusercontent.com/Motion-Project/motion-packaging/master/builddeb.sh
chmod +x builddeb.sh
./builddeb.sh 
```
wait until it builds and then 
```bash 
dpkg -i pi_buster_motion_4.3.1+git20200412-392779e-1_armhf.deb 
```
Or whatever your built version is

MichaIng, 2020-08-20:
Interesting, we use the official Debian/Raspbian package and RTSP protocol has long been available in motion on the provided version strings. Sadly for the Buster build, there is no build log available anymore, but here is the build of the version provided in Buster from Sid repo: https://buildd.debian.org/status/fetch.php?pkg=motion&arch=armhf&ver=4.1.1-1.1%2Bb2&stamp=1548693520&raw=0
And I see `netcam_rtsp` being compiled there, although a few additional entries compared to the 4.2 build.

However, as the package build is made so easy, probably we can provide those for Buster until the Debian build has been fixed in this regards, likely with Debian Bullseye only.

MichaIng, 2020-09-30:
@ickam 
Do you find time to test the above Sid packages? For a Debian bug report it would be good to know which packages/versions are.

MichaIng, 2021-09-01:
Bullseye contains a new package:
```sh
cd /tmp
wget https://deb.debian.org/debian/pool/main/m/motion/motion_4.3.2-1_armhf.deb
apt install ./motion_4.3.2-1_armhf.deb
rm motion_4.3.2-1_armhf.deb
```

MichaIng, 2021-09-18:
I'll mark this as closed for now in the hope that it works with current Bullseye motion packages. Another major issue with MotionEye is the outstanding Python 3 support, which is why it is disabled on actual Bullseye images. This is the high priority issue, requiring testing and pushing the MotionEye developer to merge all the pull requests that others contributed already: #4671

If the issue can be replicated with the current Bullseye motion package, then it is something for the Debian bug tracker: https://bugs.debian.org/cgi-bin/pkgreport.cgi?src=motion

closed 2021-09-18 by MichaIng