MichaIng/DietPi #3730: Asus Tinkerboard slow network adapter

scorpionknifes, 2020-08-23:
#### Required Information
- Date | Fri May 15 00:00:21 NZST 2020
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator | DietPi Core Team
- Pre-image | Armbian
- Hardware | ASUS Tinkerboard Board (armv71) (ID=52)
- Kernel version | Linux DietPi 5.4.28-rockchip
- Distro | buster (ID=5)
- Command | ping -c 1 -W 60 9.9.9.9
- Exit code | 2
- Software title | DietPi-Login

#### Steps to reproduce

1. use dietpi automation with dietpi.txt
2. copy dietpi.txt to /boot/dietpi.txt
3. boot with dietpi automation

#### Expected behaviour

- ping -c 1 -W 60 9.9.9.9 should wait for network connectivity
- once network connectivity is achieved continue setup

#### Actual behaviour

- ping -c 1 -W 60 9.9.9.9 throws exit code 2 and stops automatic setup
- this is thrown with message "no network connectivity"
- after approx 10 seconds there would be network connectivity but exit code is already thrown and setup is stopped.

#### Extra details

- network connection using ethernet on tinkerboard takes ages.
- tried using different IP e.g 1.1.1.1
- tried using static IP (thought was slow dhcp, but it was slow adapter on tinkerboard)
- tried dietpi-update

scorpionknifes, 2020-08-23:
I guess i just haven't set CONFIG_G_CHECK_URL_ATTEMPTS high enough
```
CONFIG_G_CHECK_URL_ATTEMPTS=1000
```
setting it to 1k seems to do the trick, also don't recommend setting CONFIG_G_CHECK_URL_TIMEOUT to 60 would just keep it to 10.

I'm gonna do some more testing before closing this issue.

MichaIng, 2020-08-23:
Many thanks for your report. Hmm a standard route/gateway check is done before, but in case of DHCP indeed this set before an actual IP has been assigned. But in such case, `connect: Network is unreachable` should not occur AFAIK but instead ping should wait for a response. `connect: Network is unreachable` means that there is no route to this network, but with a 0/0 default route there is one for every IP.

Of course we could add a delay between each ping attempt, but 1-2 seconds would have been not enough as well in your case and longer would not be reasonable for any later check. Probably I need to run some tests to verify the above assumptions, default route + no `connect: Network is unreachable` while waiting for DHCP IP after there was a default route before. Just to be sure, `CONFIG_BOOT_WAIT_FOR_NETWORK=1` or `CONFIG_BOOT_WAIT_FOR_NETWORK=2` was set and in first case the route was found during the 10 seconds period, right?

scorpionknifes, 2020-08-23:
Thanks for the quick response. I am currently using ```CONFIG_BOOT_WAIT_FOR_NETWORK=2```. I remeasured and it takes around 6-8 sec for the first route to be found (for the test to pass it would take longer). Let me know if you need any testing to be done on ASUS Tinker Board.

MichaIng, 2020-09-26:
New image has been uploaded: https://dietpi.com/downloads/images/DietPi_ASUSTB-ARMv7-Buster.7z
New kernel v5.8.10 and a fix which should prevent ping `connect: Network is unreachable` error on first boot, at least it helped in a few other cases. @scorpionknifes would be great if you could give it a try and compare 🙂.

closed 2020-10-17 by MichaIng