MichaIng/DietPi #3732: Airsonic missing from Dietpi-Software selection

linxiaopi, 2020-08-24:
#### Required Information
- DietPi version:
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=31
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version:
buster 10.4

- Kernel version:
Linux DietPi 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l GNU/Linux

- SBC model:
RPi 2 Model B (armv7l)

- Power supply used:
5v 2A

- SDcard used:
SanDisk Ultra 8GB

#### Additional Information (if applicable)
- Software title: Airsonic

#### Steps to reproduce
1. Start dietpi-software
`sudo dietpi-software`

2. Arrow down to 'Software Optimised' and press 'Enter'

3. Airsonic is not an available option despite being listed under "Media Systems" here: https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&sid=ac4c8a57674b5057d445d9657e39c206

ALSO NOT AVAILABLE BUT LISTED UNDER "Media Systems: 
-Subsonic
-Roon Server
-jRiver

#### Expected behaviour
Airsonic (and others) should be available to install

#### Actual behaviour
Airsonic is not an available option

#### Extra details
If Airsonic (and Subsonic, etc.) are no longer supported, consider updating the webpage.

MichaIng, 2020-08-24:
Many thanks for your report.

Indeed we needed to disable if for Buster+ since it didn't support Java 11: #2787

The documentation still tells that it requires Java 8: https://airsonic.github.io/docs/install/prerequisites/

It looks like modern Java version support has been added two days after I disabled it: https://github.com/airsonic/airsonic/releases

This has been certainly forgotten. I'll run a quick test to assure an in case re-enabled it. You can as well quickly test/enabled it by removing [this](https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L522) line from `/boot/dietpi/dietpi-software`.

MichaIng, 2020-08-24:
Okay, Airsonic now works fine (Subsonic does still not support Java 11), so I re-enabled it: https://github.com/MichaIng/DietPi/commit/492d26ec0273134fecdb8de94de7618d6614fa67
Changelog: https://github.com/MichaIng/DietPi/commit/836b1ee6377c841f750b9757ed6783469dab0887

closed 2020-08-31 by MichaIng