MichaIng/DietPi #3736: Generate an ID for the NanoPi K2 SBC

StephanStS, 2020-08-25:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
An DietPi image for the NanoPi K2 shall be generated.

#### Describe the solution you'd like:
An ID for the NanoPi K2 is needed which can be used during the DietPi-PREP script.

#### Additional context
With this new ID I will generate the DietPi image.

MichaIng, 2020-08-25:
Many thanks for bringing this up. NanoPi R2S is missing as well even that I have one here now. Same chip as NEO3, even the image should work, but I'll add a new ID for both of them tomorrow.

One important topic:
```
systemctl status haveged
```
Does this work on K2 or same issues as on NEO3?

StephanStS, 2020-08-25:
@MichaIng: The haveged also fails:
```
root@DietPi:~# systemctl status haveged
● haveged.service - Entropy daemon using the HAVEGE algorithm
   Loaded: loaded (/lib/systemd/system/haveged.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-08-25 21:02:09 CEST; 2h 22min ago
     Docs: man:haveged(8)
           http://www.issihosts.com/haveged/
  Process: 1422 ExecStart=/usr/sbin/haveged --Foreground --verbose=1 $DAEMON_ARGS (code=exited, status=1/FAILURE)
 Main PID: 1422 (code=exited, status=1/FAILURE)

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
root@DietPi:~# systemctl start haveged
root@DietPi:~# systemctl status haveged
● haveged.service - Entropy daemon using the HAVEGE algorithm
   Loaded: loaded (/lib/systemd/system/haveged.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-08-25 23:25:01 CEST; 3s ago
     Docs: man:haveged(8)
           http://www.issihosts.com/haveged/
  Process: 10498 ExecStart=/usr/sbin/haveged --Foreground --verbose=1 $DAEMON_ARGS (code=exited, status=1/FAILURE)
 Main PID: 10498 (code=exited, status=1/FAILURE)

Aug 25 23:25:01 DietPi systemd[1]: Started Entropy daemon using the HAVEGE algorithm.
Aug 25 23:25:01 DietPi haveged[10498]: haveged: Couldn't initialize HAVEGE rng 5
Aug 25 23:25:01 DietPi haveged[10498]: haveged starting up
Aug 25 23:25:01 DietPi systemd[1]: haveged.service: Main process exited, code=exited, status=1/FAILURE
Aug 25 23:25:01 DietPi systemd[1]: haveged.service: Failed with result 'exit-code'.
```

MichaIng, 2020-08-26:
What does this report in your case? I'll also try on by NEO3 later:
```
cat /sys/devices/system/cpu/cpu[0-9]/cache/index[0-9]/size
```

MichaIng, 2020-08-26:
I think I found it: https://github.com/jirka-h/haveged/pull/7
This fix has been added with v1.9.5: https://github.com/jirka-h/haveged/pull/7/commits/f2193587e8e51aa42b192c17ad39467931e2d4c2
Debian Buster ships haveged 1.9.1: https://packages.debian.org/buster/haveged
Bullseye ships v1.9.8, hence should contain the fix: https://packages.debian.org/bullseye/haveged
Since it does not yet have any additional Bullseye-only dependencies, besides the haveged binary package and new libhavege2 library, we could relatively simply install it on Buster systems. This is probably better than hardcoding the 16 KiB cache size, in case any kernel/firmware upgrade by Armbian enables/fixes the cache size sysfs entry.

To test:
```
cd /tmp
wget http://ftp.de.debian.org/debian/pool/main/h/haveged/libhavege2_1.9.8-4_arm64.deb
wget http://ftp.de.debian.org/debian/pool/main/h/haveged/haveged_1.9.8-4_arm64.deb
apt install ./libhavege2_1.9.8-4_arm64.deb ./haveged_1.9.8-4_arm64.deb
rm ./libhavege2_1.9.8-4_arm64.deb ./haveged_1.9.8-4_arm64.deb
apt autopurge
```
I'll just try it on my NEO3.

StephanStS, 2020-08-26:
@MichaIng: On my K2 after doing the lines given above:
```
root@DietPi:/tmp# systemctl status haveged
● haveged.service - Entropy Daemon based on the HAVEGE algorithm
   Loaded: loaded (/lib/systemd/system/haveged.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-08-26 15:29:49 CEST; 21s ago
     Docs: man:haveged(8)
           http://www.issihosts.com/haveged/
 Main PID: 2073 (haveged)
    Tasks: 1 (limit: 1960)
   Memory: 5.2M
   CGroup: /system.slice/haveged.service
           └─2073 /usr/sbin/haveged --Foreground --verbose=1 -w 1024

Aug 26 15:29:49 DietPi systemd[1]: Started Entropy Daemon based on the HAVEGE algorithm.
Aug 26 15:29:49 DietPi haveged[2073]: haveged: listening socket at 3
Aug 26 15:29:49 DietPi systemd[1]: /lib/systemd/system/haveged.service:19: Unknown lvalue 'ProtectHostname' in section 'Service', ignoring
Aug 26 15:29:49 DietPi systemd[1]: /lib/systemd/system/haveged.service:20: Unknown lvalue 'ProtectKernelLogs' in section 'Service', ignoring
Aug 26 15:29:49 DietPi haveged[2073]: haveged: ver: 1.9.8; arch: generic; vend: ; build: (gcc 9.2.1 CTV); collect: 128K
Aug 26 15:29:49 DietPi haveged[2073]: haveged: cpu: (VC); data: 16K (D V); inst: 16K (D V); idx: 11/40; sz: 15456/64452
Aug 26 15:29:49 DietPi haveged[2073]: haveged: tot tests(BA8): A:1/1 B:1/1 continuous tests(B):  last entropy estimate 8.00326
Aug 26 15:29:49 DietPi haveged[2073]: haveged: fills: 0, generated: 0
```

MichaIng, 2020-08-26:
Jep, so `data: 16K (D V); inst: 16K` shows that reverting to 16 KiB default works now. Yeah the Bullseye systemd unit contains two entries which are not yet supported on Buster systemd version, but that is okay, could be fixed by upgrading to backports, but doesn't hurt, compared to a non-functional entropy daemon.

Joulinar, 2020-08-27:
I know it's wrong place but at least it fits for creating a new device ID.

MichaIng, 2020-08-27:
Jep saw that, going to answer now.

MichaIng, 2020-08-27:
Okay done + I added Amlogic S905 detection with G_HW_CPUID=7 as this is the same SoC as Odroid C2 which allows potential driver (especially GPU) cross use, lets see.

StephanStS, 2020-08-27:
Image was generated and is ready for download.

StephanStS, 2020-08-27:
Due to the positive image generation with the new K2 ID, this issue can be closed.

The only thing I have to monitor is, that in https://dietpi.com/survey/#benchmark the ID 54 (NanoPi K2) and 55 (NanoPi R2s) will occur in the first column of the table.

closed 2020-08-27 by StephanStS

MichaIng, 2020-08-27:
> The only thing I have to monitor is, that in https://dietpi.com/survey/#benchmark the ID 54 (NanoPi K2) and 55 (NanoPi R2s) will occur in the first column of the table.

Ah, I'll upload the current version of the script to our server, so in at max one hour you'll see the SBC names 😄. Reminds me to upload a benchmark of my NEO3 as well.

StephanStS, 2020-08-27:
Ok, IDs 54 and 55 now show the SBC. Nice.