MichaIng/DietPi #3737: Automated install stuck on ping prior to any automation | RPi B+ |

emma-goldman, 2020-08-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
`G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=31
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
`
- Distro version | 'buster' or `10.4`
- Kernel version | `Linux workshop-cam1 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux`
- SBC model | RPi B+ (armv6l)
- Power supply used | Original RPi one purchased with an RPi3b
- SDcard used | SanDisk Ultra 16gb, 10/A1/U1 thingoes on it

#### Additional Information (if applicable)
- Software title | DietPi-Automation process
- Was the software title installed freshly or updated/migrated?
Never gets to installing anything outside of base stuff included in the .img

- Can this issue be replicated on a fresh installation of DietPi?
Yes, 5-6 times before I solved some hdmi issues and hooked up a monitor and keyboard to it.

#### Steps to reproduce

1. On desktop: "sudo dd bs=4M  if=DietPi_RPi-ARMv6-Buster.img of=/dev/sd* (whichever one it was at the time)
2. then copied dietpi.txt with this in it: 
```
# Added by DietPi:
AUTO_SETUP_SWAPFILE_SIZE=0
AUTO_SETUP_SWAPFILE_LOCATION=/var/swap
CONFIG_ENABLE_IPV6=0
CONFIG_PREFER_IPV4=1

#setup ssh(dropbear), wifi, gpio, motioneye & ftp
AUTO_SETUP_AUTOMATED=1                  #auto-setup c:
AUTO_SETUP_NET_WIFI_ENABLED=0           #wifi, 1 for pi zero, 0 for pi b+
#AUTO_SETUP_INSTALL_SOFTWARE_ID=104      #dropbear, installed by default if no ssh then uncomment
AUTO_SETUP_INSTALL_SOFTWARE_ID=136      #motionEye
AUTO_SETUP_INSTALL_SOFTWARE_ID=95       #ftp
```
3. wait.... and wait.... eventually ssh'd into it and htop to see what was up. first-run was on a slow loop. 
4. Reflash with a config.txt to fix the blank screen on the hdmi out and have a look while it ran first run.

#### Expected behaviour

- Unattanded/automated setup connects to the interwebs to install selected softwares

#### Actual behaviour

- Unattended/automated setup gets stuck during dietpi-firstrun when trying to check its internet connection by pinging 9.9.9.9. It fails and the whole setup fails.

#### Extra details

- When I booted the firstrun setup with a screen and keyboard attached it did the same thing, then I tried to boot again (without reflashing) and it wanted to do the "broken/unfinished first run". When it tried the last command it got stuck on pinging 9.9.9.9 to check the internet connection. I edited it to ping the lan dhcp/dns server instead after which the setup worked flawlessly.

(Looking at it now maybe it would have gotten further with IPv6 but the first two or three times the "prefer IPv4" was not on.)
firstrun log attached
[dietpi-firstrun-setup.log](https://github.com/MichaIng/DietPi/files/5126642/dietpi-firstrun-setup.log)

Joulinar, 2020-08-25:
Hi,

many thanks for your report. Once initial setup failed and you are back on command line, are you able to ping anything on the internet. Is the issue applicable for 9.9.9.9 only or for others like 8.8.8.8 or 1.1.1.1 as well?

emma-goldman, 2020-08-25:
I only tried to ping the local dns server after which it moved on with the install seeing as it actually had an internet connection - but I'll try and reproduce the error tomorrow. c:

MichaIng, 2020-08-26:
Same issue: https://github.com/MichaIng/DietPi/issues/3730
During earlier boot stage the default route/gateway is checked before it proceeds. With that, actually, the ping should not throw `connect: Network is unreachable` because a route is there. Instead it should wait for the defined time (`CONFIG_G_CHECK_URL_TIMEOUT ` by default 10s) and retry. Because if the missing route error exit it retries immediately, not giving DHCP enough time to receive an IP address.

Probably while waiting for DHCP the default route is again lost, needs testing.

emma-goldman, 2020-08-26:
Hmm.. Tried again with the same sd card, same setup and everything now and it seems to work fine. The only difference is that I'm the only one using the network at the moment. (

The pi in question is connected to an old tp-link router running openwrt as a dumb ap, which has a fritzbox upstream as the isp modem, local dns and dhcp server. Separately on the same network, there's a semi-recent asus router running asus merlin also acting as a dumb ap. At peak hours there are probably ~14-15 client devices (not counting the dumb ap:s) on the network. Maybe the fritzbox got a bit flustered and didn't forward the ping request in time. Seeing as there were about five android devices that do a lot of unnecessary call backs all the time it could be that.

I'll try this again during peak hours.

MichaIng, 2020-08-27:
Thanks for testing. I'd still not expect an overloaded router to block the ping but instead give a response delayed. But probably the longer/delayed DHCP response is what caused the missing route when ping is done.

Likely a solution would be to:
- Make `ifup@.service` `Type=oneshot`, so that `network-online.target` is really only reached when all network interfaces have fully been configured. Debian explicitly changed it to be not like that to fix some boot delays years ago, but actually nearly no Debian package uses `network-online.target` but nearly all network servers use `network.target` only, so that they may be started already when only the local loopback interface is configured.
- Our `CONFIG_BOOT_WAIT_FOR_NETWORK=2` option could then simply order `dietpi-boot.service` `After=network-online.target`. It does not wait unlimited time then, but 90 seconds max as this is the default max time that systemd units wait.
- During first boot, `dietpi-login` (where the ping and the first run update + installs are done) could check for `network-online.target` being reached before attempting any ping, or otherwise check for the default route/standard gateway to assure ping waits for the configured amount of time instead of throwing `connect: Network is unreachable`.

Definitely much to do about network, I already started to rewrite some scripts and add some functions as a basis but we want to think through everything carefully as this will include a planned larger network setup rewrite: https://github.com/MichaIng/DietPi/projects/1

MichaIng, 2020-09-13:
I uploaded a new RPi image which contains a change about how the default route is detected on boot. Probably it solves the issue in your case as well, if you want to test: https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Buster.7z

emma-goldman, 2020-09-13:
Sweet. I'll have a go with this image for a RPi Zero install I'm trying out soon.

With the old Pi B+ (original culprit) I ended up doing the install manually in the end. At this moment I'm setting up a RPi4 1GB ram using the old image seeing as I don't need 64bit for its purposes.

I think the old issue was solved using the CONFIG_BOOT_WAIT_FOR_NETWORK=2 option in dietpi.txt 

That being said it did behave strangely: the little green light wasn't flashing for a few minutes or so after I booted it up, so I logged in with dietpi through ssh and it told me
 ```
[ INFO ] DietPi-Login | [ERROR] Root login required

To finish DietPi first run setup, root permissions are required.                                                                                                     

Please login again as user "root" with password "dietpi", respectively the one you chose in "dietpi.txt".
```

So I exited and ssh:d back in with root where i was greeted with:

```
─────────────────────────────────────────────────────
 DietPi v6.30.0 : 23:28 - Sun 09/13/20
 ─────────────────────────────────────────────────────
 - LAN IP : 192.168.178.47 (eth0)
[  OK  ] DietPi-Login | Checking network connectivity
[  OK  ] DietPi-Login | Checking DNS resolver
```
Where it sat spinning for a few short moments after which it moved on with the automated install.

I do think that the original issue is resolved, because I'm not sure if the install was stuck on DNS before I ssh:d into it or if I just timed my ssh right/wrong?

I'll let this finish with the RPi4 and have a go with the new image for a webserver I wanna set up with an RPi Zero v1.3 (just got a usb hub with ethernet for it so I could try some stuff out) c:

MichaIng, 2020-09-13:
Jep great, that all looks like it should now. Yes on first login the update and setup steps require root permissions, from then on your can login with "dietpi".

A first login **was** btw always required: https://github.com/MichaIng/DietPi/issues/2520
But as of user reports a few months ago it is not anymore, at least not in all cases and on my last tests indeed with automation enabled it does not require this first manual login anymore. I'm not sure what changed, we had an autologin config for the main console tty1 applied in case of `AUTO_SETUP_AUTOMATED=1` for years, but it never worked for the very first boot (only prevented an additional login since we made an additional reboot after first update before first dietpi-software call that time). Meanwhile either in systemd something changed or some slight timing changes during boot process. However I do not yet trust that 100%, so lets see if that works with your RPi Zero as well 😉.

_The issue is that systemd initially loads all systemd units/services to check for order and dependencies before the first of them is executed, hence before our `dietpi-firstboot.service` reads `dietpi.txt` and can place the `getty@tt1.service` override for autologin when `AUTO_SETUP_AUTOMATED=1 is found. Usually systemd needs to be reloaded to recognise changes but doing this during boot might break things (e.g. overrides change unit dependencies which would require a different order while the current order has already been partly processed) or does not apply correctly. Since first boot is plausibly important, including network setup for headless systems, we must not touch anything that could break the systemd boot order while it's processing that. But obviously now systemd recognises changes by itself or some step implies a reload, or it's a question of timing in which order which unit and which target has finished/been reached in combination with hardware init state, not sure._

emma-goldman, 2020-09-13:
"A first login was btw always required: #2520" Right-ee-oh!

The RPi4 just finished setting everything up and it all worked as intended. (Haven't configured everything yet but the installs ran fine).

I'll have a go with the pi zero this week sometime c: 
In the meantime, thanks for a sweet system & great feedback! c:

eyduh, 2020-09-23:
> I'll have a go with the pi zero this week sometime c:

I thought I was running into the same issue with the piZero 1.3 but it turned out to be a hardware issue with the usb/ethernet hub - waiting on a replacement unit still.

MichaIng, 2020-09-27:
Okay I mark this issue as closed. Feel free to reopen if required.

closed 2020-09-27 by MichaIng