MichaIng/DietPi #3738: Error on SABnzbd reinstall to upgrade

simonfisher, 2020-08-25:
#### Details:
- Date           | Wed Aug 26 08:41:06 AEST 2020
- DietPi version | v6.31.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l) (ID=11)
- Kernel version | Linux DietPi 4.14.32+ #1 SMP PREEMPT Thu Apr 5 12:46:33 UTC 2018 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | python2.7 tools/make_mo.py
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. dietpi-software reinstall 139

#### Expected behaviour:

- successful reinstall/update
#### Actual behaviour:

- This error
#### Extra details:

- I'm guessing it's an issue with python compatibility
 - is it possible to install the latest python 3 SABnzbd RC? 
#### Additional logs:
```
Email MO files
Compile locale/he/LC_MESSAGES/SABemail.mo
Compile locale/pl/LC_MESSAGES/SABemail.mo
Compile locale/de/LC_MESSAGES/SABemail.mo
Compile locale/fi/LC_MESSAGES/SABemail.mo
Compile locale/nl/LC_MESSAGES/SABemail.mo
Compile locale/da/LC_MESSAGES/SABemail.mo
Compile locale/ro/LC_MESSAGES/SABemail.mo
Compile locale/sv/LC_MESSAGES/SABemail.mo
Compile locale/zh_CN/LC_MESSAGES/SABemail.mo
Compile locale/ru/LC_MESSAGES/SABemail.mo
Compile locale/es/LC_MESSAGES/SABemail.mo
Compile locale/en/LC_MESSAGES/SABemail.mo
Compile locale/sr/LC_MESSAGES/SABemail.mo
Compile locale/fr/LC_MESSAGES/SABemail.mo
Compile locale/nb/LC_MESSAGES/SABemail.mo
Compile locale/pt_BR/LC_MESSAGES/SABemail.mo
Create email templates from MO files
Create email template for fr
Traceback (most recent call last):
  File "tools/make_mo.py", line 295, in <module>
    make_templates()
  File "tools/make_mo.py", line 217, in make_templates
    translate_tmpl("email", lng)
  File "tools/make_mo.py", line 193, in translate_tmpl
    with open(EMAIL_DIR + "/%s-en.tmpl" % prefix, "r", encoding="utf-8") as src:
TypeError: 'encoding' is an invalid keyword argument for this function
```

simonfisher, 2020-08-25:
Also, now SABnzbd doesn't run:

```
[FAILED] DietPi-Services | ● sabnzbd.service - SABnzbd (DietPi)
   Loaded: loaded (/etc/systemd/system/sabnzbd.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2020-08-26 09:01:09 AEST; 189ms ago
     Docs: https://sabnzbd.org/wiki/
  Process: 17774 ExecStart=/usr/bin/python2.7 -OO /etc/sabnzbd/SABnzbd.py -b 0 -f /etc/sabnzbd/sabnzbd.ini (code=exited, status=1/FAILURE)
 Main PID: 17774 (code=exited, status=1/FAILURE)

Aug 26 09:01:09 DietPi systemd[1]: sabnzbd.service: Service RestartSec=100ms expired, scheduling restart.
Aug 26 09:01:09 DietPi systemd[1]: sabnzbd.service: Scheduled restart job, restart counter is at 5.
Aug 26 09:01:09 DietPi systemd[1]: Stopped SABnzbd (DietPi).
Aug 26 09:01:09 DietPi systemd[1]: sabnzbd.service: Start request repeated too quickly.
Aug 26 09:01:09 DietPi systemd[1]: sabnzbd.service: Failed with result 'exit-code'.
Aug 26 09:01:09 DietPi systemd[1]: Failed to start SABnzbd (DietPi).
```

Joulinar, 2020-08-25:
Hi,

many thanks for your report. I tried to install SABnzbd and at least I'm able to pass pip installation

<details>
  <summary>Install progress. Click to expand!</summary>

```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing SABnzbd: nzb download manager

[  OK  ] DietPi-Software | Checking URL: https://github.com/sabnzbd/sabnzbd/archive/master.tar.gz
[ INFO ] DietPi-Software | G_THREAD_START_0 | wget https://github.com/sabnzbd/sabnzbd/archive/master.tar.gz -O 139.tar
[ INFO ] DietPi-Software | APT install for: par2 p7zip-full libffi-dev libssl-dev, please wait...
Selecting previously unselected package libffi-dev:armhf.
(Reading database ... 28697 files and directories currently installed.)
Preparing to unpack .../libffi-dev_3.2.1-9_armhf.deb ...
Unpacking libffi-dev:armhf (3.2.1-9) ...
Selecting previously unselected package libssl-dev:armhf.
Preparing to unpack .../libssl-dev_1.1.1d-0+deb10u3+rpt1_armhf.deb ...
Unpacking libssl-dev:armhf (1.1.1d-0+deb10u3+rpt1) ...
Selecting previously unselected package p7zip-full.
Preparing to unpack .../p7zip-full_16.02+dfsg-6_armhf.deb ...
Unpacking p7zip-full (16.02+dfsg-6) ...
Selecting previously unselected package par2.
Preparing to unpack .../par2_0.8.0-1_armhf.deb ...
Unpacking par2 (0.8.0-1) ...
Setting up p7zip-full (16.02+dfsg-6) ...
Setting up libffi-dev:armhf (3.2.1-9) ...
Setting up par2 (0.8.0-1) ...
Setting up libssl-dev:armhf (1.1.1d-0+deb10u3+rpt1) ...
Processing triggers for install-info (6.5.0.dfsg.1-4+b1) ...
[  OK  ] DietPi-Software | APT install for: par2 p7zip-full libffi-dev libssl-dev
[  OK  ] DietPi-Software | G_THREAD: All threads finished
[  OK  ] DietPi-Software | tar xf 139.tar
[  OK  ] DietPi-Software | rm 139.tar
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting cheetah
  Downloading Cheetah-2.4.4.tar.gz (190 kB)
     |████████████████████████████████| 190 kB 1.3 MB/s
Collecting cryptography
  Downloading cryptography-3.0.tar.gz (534 kB)
     |████████████████████████████████| 534 kB 2.7 MB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting sabyenc
  Downloading sabyenc-3.3.6.tar.gz (9.8 kB)
Collecting Markdown>=2.0.1
  Downloading Markdown-3.1.1-py2.py3-none-any.whl (87 kB)
     |████████████████████████████████| 87 kB 1.2 MB/s
Collecting ipaddress; python_version < "3"
  Downloading ipaddress-1.0.23-py2.py3-none-any.whl (18 kB)
Collecting enum34; python_version < "3"
  Downloading enum34-1.1.10-py2-none-any.whl (11 kB)
Collecting six>=1.4.1
  Downloading six-1.15.0-py2.py3-none-any.whl (10 kB)
Processing /root/.cache/pip/wheels/b3/00/f4/3a31782a9301cd23506e4ea6e645b6c114562186bdf54296af/cffi-1.14.2-cp27-cp27mu-linux_armv7l.whl
Requirement already satisfied: setuptools>=36 in /usr/local/lib/python2.7/dist-packages (from Markdown>=2.0.1->cheetah) (44.1.1)
Collecting pycparser
  Using cached pycparser-2.20-py2.py3-none-any.whl (112 kB)
Building wheels for collected packages: cheetah, cryptography, sabyenc
  Building wheel for cheetah (setup.py) ... done
  Created wheel for cheetah: filename=Cheetah-2.4.4-cp27-cp27mu-linux_armv7l.whl size=191968 sha256=79938485ec208fce3b488c16b8e29cd87de2690efea3cb32fb12e11fb85f4512
  Stored in directory: /root/.cache/pip/wheels/90/b0/e1/c6661f8fac2bb23914d9289c76df5d3734c94be7233beb46a5
  Building wheel for cryptography (PEP 517) ... done
  Created wheel for cryptography: filename=cryptography-3.0-cp27-cp27mu-linux_armv7l.whl size=824472 sha256=6b1e547205915cf9d4eaabfdb40953d857b7f54787e3efd1247feaa9dfa2bfeb
  Stored in directory: /root/.cache/pip/wheels/75/28/ac/5ce34343273111ce829f86f19840187a711292692861654631
  Building wheel for sabyenc (setup.py) ... done
  Created wheel for sabyenc: filename=sabyenc-3.3.6-cp27-cp27mu-linux_armv7l.whl size=22316 sha256=41836430e407bf3a0a4ca4335473551b5f0fb565c353609e73964fe1189d6ffa
  Stored in directory: /root/.cache/pip/wheels/c4/cc/0a/3f534c77fce243827fe4b3120c9f6e4fddb19b9aff1d4b9525
Successfully built cheetah cryptography sabyenc
Installing collected packages: Markdown, cheetah, ipaddress, enum34, six, pycparser, cffi, cryptography, sabyenc
Successfully installed Markdown-3.1.1 cffi-1.14.2 cheetah-2.4.4 cryptography-3.0 enum34-1.1.10 ipaddress-1.0.23 pycparser-2.20 sabyenc-3.3.6 six-1.15.0
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
```
</details>

however for me installation stuck afterwards

```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Optimise and configure software

[ INFO ] DietPi-Software | Applying DietPi optimisations and configurations for RPi 3 Model B+ (armv7l), please wait...

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring SABnzbd: nzb download manager

[ INFO ] DietPi-Software | Generating initial config, please wait...

```
But yours seems to fail on compiling some email templates ??

Your forum Post for reference https://dietpi.com/phpbb/viewtopic.php?f=11&t=8018

MichaIng, 2020-08-26:
We migrated to a Python3 setup with DietPi v6.32: https://github.com/MichaIng/DietPi/pull/3657
I'll release a first beta tonight, please test it there.

simonfisher, 2020-08-26:
Thanks. Just quickly, how do I install a beta version? Couldn’t find that info

Joulinar, 2020-08-27:
Hi, new Beta is not yet available.

MichaIng, 2020-08-27:
New it is, all info: #3743

Joulinar, 2020-08-27:
ok installation of sabnzbd using python3 is working fine on Beta.

MichaIng, 2020-08-27:
Great to be reminded that we're on the pre-release branch currently while SABnzbd 3 is on stable master branch already now 👍: https://github.com/sabnzbd/sabnzbd/tree/master
Done: https://github.com/MichaIng/DietPi/commit/9c8e793ed39f55c5686f5861c20befe5d685b670

simonfisher, 2020-08-27:
Was able to update to beta, and reinstall sabnzbd successfully thanks!

MichaIng, 2020-08-27:
Many thanks for testing, great to hear!

closed 2020-08-31 by MichaIng