MichaIng/DietPi #3739: Trying to update

snoodler, 2020-08-26:
#### Details:
- Date           | Wed 26 Aug 12:17:53 BST 2020
- Bug report     | 6fdb9c44-6939-4839-9232-1aa88405a615
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 4.19.93-v7l+ #1290 SMP Fri Jan 10 16:45:11 GMT 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. ... i just tried to update with dietpi-update
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...it  gives me an error

#### Additional logs:
```
Setting up libapache2-mod-php7.3 (7.3.19-1~deb10u1) ...
dpkg: error processing package libapache2-mod-php7.3 (--configure):
 installed libapache2-mod-php7.3 package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 libapache2-mod-php7.3
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

---------------------------------------------------------------------

[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

MichaIng, 2020-08-26:
Many thanks for your report. Please run the following and post the full output here:
```
apt install libapache2-mod-php7.3
```

snoodler, 2020-08-26:
root@DietPi:~# apt install libapache2-mod-php7.3
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libapache2-mod-php7.3 is already the newest version (7.3.19-1~deb10u1).
The following package was automatically installed and is no longer required:
  libmicrodns0
Use 'apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 18 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up libapache2-mod-php7.3 (7.3.19-1~deb10u1) ...
dpkg: error processing package libapache2-mod-php7.3 (--configure):
 installed libapache2-mod-php7.3 package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 libapache2-mod-php7.3
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@DietPi:~#

MichaIng, 2020-08-26:
Could you please do and paste the following:
```
cat /var/lib/dpkg/info/libapache2-mod-php7.3.postinst
dash -x /var/lib/dpkg/info/libapache2-mod-php7.3.postinst configure '7.3.19-1~deb10u1'
```

snoodler, 2020-08-26:
root@DietPi:~# cat /var/lib/dpkg/info/libapache2-mod-php7.3.postinst
#!/bin/sh

set -e

if [ -e /usr/share/apache2/apache2-maintscript-helper ]; then
    . /usr/share/apache2/apache2-maintscript-helper

    php_enable() {
	local a2query_ret=0
	a2query -m "php7.3" > /dev/null 2>&1 || a2query_ret=$?
	if [ "$a2query_ret" -eq 0 ] ; then
	    apache2_msg "info" "$DPKG_MAINTSCRIPT_PACKAGE: not switching MPM - already enabled"
	    return 1
	elif [ "$a2query_ret" -eq 32	 ] ; then
	    apache2_msg "info" "$DPKG_MAINTSCRIPT_PACKAGE: not switching MPM - disabled by maintainer"
	    return 1
	fi

	PHP_MODULE=$(a2query -m | sed -n 's/^\(php[\.0-9]*\) (enabled.*)/\1/p')
        if [ -n "$PHP_MODULE" -a "$PHP_MODULE" != "php7.3" ]; then
	    apache2_msg "err" "$DPKG_MAINTSCRIPT_PACKAGE: $PHP_MODULE module already enabled, not enabling PHP 7.3"
	    return 1
	fi

	mpm=$(a2query -M)
	case "$(a2query -M)" in
	    prefork|itk) return 0;;
	    *) if apache2_switch_mpm prefork; then return 0; fi;;
	esac
        apache2_msg "err" "$DPKG_MAINTSCRIPT_PACKAGE: Could not switch to prefork MPM, not enabling PHP 7.3"
	return 1
    }

else
    echo "Warning: Could not load Apache 2.4 maintainer script helper."

    php_enable() {
        return 1;
    }

fi

# we've registered a trigger to handle extension updates.
if [ "$1" = "triggered" ] && [ "$2" = "/etc/php/7.3/apache2/conf.d" ]; then
    if [ -e /usr/share/apache2/apache2-maintscript-helper ]; then
	. /usr/share/apache2/apache2-maintscript-helper
	apache2_reload restart
    fi
    exit 0
elif [ "$1" != "configure" ]; then
    exit 0
fi

CONFDIR=/etc/php/7.3/apache2/conf.d

if [ "$1" = "configure" ]; then
    phpini="/etc/php/7.3/apache2/php.ini"

    if [ "apache2" = "cli" ]; then
	ucf /usr/lib/php/7.3/php.ini-production.apache2 $phpini
    else
	ucf /usr/lib/php/7.3/php.ini-production $phpini
    fi
    ucfr libapache2-mod-php7.3 $phpini

    if [ -e /usr/lib/php/php-maintscript-helper ]; then
	. /usr/lib/php/php-maintscript-helper
	
	mods=$(phpquery -M -v 7.3)
[... 494 lines cut ...]
+ [ -n 1 ]
+ return
+ apache2_invoke enmod php7.3
+ local CMD=enmod
+ local CONF=php7.3
+ local RCD_ACTION=
+ local invoke_rcd=0
+ local check_switch=
+ local invoke_string=
+ [ -x /usr/sbin/a2enmod ]
+ [ -x /usr/sbin/a2query ]
+ [ -n  ]
+ check_switch=-m
+ invoke_string=module
+ rcd_action=restart
+ local a2query_ret=0
+ a2query -m php7.3
+ a2query_ret=1
+ [ 1 -eq 0 ]
+ [ 1 -eq 32 ]
+ APACHE2_NEED_ACTION=1
+ a2enmod -m -q php7.3
+ return 1
+ exit 1
root@DietPi:~#

MichaIng, 2020-08-26:
The postinst script fails to enable php7.3 module for Apache. Can you please try this manually:
```
a2enmod php7.3
```
And check if the files are there:
```
ls -l /etc/apache2/mods-available/php7.3*
ls -l /etc/apache2/mods-enabled/php7.3*
```

snoodler, 2020-08-26:
root@DietPi:~# a2enmod php7.3
ERROR: Module php7.3 does not exist!
root@DietPi:~# ls -l /etc/apache2/mods-available/php7.3*
ls: cannot access '/etc/apache2/mods-available/php7.3*': No such file or directory
root@DietPi:~# ls -l /etc/apache2/mods-enabled/php7.3*
ls: cannot access '/etc/apache2/mods-enabled/php7.3*': No such file or directory
root@DietPi:~#

MichaIng, 2020-08-26:
Ehm, okay the files are missing, that is a good reason. But those are part of the package 🤔:
```
dpkg -L libapache2-mod-php7.3
```
Please try to reinstall:
```
apt install --reinstall libapache2-mod-php7.3
```

snoodler, 2020-08-26:
root@DietPi:~# dpkg -L libapache2-mod-php7.3
/.
/etc
/etc/apache2
/etc/apache2/mods-available
/etc/apache2/mods-available/php7.3.conf
/etc/apache2/mods-available/php7.3.load
/etc/php
/etc/php/7.3
/etc/php/7.3/apache2
/etc/php/7.3/apache2/conf.d
/usr
/usr/lib
/usr/lib/apache2
/usr/lib/apache2/modules
/usr/lib/apache2/modules/libphp7.3.so
/usr/lib/php
/usr/lib/php/7.3
/usr/lib/php/7.3/sapi
/usr/lib/php/7.3/sapi/apache2
/usr/share
/usr/share/bug
/usr/share/bug/libapache2-mod-php7.3
/usr/share/bug/libapache2-mod-php7.3/control
/usr/share/bug/libapache2-mod-php7.3/script
/usr/share/doc
/usr/share/lintian
/usr/share/lintian/overrides
/usr/share/lintian/overrides/libapache2-mod-php7.3
/usr/share/doc/libapache2-mod-php7.3
root@DietPi:~# apt install --reinstall libapache2-mod-php7.3
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  libmicrodns0
Use 'apt autoremove' to remove it.
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 18 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
E: Internal Error, No file name for libapache2-mod-php7.3:armhf
root@DietPi:~#

MichaIng, 2020-08-26:
Very strange one, the meta files must have been corrupted somehow. Found one match: https://askubuntu.com/questions/266450

Lets try to cleanup from ground:
```
/boot/dietpi/func/dietpi-set_software apt-cache  clean
apt update
apt -f install
(( $? )) && apt install --reinstall libapache2-mod-php7.3
(( $? )) && dpkg -i /var/cache/apt/archives/libapache2-mod-php7.3*.deb
apt upgrade
(( $? )) && apt install --reinstall libapache2-mod-php7.3
dpkg --configure -a
```
- Some steps are doubled and others only done if the prior step failed, based on what the guys on askubuntu.com tried. Let's see if it helps.

snoodler, 2020-08-27:
root@DietPi:~# apt -f install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libllvm8 libmicrodns0
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up libapache2-mod-php7.3 (7.3.19-1~deb10u1) ...
dpkg: error processing package libapache2-mod-php7.3 (--configure):
 installed libapache2-mod-php7.3 package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 libapache2-mod-php7.3
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@DietPi:~# apt -f install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libllvm8 libmicrodns0
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up libapache2-mod-php7.3 (7.3.19-1~deb10u1) ...
dpkg: error processing package libapache2-mod-php7.3 (--configure):
 installed libapache2-mod-php7.3 package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 libapache2-mod-php7.3
E: Sub-process /usr/bin/dpkg returned an error code (1)

MichaIng, 2020-08-27:
Yes, I expected that, please go on with the next commands, they include other install approaches.

snoodler, 2020-08-27:
I tried all of the ones in that page, still the same results... cant I just uninstall and install apache again? I dont undertsand what the problem is.

root@DietPi:~# sudo dpkg --configure -a
Setting up libapache2-mod-php7.3 (7.3.19-1~deb10u1) ...
dpkg: error processing package libapache2-mod-php7.3 (--configure):
 installed libapache2-mod-php7.3 package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 libapache2-mod-php7.3

MichaIng, 2020-08-27:
Can you please try again:
```
dpkg -i /var/cache/apt/archives/libapache2-mod-php7.3*.deb
```
I added it with  `(( $? )) &&` which made it only execute if the previous command failed, so those were meant to be executed directly behind each other via copy&paste the whole block.

However if that fails, lets verify the existence and consistency of all required files:
```
ls -l /var/cache/apt/archives/libapache2-mod-php7.3*.deb
ls -l /var/lib/dpkg/info/libapache2-mod-php7.3*
dpkg-deb -x /var/cache/apt/archives/libapache2-mod-php7.3*.deb /tmp/libapache2-mod-php7.3
ls -al /tmp/libapache2-mod-php7.3/etc/apache2/mods-available
```

And now I have an idea:
- `/etc/apache2/mods-available/php7.3.conf` is a config file: `cat /var/lib/dpkg/info/libapache2-mod-php7.3*.conffiles`
- If a config file is manually removed, dpkg with `--force-defconf`/`--force-oldconf` option (preserve changed config files automatically) will not recover it.
- If this is the reason, the following should work: `dpkg --force-confmiss --configure -a`

snoodler, 2020-08-27:
root@DietPi:~# ls -l /var/cache/apt/archives/libapache2-mod-php7.3*.deb
ls: cannot access '/var/cache/apt/archives/libapache2-mod-php7.3*.deb': No such file or directory
root@DietPi:~# ls -l /var/lib/dpkg/info/libapache2-mod-php7.3*
-rw-r--r-- 1 root root   80 Jul  5 07:46 /var/lib/dpkg/info/libapache2-mod-php7.3.conffiles
-rw-r--r-- 1 root root  679 Aug 26 12:06 /var/lib/dpkg/info/libapache2-mod-php7.3.list
-rw-r--r-- 1 root root  373 Jul  5 07:46 /var/lib/dpkg/info/libapache2-mod-php7.3.md5sums
-rwxr-xr-x 1 root root 2476 Jul  5 07:46 /var/lib/dpkg/info/libapache2-mod-php7.3.postinst
-rwxr-xr-x 1 root root 1051 Jul  5 07:46 /var/lib/dpkg/info/libapache2-mod-php7.3.postrm
-rwxr-xr-x 1 root root  828 Jul  5 07:46 /var/lib/dpkg/info/libapache2-mod-php7.3.prerm
-rw-r--r-- 1 root root   45 Jul  5 07:46 /var/lib/dpkg/info/libapache2-mod-php7.3.triggers
root@DietPi:~# dpkg-deb -x /var/cache/apt/archives/libapache2-mod-php7.3*.deb /tmp/libapache2-mod-php7.3
dpkg-deb: error: failed to read archive '/var/cache/apt/archives/libapache2-mod-php7.3*.deb': No such file or directory
root@DietPi:~# ls -al /tmp/libapache2-mod-php7.3/etc/apache2/mods-available
ls: cannot access '/tmp/libapache2-mod-php7.3/etc/apache2/mods-available': No such file or directory
root@DietPi:~#

MichaIng, 2020-08-27:
Ah the archive has been clean. Please run `apt install --reinstall libapache2-mod-php7.3` first (which downloads the archive) and then the steps again.

snoodler, 2020-08-27:
root@DietPi:~# dpkg --force-confmiss --configure -a
Setting up libapache2-mod-php7.3 (7.3.19-1~deb10u1) ...
dpkg: error processing package libapache2-mod-php7.3 (--configure):
 installed libapache2-mod-php7.3 package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 libapache2-mod-php7.3
root@DietPi:~#

snoodler, 2020-08-27:
root@DietPi:~# apt install --reinstall libapache2-mod-php7.3
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libllvm8 libmicrodns0
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
E: Internal Error, No file name for libapache2-mod-php7.3:armhf
root@DietPi:~#

MichaIng, 2020-08-27:
Okay, lets try it manually:
```
cd /tmp
wget http://raspbian.raspberrypi.org/raspbian/pool/main/p/php7.3/libapache2-mod-php7.3_7.3.19-1~deb10u1_armhf.deb
dpkg --force-confmiss -i libapache2-mod-php7.3_7.3.19-1~deb10u1_armhf.deb
```

snoodler, 2020-08-27:
that worked, thank you dietpi-update was successful

closed 2020-08-27 by snoodler

MichaIng, 2020-08-27:
Finally, good to know and keep that option in mind. Of course such basically required config files should not be removed in the first place, but sometimes things happen accidentally.