MichaIng/DietPi #3745: ?

quadews, 2020-08-28:
#### Details:
- Date           | Fri Aug 28 08:00:07 BST 2020
- DietPi version | v6.31.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi Zero (armv6l) (ID=1)
- Kernel version | Linux DietPi 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | wget --spider -t 1 -T 10 https://builds.cuberite.org/job/Cuberite%20Linux%20raspi-armhf%20Master/lastSuccessfulBuild/artifact/Cuberite.tar.gz
- Exit code      | 8
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Spider mode enabled. Check if remote file exists.
--2020-08-28 07:59:59--  https://builds.cuberite.org/job/Cuberite%20Linux%20raspi-armhf%20Master/lastSuccessfulBuild/artifact/Cuberite.tar.gz
Resolving builds.cuberite.org (builds.cuberite.org)... 91.121.65.133, 2001:41d0:1:7485::1
Connecting to builds.cuberite.org (builds.cuberite.org)|91.121.65.133|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
Remote file does not exist -- broken link!!!
```
****

Joulinar, 2020-08-28:
Hi,

many thanks for your report. Currently the download link is broken. You can use following instruction as workaround. 

https://github.com/MichaIng/DietPi/issues/3709#issuecomment-671851600

a permanent fix will be implemented with upcoming DietPi release 6.32

closed 2020-08-28 by quadews

MichaIng, 2020-08-28:
You can test it with the DietPi beta: https://github.com/MichaIng/DietPi/issues/3743
There was another workaround required for RPi Zero since the current official Cuberite armhf builds stopped supporting ARMv6: #3664
Since I have no ARMv6 RPi model, would be great to have another tester 🙂.

quadews, 2020-08-28:
[image]
its working soo good.
i will use beta in soon time
thanks for the suggestion  👍

MichaIng, 2020-08-28:
Wait, did you just change the download link and that was enough to get it running on RPi Zero?
This would mean the current builds support ARMv6 again, great but I would have expected some info about it here 😄: https://github.com/cuberite/cuberite/issues/4774
@CactiChameleon9 can you verify?

quadews, 2020-08-28:
current builds not working its get segmentation fail im use this version 

https://dietpi.com/downloads/binaries/buster/cuberite_armv6l.7z

MichaIng, 2020-08-28:
Ah okay, so it's still as expected. Good, on ARMv6 we place this binary automatically with v6.32. Since the build is quite simple, we should be able to update the binary on regular basis.